Marus/cortex-debug #1047: help: fail to set break point，debug console output：No source file named vscode-local

cmbjxxiao, 2024-09-25:
1. Below is the debug console output after pausing

-> ~"208\t    LL_PERIPH_WE(LL_PERIPH_ALL);\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"208\t    LL_PERIPH_WE(LL_PERIPH_ALL);\n"}]}
208	    LL_PERIPH_WE(LL_PERIPH_ALL);
-> *stopped,reason="breakpoint-hit",disp="del",bkptno="1",frame={addr="0x000a048a",func="main",args=[],file="../../../device/board/yyy/xxxx_board/kkk/board/main.c",fullname="\\\\wsl$\\Ubuntu-20.04\\home\\ccc\\code\\Mytest\\device\\board\\yyy\\xxxx_board\\kkk\\board\\main.c",line="208",arch="armv7e-m"},thread-id="1",stopped-threads="all"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["reason","breakpoint-hit"],["disp","del"],["bkptno","1"],["frame",[["addr","0x000a048a"],["func","main"],["args",[]],["file","../../../device/board/yyy/xxxx_board/kkk/board/main.c"],["fullname","\\\\wsl$\\Ubuntu-20.04\\home\\ccc\\code\\Mytest\\device\\board\\yyy\\xxxx_board\\kkk\\board\\main.c"],["line","208"],["arch","armv7e-m"]]],["thread-id","1"],["stopped-threads","all"]]}]}
mi2.status = stopped
-> =breakpoint-deleted,id="1"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"breakpoint-deleted","output":[["id","1"]]}]}
21-thread-list-ids
-> 21^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
GDB -> App: {"output":"","token":21,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
22-thread-info 1
-> 22^done,threads=[{id="1",target-id="Thread 57005",frame={level="0",addr="0x000a048a",func="main",args=[],file="../../../device/board/yyy/xxxx_board/kkk/board/main.c",fullname="\\\\wsl$\\Ubuntu-20.04\\home\\ccc\\code\\Mytest\\device\\board\\yyy\\xxxx_board\\kkk\\board\\main.c",line="208",arch="armv7e-m"},state="stopped"}]
GDB -> App: {"output":"","token":22,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","1"],["target-id","Thread 57005"],["frame",[["level","0"],["addr","0x000a048a"],["func","main"],["args",[]],["file","../../../device/board/yyy/xxxx_board/kkk/board/main.c"],["fullname","\\\\wsl$\\Ubuntu-20.04\\home\\ccc\\code\\Mytest\\device\\board\\yyy\\xxxx_board\\kkk\\board\\main.c"],["line","208"],["arch","armv7e-m"]]],["state","stopped"]]]]]}}
23-stack-list-frames --thread 1 0 19
-> 23^done,stack=[frame={level="0",addr="0x000a048a",func="main",file="../../../device/board/yyy/xxxx_board/kkk/board/main.c",fullname="\\\\wsl$\\Ubuntu-20.04\\home\\ccc\\code\\Mytest\\device\\board\\yyy\\xxxx_board\\kkk\\board\\main.c",line="208",arch="armv7e-m"}]
GDB -> App: {"output":"","token":23,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0x000a048a"],["func","main"],["file","../../../device/board/yyy/xxxx_board/kkk/board/main.c"],["fullname","\\\\wsl$\\Ubuntu-20.04\\home\\ccc\\code\\Mytest\\device\\board\\yyy\\xxxx_board\\kkk\\board\\main.c"],["line","208"],["arch","armv7e-m"]]]]]]}}
24-stack-info-frame --thread 1 --frame 0
-> 24^done,frame={level="0",addr="0x000a048a",func="main",file="../../../device/board/yyy/xxxx_board/kkk/board/main.c",fullname="\\\\wsl$\\Ubuntu-20.04\\home\\ccc\\code\\Mytest\\device\\board\\yyy\\xxxx_board\\kkk\\board\\main.c",line="208",arch="armv7e-m"}
GDB -> App: {"output":"","token":24,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["frame",[["level","0"],["addr","0x000a048a"],["func","main"],["file","../../../device/board/yyy/xxxx_board/kkk/board/main.c"],["fullname","\\\\wsl$\\Ubuntu-20.04\\home\\ccc\\code\\Mytest\\device\\board\\yyy\\xxxx_board\\kkk\\board\\main.c"],["line","208"],["arch","armv7e-m"]]]]}}
25-stack-list-variables --thread 1 --frame 0 --simple-values
-> 25^done,variables=[{name="ret",type="int",value="537264120"}]
GDB -> App: {"output":"","token":25,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["variables",[[["name","ret"],["type","int"],["value","537264120"]]]]]}}
26-var-create --thread 1 --frame 0 var_ret_4096 @ "ret"
-> 26^done,name="var_ret_4096",numchild="0",value="537264120",type="int",has_more="0"
GDB -> App: {"output":"","token":26,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["name","var_ret_4096"],["numchild","0"],["value","537264120"],["type","int"],["has_more","0"]]}}

2. Below is the debug console output after setting breakpoint

28-break-insert "vscode-local:/Ubuntu-20.04/home/ccc/code/Mytest/device/board/yyy/xxxx_board/kkk/board/main.c:212"
-> 28^error,msg="No source file named vscode-local."
GDB -> App: {"output":"","token":28,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","No source file named vscode-local."]]}}

cmbjxxiao, 2024-09-25:
develop env:  vscode + wsl + cortex debug

main.c path in windows: \\wsl.localhost\Ubuntu-20.04\home\ccc\code\Mytest\device\board\yyy\xxxx_board\kkk\board\main.c

launch.json:
{
            "name": "Debug with JLink",
            "type": "cortex-debug",
            "request": "launch",
            "executable": "out/xxxx_board/xxxx_product_demo/unstripped/bin/xxxx_board.elf",
            "cwd":"${workspaceFolder}",
            "runToEntryPoint": "main",
            "servertype": "jlink",
            "device": "xxxx",
            "showDevDebugOutput": "both",
 }

cmbjxxiao, 2024-09-25:
after pressing pausing button, the debug gui is work fine, and the debug console log shows main.c's fullname is started by "\\wsl$\Ubuntu20.04"

but after click mouse to set break point,  the console output error message and main.c's fullname is started by "vscode-local:/Ubuntu-20.04"

haneefdm, 2024-09-25:
> develop env: vscode + wsl + cortex debug

I don't think this is enough info for me to help. I can't tell where you are compiling, linking or debugging. You have to be very aware of the OS and your current directory. From what I remember, you have to do everything from WSL so that windows path names don't even appear. This is what one vendor recommends (ignore stuff that is not relevant)

https://www.infineon.com/dgdl/Infineon-Running_ModusToolbox_TMusing_Windows_Subsystem_for_LinuxWSL-ApplicationNotes-v02_00-EN.pdf?fileId=8ac78c8c8afe5bd0018b4443a412010e&da=t

Maybe others can help.

Just so you know, Cortex-Debug does not do anything special for WSL and doesn't even know what WSL is. Pleaople have used gdb features such as substitute paths, sourcemap etc to overcome mixed environments. Those are gdb features and you can use them via launch json or .gdbinit

closed 2025-01-16 by haneefdm