Marus/cortex-debug #1048: Attach Session fails to issue command "ThreadExtraInfo" in contrast to Launch Session.

thood-masterclock, 2024-10-07:
**Description**

When running an attach command on a running target, then breaking on a breakpoint, the titles of the threads in the call stacks are not the custom strings created by my JLink plugin, but rather Thread {ThreadID} as shown in the screenshots. After working with SEGGER, and having them reproduce the problem, they said that the debugging client (VSCode/Cortex-Debug) was not issuing the "ThreadExtraInfo" command in the attach session.

This is in contrast to the launch session, which works perfectly.

**To Reproduce**

Attach to a running session with multiple threads, then break with a breakpoint. The threads should have a custom string accessible from the GDB server.

**Expected behavior**

The call stack view displays a custom string for each thread name reported by the GDB server.

**Screenshots**

Attach session (incorrect):
[image]
Launch session (correct):
[image]

**Environment**

WSL Ubuntu 24.04, Windows, Ubuntu 24.04
JLink Tools V798i
Custom JLink RTOS plugin (should be transparent to Cortex-Debug)
Cortex-Debug V1.12.1

**launch.json`**

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Launch G6 Application",
            "cwd": "${workspaceFolder}",
            "executable": "Bin/G6-Application-Debug/K-Controller_G6/MfdPrimary_5",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "jlink",
            "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
            "device": "MK66FN2M0xxx18",
            "interface": "jtag",
            "rtos": "${workspaceFolder}/MQX_GDB_JLink_Plugin/build/release/libRTOSPlugin_MQX42",
            "armToolchainPath": "/usr/bin"
        },
        {
            "name": "Attach G6 Application",
            "cwd": "${workspaceFolder}",
            "executable": "Bin/G6-Application-Debug/K-Controller_G6/MfdPrimary_5",
            "request": "attach",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "jlink",
            "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
            "device": "MK66FN2M0xxx18",
            "interface": "jtag",
            "rtos": "${workspaceFolder}/MQX_GDB_JLink_Plugin/build/release/libRTOSPlugin_MQX42",
            "armToolchainPath": "/usr/bin"
        }
    ]
}

thood-mc, 2024-10-09:
After looking through the source code in gdb.ts and doing some experimentation, it seems like the issue isn't due to different behavior when parsing the string (know to mi2 as details=""). Running "info threads" through the standard GDB console also doesn't report the "details" to the programmer.

However, in bare gdb-multiarch connected to the server, it reports this information just fine, both to the console and mi2 interpreters. Seems to me that something about the state of the GDB client is disabling this part of the thread information.

thood-mc, 2024-10-25:
The issue turned out to be in GDB itself, though this project might have affected this specific condition. Closing for now.

closed 2024-10-25 by thood-mc