Marus/cortex-debug #1051: reset after setting PC (on RT1170)

mastupristi, 2024-10-12:
I'm using vscode 1.94.0, cortex-debug 1.12.1 preview, Segger JLink 7.98h and *arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi* toolchain.
JLink script and settings file are copied from NXP SDK:
- [evkbmimxrt1170_connect_cm4_cm4side.jlinkscript](https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/main/evkbmimxrt1170/demo_apps/hello_world_secondary_core_boot/cm4/evkbmimxrt1170_connect_cm4_cm4side.jlinkscript)

The board I am testing with is NXP MIMXRT1170 EVKB.
The problem I'm having is a reset issued after writing the flash.
In fact, writing the flash also sets the PC, the subsequent reset nullifies this setting.

```
/opt/SEGGER/JLink/JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MIMXRT1176xxxA_M4 -jlinkscriptfile /home/max/Lavori/ama/srcs/repos/ama-testboard/FW/CM4/evkbmimxrt1170_connect_cm4_cm4side.jlinkscript -settingsfile /home/max/Lavori/ama/srcs/repos/ama-testboard/FW/CM4/JLink_Debug_SettingsFile.jlink -endian little -noir -vd -halt -reportuseraction -gui
SEGGER J-Link GDB Server V7.98h Command Line Version
 
JLinkARM.dll V7.98h (DLL compiled Sep 11 2024 14:25:50)
 
Command line: -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MIMXRT1176xxxA_M4 -jlinkscriptfile /home/max/Lavori/ama/srcs/repos/ama-testboard/FW/CM4/evkbmimxrt1170_connect_cm4_cm4side.jlinkscript -settingsfile /home/max/Lavori/ama/srcs/repos/ama-testboard/FW/CM4/JLink_Debug_SettingsFile.jlink -endian little -noir -vd -halt -reportuseraction -gui
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               on
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 /home/max/Lavori/ama/srcs/repos/ama-testboard/FW/CM4/evkbmimxrt1170_connect_cm4_cm4side.jlinkscript
J-Link settings file:          /home/max/Lavori/ama/srcs/repos/ama-testboard/FW/CM4/JLink_Debug_SettingsFile.jlink
------Target related settings------
Target device:                 MIMXRT1176xxxA_M4
Target device parameters:      none
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little
 
Connecting to J-Link...
J-Link is connected.
Device "MIMXRT1176XXXA_M4" selected.
Firmware: J-Link Ultra V4 compiled Sep 22 2022 15:00:10
Hardware: V4.00
S/N: 504400379
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.24 V
Listening on TCP/IP port 50000
Connecting to target...
InitTarget() start
InitTarget() end - Took 13.1ms
Found SW-DP with ID 0x6BA02477
DPIDR: 0x6BA02477
CoreSight SoC-400 or earlier
AP map detection skipped. Manually configured AP map found.
AP[0]: AHB-AP (IDR: Not set, ADDR: 0x00000000)
AP[1]: AHB-AP (IDR: Not set, ADDR: 0x00000000)
AP[2]: APB-AP (IDR: Not set, ADDR: 0x00000000)
AP[1]: Core found
AP[1]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
[0][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
[0][1]: E0001000 CID B105E00D PID 003BB002 DWT
[0][2]: E0002000 CID B105E00D PID 002BB003 FPB
[0][3]: E0000000 CID B105E00D PID 003BB001 ITM
[0][5]: E0041000 CID B105900D PID 000BB925 ETM
[0][7]: E0043000 CID B105900D PID 001BB908 CSTF
[0][8]: E0042000 CID B105900D PID 005BB906 CTI
Halting core...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
GDB client (conn. 12) requested target.xml from GDB Server
Reading common registers: Read register 'r0' (4 bytes) from hardware: 0x00000000
Read register 'r1' (4 bytes) from hardware: 0x00000000
Read register 'r2' (4 bytes) from hardware: 0x00000000
Read register 'r3' (4 bytes) from hardware: 0x00000000
Read register 'r4' (4 bytes) from hardware: 0x00000000
Read register 'r5' (4 bytes) from hardware: 0x00000000
Read register 'r6' (4 bytes) from hardware: 0x00000000
Read register 'r7' (4 bytes) from hardware: 0x00000000
Read register 'r8' (4 bytes) from hardware: 0x00000000
Read register 'r9' (4 bytes) from hardware: 0x00000000
Read register 'r10' (4 bytes) from hardware: 0x00000000
Read register 'r11' (4 bytes) from hardware: 0x00000000
Read register 'r12' (4 bytes) from hardware: 0x00000000
Read register 'sp' (4 bytes) from hardware: 0x00102020
Read register 'lr' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'pc' (4 bytes) from hardware: 0x40F02300
Read register 'xpsr' (4 bytes) from hardware: 0x00000001
Read 4 bytes @ address 0x0023F040 (Data = 0x0000E7FE)
Read 2 bytes @ address 0x0023F040 (Data = 0xE7FE)
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x0023F040)
Received monitor command: reset
ResetTarget() start                                <---- first reset issued
HandleBeforeMemAccessWrite() start
HandleBeforeMemAccessWrite() end - Took 529us
CM4 SRC reset
ResetTarget() end - Took 11.4ms
Resetting target
Downloading 1200 bytes @ address 0x08002000 - Verified OK
Downloading 936 bytes @ address 0x080024B0 - Verified OK
Downloading 16088 bytes @ address 0x08002858 - Verified OK
Downloading 16192 bytes @ address 0x08006730 - Verified OK
Downloading 5040 bytes @ address 0x0800A670 - Verified OK
Downloading 260 bytes @ address 0x0800BA20 - Verified OK
Downloading 16108 bytes @ address 0x0800BB24 - Verified OK
Downloading 16032 bytes @ address 0x0800FA10 - Verified OK
Downloading 11908 bytes @ address 0x080138B0 - Verified OK
Downloading 36 bytes @ address 0x08016734 - Verified OK
Downloading 8 bytes @ address 0x08016758 - Verified OK
Downloading 2540 bytes @ address 0x08016760 - Verified OK
Writing register 'pc' = 0x0800206C                 <---- writing flash and set PC
HandleBeforeFlashProg() start
HandleBeforeFlashProg() end - Took 2.20ms
J-Link: Flash download: Bank 0 @ 0x30000000: Skipped. Contents already match
Received monitor command: reset
ResetTarget() start                              <---- second reset issued  (nullify PC setting)
```

one workaround I found is to use postlaunch commands to write the elf in flash one more time only to have the PC rewrite:

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "request": "launch",
            "type": "cortex-debug",
            "name": "CM4-Test amaLogic Start",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/FW/image/CM4-Test.elf",
            "servertype": "jlink",
            "device": "MIMXRT1176xxxA_M4",
            "interface": "swd",
            "svdFile": "${workspaceRoot}/FW/bsp/device/MIMXRT1176_cm4.svd",
            "jlinkscript": "${workspaceRoot}/evkbmimxrt1170_connect_cm4_cm4side.jlinkscript",
            "serverArgs": [
                "-settingsfile",
                "${workspaceRoot}/JLink_Debug_SettingsFile.jlink",
                "-endian",
                "little",
                "-noir",
                "-vd",
                "-halt",
                "-reportuseraction",
                "-gui"
            ],
            "runToEntryPoint": "main",
            "preLaunchTask": "Build AmaTestboard CM4",
            "preLaunchCommands": [
                "directory ${workspaceRoot}/FW"
            ],
            "postLaunchCommands": [
                 "monitor clrbp",
                 "monitor halt",
                 "monitor regs",
                 "file ${workspaceRoot}/FW/image/CM4-Test.elf",
                 "load",
                 "monitor halt",
                 "tb main",
            ],
        }
    ]
}
```

MCUXpresso IDE has a checkbox for not issuing the second reset:

[image]

Where can I find the cortex-debug equivalent?

haneefdm, 2024-10-12:
Have you considered using `overrideLaunchCommands` instead of `postLaunchCommands` Also, don't duplicate what we do in your customizations or it will become a mess, and, most things in it are not even needed.

I have to close this issue as I can't see what the issue is. It is up to you to customize 

> The problem I'm having is a reset issued after writing the flash.
In fact, writing the flash also sets the PC, the subsequent reset nullifies this setting.

Yes, we have to do that so that after programming, the execution starts at the reset vector and not at some random location in some random memory. Like I said, you can override this...

closed 2024-10-12 by haneefdm

mastupristi, 2024-10-14:
> Have you considered using `overrideLaunchCommands` instead of `postLaunchCommands` Also, don't duplicate what we do in your customizations or it will become a mess, and, most things in it are not even needed.

I will try using `overrideLaunchCommands` then. Where can I find the list of steps that are done by default?

> I have to close this issue as I can't see what the issue is. It is up to you to customize
Yes, no problem at all. I wrote here because I couldn't find a community (mailing list or forum) in which to start a discussion.

> Yes, we have to do that so that after programming, the execution starts at the reset vector and not at some random location in some random memory. 
This is not always the scenario. In any case, writing the flash (I don't know if also the ram) also causes the PC to be written, loading the entry point into it. So the execution would not start at a random location but at the right address,

best regards

haneefdm, 2024-10-14:
See https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood

If you enable debug, every command sent to gdb is shown.