Marus/cortex-debug #1053: why the log file‘ s time timestamp  is messy code?

woquchonglang, 2024-10-17:
**Describe the bug**

the log file‘ s time timestamp  is messy code

**Screenshots**

[image]

[image]

**Environment (please complete the following information):**

OS: arch linux 6.11.3-zen

 - Cortex-Debug (vscode plugin)

 - Compiler Toolchain Version: [ arn-none-eabi 14.2 ]

**Please include `launch.json`**
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "showDevDebugOutput": "none",
      "cwd": "${workspaceRoot}",
      "executable": "./cmake-build-debug/board_c.elf",
      "name": "DebugSTM32",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      "device": "stm32f446",
      "configFiles": [
        // "interface/cmsis-dap.cfg",
        "interface/stlink.cfg",
        "target/stm32f4x.cfg"
      ],
      "armToolchainPath": "/usr/bin",
      "svdFile": "STM32F446x.svd",
      "liveWatch": {
        "enabled": true,
        "samplesPerSecond": 1
      },
      "preLaunchTask": "build",
      "rttConfig": {
        "enabled": true,
        "address": "auto",
        "noclear": true,

        "decoders": [
          {      
            "type": "console",
            "label": "cnt",
            // "encoding": "unsigned",
            // "iencoding": "utf8",
            "logfile": "${workspaceRoot}/data/rtt_log",
            "port":0,
          },
          // {      
          //   "type": "binary",
          //   "label": "cnt_bin",
          //   // "encoding": "unsigned",
          //   // "iencoding": "utf8",
          //   "port": 1,
          // },
          {      
            "type": "graph",
            "label": "cnt",
            // "encoding": "signed",
            "port": 0,
            "graphId": "1",
            "scale": 1.5,
          }
        ]
      },
      // "swoConfig": {
      //   "enabled": true,
      //   "decoders": [
      //     {
      //       "type": "console",
      //       "port": 3,
      //       // "encoding": "unsigned",
      //       // "graphId": "1",
      //       // "scale": 1
      //     },
      //     // {
      //     //   "type": "graph",
      //     //   "port": 0,
      //     //   "encoding": "unsigned",
      //     //   "graphId": "1",
      //     //   "scale": 1
      //     // }
      //   ],
      //   "cpuFrequency": 168000000,
      //   "swoFrequency": 2000000,
      //   "source": "probe"
      // },
      "graphConfig": [
        {
          "label": "Graph 1",
          "timespan": 5,
          "type": "realtime",
          "annotate": true,
          "maximum": 100,
          "minimum": 0,
          "plots": [
            {
              "graphId": "1",
              "label": "cnt",
              "color": "#53753c"
              
            },
            // {
            //   "graphId": "1",
            //   "label": "data 2",
            //   "color": "#955f20"
            // }
          ],
        }
      ]
    }
  ]
}
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Paste Debug Console contents here
```

woquchonglang, 2024-10-17:
BTW, Can I cancel the timestamp?

haneefdm, 2024-10-17:
It is NOT messy. It is the raw output of your firmware. It appears to be writing ANSI codes intended for a console. A terminal (not all) typically takes those ANSI codes to make changes to the screen including cursor movement, font/color changes, etc. You can even write games/animation with it.

So, this is not a bug. It is not our job to turn the logfile into a terminal.

closed 2024-10-17 by haneefdm

woquchonglang, 2024-10-17:
fine, I found the code about LOGI, but what my teminal output is "INFO: xxx", the logfile output is just the "messy code", should  I change the encode or code?
```
#define LOGI(format, ...) \
    LOG_PROTO("INFO: ", RTT_CTRL_TEXT_BRIGHT_GREEN, format, ##__VA_ARGS__)
```

haneefdm, 2024-10-17:
I don't know what you or your team members prefer. You can also output the same to two channels, one intended for the console and one for the logfile and modify your #define to output both.

There are also programs/scripts to strip ANSI codes that you can look at

https://stackoverflow.com/questions/17998978/removing-colors-from-output

woquchonglang, 2024-10-17:
> I don't know what you or your team members prefer. You can also output the same to two channels, one intended for the console and one for the logfile and modify your #define to output both.
>
> There [...]

thank you very much ，what 's else，I only can use the port 0，when I set printf to the buffer 1 ，it doesn't work，and I have changed the rttconfig to port 1（by the stlink to debug）