Marus/cortex-debug #1054: "Remote replied unexpectedly to 'vMustReplyEmpty'" with OpenOCD

marcusb, 2024-10-21:
I am debugging a Marvell mw320 target with OpenOCD. The debugger startup fails with:
```
Failed to launch GDB: Remote replied unexpectedly to 'vMustReplyEmpty': PacketSize=4000;qXfer:memory-map:read-;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+ (from target-select extended-remote localhost:50000)
```

**To Reproduce**

Start a debug session with the target connected.

After starting OpenOCD and connecting to the remote debugger, this error is produced on the debug console, and it does not connect to the remote.

**Expected behavior**

It should connect to remote and allow me to debug the program.

**Environment (please complete the following information):**

 - Cortex-Debug Version: 1.12.1
 - OS: Debian Linux trixie/sid
 - GDB Version: 15.1
 - Compiler Toolchain Version: arm-none-eabi-gcc version 13.2.1 20231009 (15:13.2.rel1-2)
 - OpenOCD 0.12.0 (Debian)

**Please include `launch.json`**
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/build/sesame.axf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "openocd",
            "gdbPath": "/usr/bin/gdb-multiarch",
            "searchDir": [
                "${workspaceFolder}/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/interface",
                "${workspaceFolder}/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD"
            ],
            "configFiles": [
                "${workspaceFolder}/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/interface/ftdi.cfg",
                "${workspaceFolder}/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/openocd.cfg"
            ],
            "openOCDLaunchCommands": [
                "init",
                "load ${workspaceFolder}/build/sesame.axf 0x10013d"
            ],
            "objdumpPath": "/usr/bin/arm-none-eabi-objdump",
            "showDevDebugOutput": "raw",
            
        }
    ]
}
```

**Attach text from `Debug Console`**

```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Cortex Debug",
    "cwd": "/home/marcus/src/FreeRTOS",
    "executable": "/home/marcus/src/FreeRTOS/build/sesame.axf",
    "request": "launch",
    "type": "cortex-debug",
    "runToEntryPoint": "main",
    "servertype": "openocd",
    "gdbPath": "/usr/bin/gdb-multiarch",
    "searchDir": [
        "/home/marcus/src/FreeRTOS/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/interface",
        "/home/marcus/src/FreeRTOS/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD"
    ],
    "configFiles": [
        "/home/marcus/src/FreeRTOS/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/interface/ftdi.cfg",
        "/home/marcus/src/FreeRTOS/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/openocd.cfg"
    ],
    "openOCDLaunchCommands": [
        "init",
        "load /home/marcus/src/FreeRTOS/build/sesame.axf 0x10013d"
    ],
    "objdumpPath": "/usr/bin/arm-none-eabi-objdump",
    "showDevDebugOutput": "raw",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 60000,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "toolchainPrefix": "arm-none-eabi",
    "extensionPath": "/home/marcus/.vscode/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "a81e0bcf-2796-4dd9-b152-fc4c1bac2f10",
    "pvtShowDevDebugOutput": "raw"
}
Reading symbols from /usr/bin/arm-none-eabi-objdump --syms -C -h -w /home/marcus/src/FreeRTOS/build/sesame.axf
Reading symbols from /usr/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/marcus/src/FreeRTOS/build/sesame.axf
Launching GDB: /usr/bin/gdb-multiarch -q --interpreter=mi2
1-gdb-version
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/marcus/src/FreeRTOS/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/interface -s /home/marcus/src/FreeRTOS/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD -f /home/marcus/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f /home/marcus/src/FreeRTOS/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/interface/ftdi.cfg -f /home/marcus/src/FreeRTOS/app/marvell/WMSDK/mw320/sdk/tools/OpenOCD/openocd.cfg -c init -c "load /home/marcus/src/FreeRTOS/build/sesame.axf 0x10013d"
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 90 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (Debian 15.1-1) 15.1\n"
-> ~"Copyright (C) 2024 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"x86_64-linux-gnu\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <"
-> ~"http://www.gnu.org/software/gdb/documentation/>.\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /home/marcus/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /home/marcus/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-file-exec-and-symbols "/home/marcus/src/FreeRTOS/build/sesame.axf"
-> 10^done
Finished reading symbols from nm: Time: 346 ms
11-target-select extended-remote localhost:50000
-> 11^error,msg="Remote replied unexpectedly to 'vMustReplyEmpty': PacketSize=4000;qXfer:memory-map:read-;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+"
12-interpreter-exec console "monitor reset halt"
Failed to launch GDB: Remote replied unexpectedly to 'vMustReplyEmpty': PacketSize=4000;qXfer:memory-map:read-;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+ (from target-select extended-remote localhost:50000)

```

**Additional context**

When starting OpenOCD and gdb outside of VS.Code on the command line, I am able to connect to the target.

haneefdm, 2024-10-21:
What is going on in the terminal tab where you see the gdb-server output appears?

We only sent 11 commands to gdb before failure. You can see how OpenOCD and GDB are launched and you can launch them the same way and send the same gdb commands and see what happens It appears gdb is not happy with what openocd is sending. We don't talk to openocd at all. Maybe a very old/incomplete version of openocd?

```
11-target-select extended-remote localhost:50000
-> 11^error,msg="Remote replied unexpectedly to 'vMustReplyEmpty': PacketSize=4000;qXfer:memory-map:read-;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+"
```

marcusb, 2024-10-22:
OpenOCD is the latest version, and it appears to work when sending the commands manually.

jeronimoagullo, 2024-12-03:
Hi! I am facing exactly the same issue trying to debug an esp32s3 board with a Zephyr app (I cam debug it directly using `west debug` command which calls the same openocd executable).

My launch.json file:

```json
{
            "name": "ESP32-S3",
            "type": "cortex-debug",
            "request": "attach",
            "servertype": "openocd",
            "interface": "jtag",
            "armToolchainPath": "C:/Workspace/zephyrSDK/zephyr-sdk-0.17.0/xtensa-espressif_esp32s3_zephyr-elf/bin",
            "toolchainPrefix": "xtensa-espressif_esp32s3_zephyr-elf",    
            "gdbPath": "C:/Workspace/zephyrSDK/zephyr-sdk-0.17.0/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gdb.exe",
            "runToEntryPoint": "main",
            "cwd": "C:/Workspace/zephyrproject/esp32s3_ws/display",
            "executable": "build/zephyr/zephyr.elf",
            "configFiles": [
                "C:/Workspace/zephyrproject/zephyr_jero/boards/espressif/esp32s3_devkitm/support/openocd.cfg"  // Replace with your OpenOCD config
            ],
        }
```

haneefdm, 2024-12-03:
@jeronimoagullo How about output from both cortex-debug and the `west debug`.. Enable debug/verbose where appropriate.

* I would like to see how openocd was launched. (command-line, confilg files etc.)
* I want to see every command sent to gdb and its responses if available.
* 
People saying it works from the command line and not providing the input/output is not helpful

jeronimoagullo, 2024-12-04:
@haneefdm you are totally rigth. I attach both logs:

With `west -v debug` using Zephyr:

```
-- west debug: rebuilding
cmake version 3.21.1 is OK; minimum version is 3.13.1
Running CMake: 'C:\Program Files (x86)\CMake\bin\cmake.EXE' --build 'C:\Workspace\zephyrproject\esp32s3_ws\display\build'
ninja: no work to do.
-- west debug: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
runners.openocd: openocd -s 'C:/Workspace/zephyrproject/zephyr_jero/boards/espressif/esp32s3_devkitm\support' -f 'C:/Workspace/zephyrproject/zephyr_jero/boards/espressif/esp32s3_devkitm\support\openocd.cfg' -c 'tcl_port 6333' -c 'telnet_port 4444' -c 'gdb_port 3333'
runners.openocd: C:/Workspace/zephyrSDK/zephyr-sdk-0.17.0/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gdb-py.exe -ex 'target extended-remote :3333' C:/Workspace/zephyrproject/esp32s3_ws/display/build/zephyr/zephyr.elf -ex 'set remote hardware-watchpoint-limit 2' -ex 'maintenance flush register-cache' -ex 'mon reset halt' -ex 'thb main'
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
DEPRECATED! use 'tcl port' not 'tcl_port'
DEPRECATED! use 'gdb port', not 'gdb_port'
Info : Listening on port 6333 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (F4:12:FA:43:6E:48)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.tap0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.tap1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Examination succeed
Info : [esp32s3.cpu1] Examination succeed
Info : [esp32s3.cpu0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40375F0C, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32s3.cpu1] Target halted, PC=0x40043A40, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (1) - (Power on reset)
GNU gdb (Zephyr SDK 0.17.0) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=xtensa-espressif_esp32s3_zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/zephyrproject-rtos/sdk-ng/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from C:/Workspace/zephyrproject/esp32s3_ws/display/build/zephyr/zephyr.elf...
Info : accepting 'gdb' connection on tcp/3333
Remote debugging using :3333
Info : [esp32s3.cpu0] Target halted, PC=0x40375F0C, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x40043A40, debug_reason=00000000
Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, debug_reason=00000001
Warn : Failed to get flash maps (-6)!
Warn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.
Warn : Failed to get flash mappings (-4)!
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, debug_reason=00000001
Warn : Failed to get flash maps (-6)!
[... 137 lines cut ...]
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "serverpath": "C:/idf-tools/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin/openocd.exe",
    "searchDir": [],
    "toolchainPath": "C:/Workspace/zephyrSDK/zephyr-sdk-0.17.0/xtensa-espressif_esp32s3_zephyr-elf/bin",
    "extensionPath": "c:/.vscode/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": false,
    "pvtVersion": "1.12.1",
    "__sessionId": "c9039f3f-ac12-428a-aed8-c9e7a776a270",
    "pvtShowDevDebugOutput": "raw"
}
```

jeronimoagullo, 2024-12-04:
Besides, I can run the debug session directly by terminal launching the GDB server with:

```
.\xtensa-espressif_esp32s3_zephyr-elf-gdb.exe -ex 'file C:/Workspace/zephyrproject/esp32s3_ws/display/build/zephyr/zephyr.elf' -ex 'target extended-remote
:3333'
```

and openocd with:

```
openocd -s 'C:/Workspace/zephyrproject/zephyr_jero/boards/espressif/esp32s3_devkitm/support' -f 'C:/Workspace/zephyrproject/zephyr_jero/boards/espressif/esp32s3_devkitm/support/openocd.cfg' -c 'tcl_port 6333' -c
 'telnet_port 4444' -c 'gdb_port 3333'
```

haneefdm, 2024-12-04:
I wish you listed a few more lines from cortex-debug console where I can see the various programs being launched.

I spot several differences already.

**Versions of openocd are different** I don't know how much of a difference that makes. The timestamp can be meaningless and versions don't mean much for device support.
```
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
vs
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
```

No search path specified for cortex-debug. This is done via `searchDirs` in launch.json
```
-s 'C:/Workspace/zephyrproject/zephyr_jero/boards/espressif/esp32s3_devkitm\support'
```
I would have expected (unless it is the default/redundant).
```
    "searchDirs": [
        'C:/Workspace/zephyrproject/zephyr_jero/boards/espressif/esp32s3_devkitm\support"
    ]
```
Openocd contains all the magic for your device. Not gdb and not us.

You may also be using different versions of gdb. I can't tell

jeronimoagullo, 2024-12-04:
@haneefdm you gave me the key to get it working, my `cortex-debug.openocdPath` setting was incorrected (maybe from a past project) and it was using a different version of openocd. Now, it is using the same openocd executable and working smoothly. Thanks!

haneefdm, 2024-12-04:
I am closing this because the original PR did not give enough details, so nothing to do.

closed 2024-12-04 by haneefdm