Marus/cortex-debug #1055: Debugger trying to debug non-existant example.elf which is not in project

Charry2014, 2024-10-22:
**Describe the bug**
This is more a request for help and explanation than reporting a bug - The debugger is always trying to debug a file called example.elf which does not exist and is not mentioned in any makefile I can find - where is this name coming from? The project does not build this file.

On starting the debugger VS Code presents a dialog -
Failed to launch GDB: /Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf: No such file or directory. (from interpreter-exec console "load \"/Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf\"")

This is building a FreeRTOS project for the Pi 2040. The project builds two .elf files - on_core_zero.elf and on_core_one.elf. If one of these is renamed to example.elf then debugging proceeds normally.

**To Reproduce**
Installed plugins are the standard Raspberry Pi plugin for VS Code, and of course Cortex Debug. The required tools (Pico SDK etc.) are all installed.
OpenOCD is installed using xPack to work around [this issue](https://github.com/raspberrypi/pico-vscode/issues/102).
To set up the project I cloned the FreeRTOS Kernel and Community Examples into the same folder, and then selected (at random) the FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore folder to make the project. 
Having added this folder to the VS Code workspace the project was created and after setting some environment variables it builds fine.

setup.sh
```
#!/bin/bash

if [ ! -d ./build ]; then
    mkdir ./build
fi
cd ./build
export PICO_SDK_PATH="$HOME"/.pico-sdk
export FREERTOS_KERNEL_PATH="$HOME"/work/projects/vscfreetos/FreeRTOS-Kernel
export PICO_TOOLCHAIN_PATH="$HOME"/.pico-sdk/toolchain/13_2_Rel1/bin

cmake -G Ninja ..
```

**Expected behavior**

The debugger should debug the file mentioned in the executable section of launch.json
``` "executable": "${workspaceRoot}/build/on_core_zero.elf",```

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension) 1.12.1
 - OS: macOS Sonoma 14.6.1
 - GDB Version: 15.2
 - OpenOCD version: xPack Open On-Chip Debugger 0.12.0+dev-01685-gb9224c0c0-dirty (2024-08-02-22:53)
 - Compiler Toolchain Version: [e.g. arn-none-eabi V 11.1]

Launch.json:
```
        {
            "name": "Pico Debug (Cortex-Debug)",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/on_core_zero.elf",
            "request": "launch",
            "showDevDebugOutput": "raw",
            "type": "cortex-debug",
            "servertype": "openocd",
            "gdbPath": "${command:raspberry-pi-pico.getGDBPath}",
            "device": "${command:raspberry-pi-pico.getChipUppercase}",
            "configFiles": [
                "interface/cmsis-dap.cfg",
                "target/${command:raspberry-pi-pico.getTarget}.cfg"
            ],
            "svdFile": "${userHome}/.pico-sdk/sdk/2.0.0/src/${command:raspberry-pi-pico.getChip}/hardware_regs/${command:raspberry-pi-pico.getChipUppercase}.svd",
            "runToEntryPoint": "main",
            // Fix for no_flash binaries, where monitor reset halt doesn't do what is expected
            // Also works fine for flash binaries
            "overrideLaunchCommands": [
                "monitor reset init",
                "load \"${command:raspberry-pi-pico.launchTargetPath}\""
            ],
            "openOCDLaunchCommands": [
                "adapter speed 5000"
            ]
        },
[... 120 lines cut ...]
-> ~"6065\t        while( uxDeletedTasksWaitingCleanUp > ( UBaseType_t ) 0U )\n"
6065	        while( uxDeletedTasksWaitingCleanUp > ( UBaseType_t ) 0U )
-> *stopped,frame={addr="0x10001bf8",func="prvCheckTasksWaitingTermination",args=[],file="/Users/toby/work/projects/vscfreetos/FreeRTOS-Kernel/tasks.c",fullname="/Users/toby/work/projects/vscfreetos/FreeRTOS-Kernel/tasks.c",line="6065",arch="armv6s-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 11^connected
12-interpreter-exec console "monitor reset init"
-> @"[rp2040.core0] halted due to breakpoint, current mode: Thread \n"
[rp2040.core0] halted due to breakpoint, current mode: Thread
-> @"xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00\n"
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
-> @"[rp2040.core1] halted due to debug-request, current mode: Thread \n"
[rp2040.core1] halted due to debug-request, current mode: Thread
-> @"xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00\n"
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
-> 12^done
13-interpreter-exec console "load \"/Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf\""
-> &"/Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf: No such file or directory.\n"
/Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf: No such file or directory.
-> 13^error,msg="/Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf: No such file or directory."
Failed to launch GDB: /Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf: No such file or directory. (from interpreter-exec console "load \"/Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf\"")
```

**Additional context**
This is my first step into debugging Pi FreeRTOS projects in C so any help gratefully received.

haneefdm, 2024-10-22:
You are the one forcing the load of example.elf from your launch.json
```JSON
            "overrideLaunchCommands": [
                "monitor reset init",
                "load \"${command:raspberry-pi-pico.launchTargetPath}\""
            ],
```

In the debug console, you will see the above into]]
```JSON
    "overrideLaunchCommands": [
        "monitor reset init",
        "load \"/Users/toby/work/projects/vscfreetos/FreeRTOS-Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/build/example.elf\""
    ],
```

Please contact whoever gave you this template. When you override things, you are on your own and have to understand what is going on.

closed 2024-10-22 by haneefdm

Charry2014, 2024-10-22:
Thank you, thank you. That is the magic - kicking myself for missing that, but many thanks for your help.

The template doing this most likely is the standard Raspberry Pi extension for VS Code, which I do not understand - but I am learning. Thank you again for your help.