Marus/cortex-debug #1056: Core peripherals view missing ?

eMKa94, 2024-10-29:
As far as i remember, some time ago i was able to view the registers related to the core (CM4) like NVIC, Systick, etc...

No i can only see the MCU peripherals. 

I was trying two methods of providing a SVD info: directly from SVD file downloaded from vendor and by giving the cmsisPACK reference. Both with the same result: MCU peripherals and core registers available, but not core peripherals... 

How to see them ?

eMKa94, 2024-10-30:
Not exactly...

The core registers are available but besides the registers  core related peripherals with their own register set should be visible too.

Like in the STM32 Cube IDE in the SFR view:
[image]
 
Some time ago if i remember correctly they were available even if i didn't specify the SVD file for the debug session so i'm assuming that the problem is not that they're missing in the SVD...

haneefdm, 2024-10-30:
If the SVD describes those peripherals, then they will be displayed. Does ST's svd file have those peripherals?

And, no, we never automatically showed things like NVIC/MPU/etc. And, I don't think we will ever do that. There are too many versions of these peripherals, we don't do discovery probes of the ROMTABLE. We are now also supporting non ARM architectures.

closed 2024-10-30 by haneefdm

eMKa94, 2024-10-30:
OK I understood.
Thanks for the reply

PhilippHaefele, 2024-10-30:
BTW I found the project with the patched SVD files: https://stm32-rs.github.io/stm32-rs/

You can download the patched SVD with the click on the corresponding `(svd)` next to the device (family) name.