Marus/cortex-debug #1060: Allow string for cpuFrequency and swoFrequency

mm-mm-bb, 2024-11-12:
I need to support different chip with clock speed.
The setup is:
- in the CMakePresets.json, each preset define a cacheVariables with the desired clock speed
- cmake will save that variables inside of the .vscode/settings.json
- in the launch.json i assign the variable to the "swoConfig.cpuFrequency"

Unfortunately, while this work for stuff like the swoPort, on the cpuFrequency field i get error "Expected "number"."

**Describe the solution you'd like**
Allow cpuFrequency  and swoFrequency (but really any numeric field) to be a string.
Something similar was implemented here: https://github.com/microsoft/vscode-js-debug/pull/1880/files

**Describe alternatives you've considered**
CMake would have to edit the launch.json and not only the setting.json.
(launch.json using cacheVariables  directly would NOT fix, as seems ${} must be in quotes to work)

haneefdm, 2024-11-12:
Show me what your launch.json line for `swoConfig.cpuFrequency` looks like now and after the change. I am not understanding

mm-mm-bb, 2025-04-07:
I am thinking about this hack:
 - create a little script that can edit the settings.json with values from parameters
 - create a target that run the script with proper values
 - add that target as build dependency

hopefully the script run early enough for the plugin to use the new setting?  
If is guarantee (maybe by a config) that the plugin will always reload the setting.json before using it, should be a decent way to share data from the CMake/build system to the plugin.

What do you think about this trick? Maybe a dedicated file instead of using the settings.json would be cleaner, but seems a decent hack.