Marus/cortex-debug #1063: gdb starts without executable file, and does not load it or symbols

ponyatov, 2024-11-21:
```json
            "name": "STM32F407G-DISC1",
            "type": "cortex-debug",
            "servertype": "openocd",
            "request": "launch",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceFolder}/tmp/STM32F407G-DISC1/${workspaceFolderBasename}.elf",
            "toolchainPrefix": "arm-none-eabi",
            "gdbPath": "gdb-multiarch",
            "overrideLaunchCommands": [
                "target remote :12345",
                "file ${workspaceFolder}/tmp/STM32F407G-DISC1/${workspaceFolderBasename}.elf",
```

ponyatov, 2024-11-21:
partially fixed with native debug plugin: 

```json
        // "marus25.cortex-debug",     // overengineered, does not load .elf on debug start
        "webfreak.debug",              // ok
```

`cortex-debug` have no config parameters (?) to fully tune command line for gdb start, such as:
```sh
${gdbpath} -q -x ${workspaceFolder}/.gdbinit ${executable}
```
so I just moved to classical native debug:
```json
        {
            "name": "STM32F407G-DISC1",
            "type": "gdb",
            "request": "attach",
            "executable": "${workspaceFolder}/tmp/STM32F407G-DISC1/${workspaceFolderBasename}.elf",
            "target": ":12345",
            "remote": true,
            "gdbpath": "/usr/bin/gdb-multiarch",
            "cwd": "${workspaceRoot}",
            "autorun": [
                "b Reset_Handler",
                "b main",
                "monitor reset halt"
            ]
        },
```
, run `openocd -f ${workspaceFolder}/openocd.cfg` in a separate task, and now just have no any problems

haneefdm, 2024-11-22:
not sure you read our docs/wiki or looked at examples on web. Since you have another solution and can't see any issue here, i am closing this

closed 2024-11-22 by haneefdm