Marus/cortex-debug #1064: STM32F042F6P6 Cant program or debug

D4NNYH, 2024-11-28:
**Describe the bug**
Upon starting debug session everything appears to be working however program has not uploaded and is not running, deugger appears to allow pausing the and resuming code but actually is not doing anything.
Uploading the same .elf file using stm32cubeprogrammer (gui version) works and code runs.

**Expected behavior**

Program uploads, debug session begins

**Screenshots**

[comment]: <> If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (this extension): v1.12.1
 - OS: Windows 11
 - GDB Version: 7.9.0
 - Compiler Toolchain Version: not sure (whatever comes with stm32cubeCLT 1.17.0)

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
Paste launch.json contents here
```

**Attach text from `Debug Console`**

Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Build & Debug Microcontroller - ST-Link",
    "cwd": "C:\\Users\\user\\Documents\\VSCode\\STM\\UrbanWatt\\System_1--RxF042F6P6",
    "type": "cortex-debug",
    "executable": "C:/Users/user/Documents/VSCode/STM/UrbanWatt/System_1--RxF042F6P6/build/Debug/System_1--RxF042F6P6.elf",
    "request": "launch",
    "servertype": "stlink",
    "device": "STM32F042F6Px",
    "interface": "swd",
    "serialNumber": "",
    "runToEntryPoint": "main",
    "svdFile": "c:\\ST\\STM32CubeCLT_1.17.0/STMicroelectronics_CMSIS_SVD/STM32F0x2.svd",
    "v1": false,
    "serverpath": "c:\\ST\\STM32CubeCLT_1.17.0/STLink-gdb-server/bin/ST-LINK_gdbserver",
    "stm32cubeprogrammer": "c:\\ST\\STM32CubeCLT_1.17.0/STM32CubeProgrammer/bin",
    "stlinkPath": "c:\\ST\\STM32CubeCLT_1.17.0/STLink-gdb-server/bin/ST-LINK_gdbserver",
    "armToolchainPath": "c:\\ST\\STM32CubeCLT_1.17.0/GNU-tools-for-STM32/bin",
    "gdbPath": "c:\\ST\\STM32CubeCLT_1.17.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb",
    "serverArgs": [
        "-m",
        "0"
    ],
    "showDevDebugOutput": "raw",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
[... 108 lines cut ...]
-> *running,thread-id="all"
mi2.status = running
19-exec-interrupt
-> 19^done
-> ~"\nProgram"

Program
-> ~" received signal SIGINT, Interrupt.\n"
 received signal SIGINT, Interrupt.
-> ~"0x1fffc8bc in ?? ()\n"
0x1fffc8bc in ?? ()
-> *stopped,reason="signal-received",signal-name="SIGINT",signal-meaning="Interrupt",frame={addr="0x1fffc8bc",func="??",args=[],arch="armv6s-m"},thread-id="1",stopped-threads="all",core="0"
mi2.status = stopped
20-break-delete
-> 20^done
21-target-disconnect
-> =thread-exited,id="1",group-id="i1"
-> =thread-group-exited,id="i1"
-> 21^done
-gdb-exit
-> ^exit
GDB session ended. exit-code: 0

**Additional context**
Add any other context about the problem here.

haneefdm, 2024-11-29:
I don't know what you mean by program not loading. How did you determine that?

```
10-file-exec-and-symbols "C:/Users/user/Documents/VSCode/STM/UrbanWatt/System_1--RxF042F6P6/build/Debug/System_1--RxF042F6P6.elf"
-> 10^done
```

The above says we told gdb what program is being debugged.
```
13-target-download
-> 13+download,{section=".isr_vector",section-size="192",total-size="810864"}
-> 13+download,{section=".isr_vector",section-sent="192",section-size="192",total-sent="192",total-size="810864"}
-> 13+download,{section=".text",section-size="10988",total-size="810864"}
-> 13+download,{section=".rodata",section-size="112",total-size="810864"}
-> 13+download,{section=".init_array",section-size="4",total-size="810864"}
-> 13+download,{section=".fini_array",section-size="4",total-size="810864"}
-> 13+download,{section=".data",section-size="104",total-size="810864"}
-> 13^done,address="0x080027e4",load-size="11404",transfer-rate="43256",write-rate="1900"
14-interpreter-exec console "monitor reset"
-> @"STM32 Successfully completed reset operation (System reset)\n"
STM32 Successfully completed reset operation (System reset)
-> 14^done
```
We asked gdb to download the current program and it seems to have done it properly. This is normal for 99% of the usecases. Those that have multiple cores and or security images, will require customization.

I don't know how you configured the ST tools. ST also has similar debug output, btw

You can look at every line of the debug output and see what is on/wrong,

D4NNYH, 2024-11-29:
Sorry for any confusion, essentially the program only runs (LEDs flash etc) if i manually flash the code using stm32cubeprogrammer. When doing this via cortex debugger it says this has happened however nothing is functioning, i attempt to pause the operations and add breakpoints but it appears to have no effect (other than showing the pause button is pressed and the program is "paused"). It works fine on other microcontrollers i use, however it seems to have problems only with the f042f6p6, again the chip functions fine when flashing without using the debugger

haneefdm, 2024-11-30:
You can customize loading your FW using VSCode facilities launch.json, tasks.json (as a predebug action), and there are a ton of ways you can also configure cortex-debug as well

closed 2024-11-30 by haneefdm

D4NNYH, 2024-11-30:
I dont suppose youd be able to point me to what in my launch.json is causing the debugger to think it is functioning fine when nothing is actually happening please? (sorry i thought id attached this before but had missed it out)