Marus/cortex-debug #1065: rttConfig.enable does nothing (arm gdb + blackmagicprobe on stlink). here is the proof + workaround

albmef, 2024-12-01:
hi !
below is the proof that enabling rtt with just rttconfig.enable just does not work (on my config)

setup: 
-VSCODE on archlinux, with cortex debug extension
- board st nucleo-f441re with blackmagicprobe (target = cortexm, stm and rtt enabled)

prior each test i manually disable the rtt by either doing a cold start, or typing monitor rtt disable in a gdb CLI

first test: 
launch.json

```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "BMP Launch RTT",
    "type": "cortex-debug",
    "request": "launch",
    "executable": "/home/alban/zephyr_workspace/blinky_f411re_BMP/build/zephyr/zephyr.elf",
    "cwd": "/home/alban/zephyr_workspace/blinky_f411re_BMP",
    "servertype": "bmp",
    "BMPGDBSerialPort": "/dev/ttyBmpGdb",
    "interface": "swd",
    "runToEntryPoint": "main",
    "breakAfterReset": true,
    "showDevDebugOutput": "raw",
    "svdPath": "/home/alban/zephyr_workspace/blinky_f411re_BMP/boards/arm/nucleo_f411re_bmp/support/STM32F411.svd",
    "rttConfig": {
        "enabled": true,
        "decoders": []
    },
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "powerOverBMP": "lastState",
    "targetId": 1,
    "toolchainPath": "/home/alban/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/",
    "toolchainPrefix": "arm-zephyr-eabi",
    "extensionPath": "/home/alban/.vscode-oss/extensions/marus25.cortex-debug-1.12.1-universal",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "59008cd4-aa6a-4bcc-bbd6-345e87a6d881",
    "pvtShowDevDebugOutput": "raw"
}
Reading symbols from /home/alban/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump --syms -C -h -w /home/alban/zephyr_workspace/blinky_f411re_BMP/build/zephyr/zephyr.elf
Reading symbols from /home/alban/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-nm --defined-only -S -l -C -p /home/alban/zephyr_workspace/blinky_f411re_BMP/build/zephyr/zephyr.elf
Launching GDB: /home/alban/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb -q --interpreter=mi2
1-gdb-version
Finished reading symbols from objdump: Time: 49 ms
[... 426 lines cut ...]
8:LED state: ON
[00:00:07.002,000] <dbg> main: main: 8:LED state: ON
9:LED state: OFF
[00:00:08.002,000] <dbg> main: main: 9:LED state: OFF
10:LED state: ON
[00:00:09.002,000] <dbg> main: main: 10:LED state: ON
11:LED state: OFF
[00:00:10.003,000] <dbg> main: main: 11:LED state: OFF
12:LED state: ON
[00:00:11.003,000] <dbg> main: main: 12:LED state: ON
13:LED state: OFF
[00:00:12.003,000] <dbg> main: mai*** Booting Zephyr OS build v4.0.0-514-g684324019635 ***
*** Booting Zephyr OS build v4.0.0-514-g684324019635 ***
1:LED state: OFF
[00:00:00.000,000] <dbg> main: main: 1:LED state: OFF
2:LED state: ON
[00:00:01.000,000] <dbg> main: main: 2:LED state: ON
3:LED state: OFF
[00:00:02.000,000] <dbg> main: main: 3:LED state: OFF
```

TADAAAWWWWW !

feel free to ask if you need more info i'd be happy to help
AM

haneefdm, 2025-01-16:
We don't have BMP hardware. We only enabled what we could test. Questions/comments

* All you had to do was to use `"monitor rtt enable"`. OpenOCD and JLink require much more setup than that.
* Where does the output show up? We normally direct RTT output to the TERMINAL tab just like SWO
* BMP uses serial ports for output so not sure where/what serial port you used. Did you just use something like PuTTY to capture the serial output?
* How does BMP know where the RTT buffer is? And, what to look for? Searching the entire memory space may not be a good idea.

Usage or RTT implies that BMP implemented what SEGGER does.

> prior each test i manually disable the rtt by either doing a cold start, or typing monitor rtt disable in a gdb CLI

What happens if you don't disable it when the debug session ends?

CC: @ssimek Do you know about this? RTT is far more performant than SWO can ever be. Both require manual instrumentation and flow control. Since SEGGER put it in the public domain, more gdb servers are implementing it.

ssimek, 2025-01-16:
@albmef true, rttConfig is currently not handled on BMP but it easily could, I'll look into it.

@haneefdm the simple config is what BMP is about :) And yes, BMP has RTT support, [just not in the standard builds for some reason](https://black-magic.org/usage/rtt.html). I have never played with it until now, I just remember looking at it some time ago but I was quite happy with SWO until now. SEGGER's [performance claim](https://kb.segger.com/RTT) is marketing BS to a degree, they are comparing filling a memory buffer that will be later read by the debug probe with synchronously waiting on a UART-like device. Running SWO at reasonable speeds (several MHz) removes most of the waiting. They also need a pretty beefy MCU to make the claim that they were outputting 82 characters in a microsecond - an average ~80 MHz one would hardly be able to output more than one character in one CPU clock cycle 🙄

haneefdm, 2025-01-16:
> SEGGER's [performance claim](https://kb.segger.com/RTT) is marketing BS to a degree, they are comparing filling a memory buffer that will be later read by the debug probe with synchronously waiting on a UART-like device.

Yes, I know and I have tried it in a real application. While it is faster than SWO, definitely not 100x. The 100x is mostly FW performance (cycles) for writing to SRAM vs SWO writing to registers with a tiny FIFO. Not the gdb-server reading the RTT buffer. I was limited by the probe USB performance though. And, our board TX-UART (SWO) is unreliable even at 1 Mbit. It doesn't matter how fast the processor could run. Our IOs were limited to 33 MHz toggle rate as well.

In my application I was not just banging on the buffer in a loop. I had things to sense and calculate and then shove semi-large quantities of bytes into a buffer for the host to see. We also could not afford a pin to dedicate to SWO and using the SWD pins was more convenient.

Looking at https://black-magic.org/usage/rtt.html they do have the other config items for RTT. Most important of those is this

```
monitor rtt ram startaddress endaddress
```

There is also a question of timing. Since the RTT control block is initialized in FW sometime after main, looking for it will fail if done at reset. Or you may pickup stale/wrong information from a previous run. Especially if the linker moved the control block.

Btw, some of the new crop of chips may not even have SWO. I was already part of developing one chip with no SWO because it was a multi-core setup and had to go with ARMs recommendation.

ssimek, 2025-01-17:
Yep, I'm definitely intrigued in RTT and will work on adding proper support with the BMP. Regarding the config - I presume they are doing an autoscan if the addresses are not specified, I guess JLink is doing the same from what I understood.

There definitely are cores without SWO, for example M0/M0+, and I completely understand the need for an extra pin can be an issue sometimes. Another thing that comes to mind is the USB-C SBU1/2 connection to SWCLK/SWDIO would be a lot more useful with RTT.

haneefdm, 2025-01-17:
>  Regarding the config - I presume they are doing an autoscan if the addresses are not specified, I guess JLink is doing the same from what I understood.

Yes, but it takes quite a bit of SWD bandwidth to scan the entire SRAM. One scan may not work. They could also find the wrong control block. With other gdb-servers, we typically know the exact address and we pass that along. It is almost free.

Just assigned this to you.

PaulFirs, 2025-05-10:
Are there any updates on this issue? It seems to be a pretty relevant question.

ssimek, 2025-05-12:
@PaulFirs sorry, I haven't really been able to spend time working on this for various reasons, and I don't expect that to change in the near future. It should be quite easy to add, though, if you feel like it.