Marus/cortex-debug #1071: Support for STM32CubeCLT

srberard, 2025-01-13:
**Is your feature request related to a problem? Please describe.**
I'm working with the STM32CubeCLT tools from ST Micro.  STM32CubeCLT is a toolset for third-party integrated development environment (IDE) providers, allowing the use of STMicroelectronics proprietary tools within their own IDE frameworks.

I am using these instead of the STM32CubeIDE as I'm using this under WSL2 with VS Code.  I've had issues with the full IDE under WSL and found [these](https://github.com/robinhellmers/developer_setup_stm32_wsl) instructions on how to setup the CLT tools with WSL.

All of this works, however, I have to manually set the configuration values as the paths are not automatically found.  While this works, it needs to be updated every time the tool version changes.  Cortex Debug is able to automatically find the paths for the tools under the full IDE.

**Describe the solution you'd like**
I'd like Cortex Debug to automatically detect the path when using the CLT.  It is my understanding that the CLT is the recommended toolset when using 3rd party IDEs such as VS Code.

**Describe alternatives you've considered**
Manual configuration of Cortex Debug parameters.  In my case, I've installed STM32CubeCLT version 1.17.0.  This requires me to set the following parameter:
```
"cortex-debug.stm32cubeprogrammer.linux": "/opt/st/stm32cubeclt_1.17.0/STM32CubeProgrammer/bin",
```
I believe Cortex Debug assumes a base path of `STM32CubeIDE_` and the CLT uses a different base install path.

**Additional context**
Information on the STM32CubeCLT tools can be found [here](https://www.st.com/en/development-tools/stm32cubeclt.html).

haneefdm, 2025-01-13:
Thanks for letting me know about STM32CubeCLT. I will look into it.

Q? What if the user has both the IDE and CLT installed? FYI: we don't detect if STM32CubeIDE is being used. No way to do that. We just look for an STM32CubeIDE install in its default location.

I wish STM32CubeIDE itself used the CLT and that way, there will be one version of the truth.

Btw, I just installed the CLT on my Mac and I couldn't tell where it was installed. Had to go hunting for it

Andreychik32, 2025-01-13:
> I couldn't tell where it was installed

FYI: it gets installed under `/opt/ST` path.

haneefdm, 2025-01-13:
Yes, I now know. I found the path in the uninstall documentation. Not where I was expecting.

srberard, 2025-01-13:
Regarding your question on the IDE and CLT?  My suggestion would be to see if the IDE stuff is there and, if so, use that.  Otherwise continue and look for the CLT.  

My assumption is that one would use or the other, not both.  But I might be mistaken.

schiltztool, 2025-01-21:
Hopping on this to bump searching for CLT first. We use it both on developers' machines and our build server.

It's installed to `C:\ST\STM32CubeCLT` on windows fyi

haneefdm, 2025-01-21:
It is already merged and undergoing testing.

https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md#v1130-pre4

stay tuned for a pre-release