Marus/cortex-debug #1075: Path expansion seems broken in c1.13.0-pre1 for executable path

dancollins, 2025-01-16:
**Describe the bug**
Moving from v1.12.1 to v1.13.0-pre1, the extension no longer "works". Without touching launch.json between versions, it now fails to launch gdb - and the cause seems to be path expansion. My suspicion is this is a Windows-only failure (due to the way paths work in Windows).

**To Reproduce**
Steps to reproduce the behavior:
1. Have a working setup with version 1.12.1, including the `${workspaceFolder}` variable as part of the *executable* path
2. Try to launch a session - GDB may fail to start, saying the file could not be found.
3. Observe the file exists, but the path used isn't quite right.

**Expected behavior**

Ignoring any other pre-release issues, this should launch GDB with the specified executable.

**Screenshots**

[image]

**Environment (please complete the following information):**

 - Cortex-Debug Version 1.13.0-pre1
 - OS: Windows 11
 - GDB Version: GNU gdb (Zephyr SDK 0.16.5-1) 12.1
 - Compiler Toolchain Version: arm-zephyr-eabi-gcc.exe (Zephyr SDK 0.16.5-1) 12.2.0

**launch.json**

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "PRV Station 9151",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder:build}/prv_station/9151/asset_tracker_v2/zephyr/zephyr.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "nRF9151_xxCA",
            "rtos": "Zephyr",
            "loadFiles": [
                "${workspaceFolder:build}/prv_station/9151/merged.hex",
            ],
            "symbolFiles": [
                "${workspaceFolder:build}/prv_station/9151/mcuboot/zephyr/zephyr.elf",
                "${workspaceFolder:build}/prv_station/9151/asset_tracker_v2/zephyr/zephyr.elf"
            ],
            "showDevDebugOutput": "raw"
        }
    ]
}
```

**Debug Console**

See attached.

dancollins, 2025-01-16:
[cortex-debug-1.13.0-pre1-logs.txt](https://github.com/user-attachments/files/18431782/cortex-debug-1.13.0-pre1-logs.txt)

haneefdm, 2025-01-16:
We don't expand the paths. VSCode does. However, gdb does not handle Windows paths so we have to convert what gdb sees into forward slashed path names. Missed massaging the executable when we changed how the executable is given to gdb.

I will have an updated build that works on windows tomorrow.

haneefdm, 2025-01-16:
Can you try out the following latest release?

https://github.com/Marus/cortex-debug/releases

dancollins, 2025-01-16:
Fixed! Thanks :)

haneefdm, 2025-01-16:
Thanks for checking it out.

Sorry to say but more changes are coming that can affect how programs and symbols are loaded because of #1007 There are far too many customizations to keep everything straight.

dancollins, 2025-01-20:
Another example is when using `add-symbol-file` commands - the path is expanded but not escaped correctly:

```
            "preLaunchCommands": [
                "add-symbol-file ${workspaceFolder:build}/prv_station/9151/mcuboot/zephyr/zephyr.elf",
            ]```

... which results in ...

```Program stopped, probably due to a reset and/or halt issued by debugger
add symbol table from file "C:puild/prv_station/9151/mcuboot/zephyr/zephyr.elf"
(y or n) [answered Y; input not from terminal]
C:puild/prv_station/9151/mcuboot/zephyr/zephyr.elf: No such file or directory.
Failed to launch GDB: C:puild/prv_station/9151/mcuboot/zephyr/zephyr.elf: No such file or directory. (from interpreter-exec console "add-symbol-file C:\p\build/prv_station/9151/mcuboot/zephyr/zephyr.elf")```

I was previously using `symbolFiles` but I found this was then not providing the executable to GDB which caused other issues. If you want, I can raise a bug to elaborate - but the `preLaunchCommands` seems to do what I need for now.

haneefdm, 2025-01-20:
We don't edit/touch the pre/post/override commands. This is because it can contain ANYTHING and it could be meant for gdb or for openocd/jlink/pyocd/stlink/etc. You have to use the appropriate number of backslashes and quotes to make it through VSCode and what ever the destination is.

```
    "preLaunchCommands": [
        "add-symbol-file \\\"${workspaceFolder:build}/prv_station/9151/mcuboot/zephyr/zephyr.elf\\\"",
    ]
```

Also, if you use add symbols (instead of `symbolFiles`) this way, we don't know those symbols and we will not list them in the globals/statis/disassembly. This is okay for some people

closed 2025-01-22 by haneefdm