Marus/cortex-debug #1082: "Failed to launch OpenOCD GDB Server: Timeout." Nuvoton Nu Link Debugger

ichbinbork, 2025-01-19:
**Describe the bug**
I made all the setup for Cortex Debug to debug the project. However, I got the "Failed to launch OpenOCD GDB Server: Timeout." error. 

Here is the command that Cortex Debug runs. I run the same command from the terminal and I reprogram MCU without any problem.
```
"C:/Program Files (x86)/Nuvoton Tools/OpenOCD/bin/openocd.exe" -c "gdb_port 50003" -c "tcl_port 50004" -c "telnet_port 50005" -s "C:\\Users\\megab\\Desktop\\Nuvoton\\NuvotonWorkspace\\MicroBMS" -f "c:/Users/megab/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl" -f "C:/Program Files (x86)/Nuvoton Tools/OpenOCD/scripts/interface/nulink.cfg" -f "C:/Program Files (x86)/Nuvoton Tools/OpenOCD/scripts/target/numicroM0.cfg"
```

**Screenshots**

[image]

 - Cortex-Debug Version : 1.12.1
 - OS: Windows11
 - GDB Version: [e.g. 1.11.1]
 - Compiler Toolchain Version: GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10)

**Please include `launch.json`**

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        
        {
            
            "gdbTarget": "localhost:3333",
            "cwd": "${workspaceRoot}",
            "executable": "C:/Users/megab/Desktop/Nuvoton/NuvotonWorkspace/MicroBMS/Debug/SPI_Master.elf",
            "name": "Debug with OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "serverpath": "C:/Program Files (x86)/Nuvoton Tools/OpenOCD/bin/openocd.exe",
            "servertype": "openocd",
            "configFiles": [
                "C:/Program Files (x86)/Nuvoton Tools/OpenOCD/scripts/interface/nulink.cfg",
                "C:/Program Files (x86)/Nuvoton Tools/OpenOCD/scripts/target/numicroM0.cfg"
            ],
            "postLaunchCommands": ["set mem inaccessible-by-default off"],
            "searchDir": [],
            "runToEntryPoint": "main",
            "showDevDebugOutput": "raw",
            "gdbPath": "C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.10/bin/arm-none-eabi-gdb.exe"
        }
    ]
}
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Paste Debug Console contents here

[log.txt](https://github.com/user-attachments/files/18470203/log.txt)

```

**Additional context**
You might know Nuvoton uses custom OpenOCD designed by the firm. So I'm using that version 

```
PS C:\Program Files (x86)\Nuvoton Tools\OpenOCD\bin> .\openocd.exe --version
Open On-Chip Debugger 0.10.022.0-dev-00493-gd6db465b-dirty (2023-12-14-15:04)
Licensed under GNU GPL v2
```

haneefdm, 2025-01-19:
Even if it uses a custom openocd, hopefully, it respects the command line options. Looks like it recognized the board but it never opened the gdb tcp port 50000. The timeout is 20 seconds I think

You have to find out why. If it does not open any ports, how can we/gdb talk to it? Run openocd from the command like using the same options as we did. If it did open the ports and waits for a gdb connection show that output.

haneefdm, 2025-01-19:
Focus on the TERMINAL(gdb-server) tab for this issue

ichbinbork, 2025-01-20:
_**" Run openocd from the command like using the same options as we did. If it did open the ports and waits for a gdb connection show that output."**_. Actually, I did it. You can find in the screenshot.

haneefdm, 2025-01-20:
Sorry, I missed what in the screenshot. I didn't think your `openocd` was that ancient. We have to launch a gdb server and wait for the server to initialize itself before we allow gdb to connect to it. Note that we don't talk to openocd at all. We just launch and introduce gdb to openocd and let them shake hands and do their dance

For openocd, we look for a pattern on stdio to see if the server is ready. Normally, openocd creates a TCP port and lats the user know that it is safe to connect. Here is the regex for openocd

https://github.com/Marus/cortex-debug/blob/61f3ba5fbe3b4a470c2de8f3d997f31b46b7d2be/src/openocd.ts#L274

In launch.json, you can supply your own regex using `overrideGDBServerStartedRegex`. Maybe you can do the following
```JSON
"overrideGDBServerStartedRegex": "^Info\s:\saccepting 'gdb' connection[^\n]*"
```
https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

For some reason, I don't see this message from openocd inside VSCode Terminal, but I saw it in your screenshot. Make sure I don't have typos. Experiment a bit.

If 10 seconds is not enough time there is not much we can do with the current release.

haneefdm, 2025-01-20:
Please close this issue if it is resolved.

ichbinbork, 2025-01-20:
Hey, I haven't tested it yet, but as far as I can see, there's not much to be done. Now, I'll close the issue. If it works, I'll let you know

closed 2025-01-20 by ichbinbork