Marus/cortex-debug #1083: RTT Terminal Not Working in VS Code with Remote J-Link Debugging

PaulFirs, 2025-01-23:
## Issue Description:

I am encountering issues getting the RTT terminal to work in VS Code while using a remote J-Link debugger. I am connecting to a J-Link GDB server running on a remote machine. I can successfully connect to the remote GDB server with VS Code, and I can debug my application, but 
RTT terminal is not created.

### Remote J-Link GDB Server:
The J-Link GDB server is running on a remote machine with IP address 192.168.1.199.
The server is started with the following command:
`./JLinkGDBServer -select usb=772659956 -USB -device STM32F407VE -endian little -speed 4000 -rtttelnetport 19021 -if swd -rtos GDBServer/RTOSPlugin_FreeRTOS.so`

Linux version 6.11.4-amd64 (debian-kernel@lists.debian.org) (x86_64-linux-gnu-gcc-14 (Debian 14.2.0-7) 14.2.0, GNU ld (GNU Binutils for Debian) 2.43.1) #1 SMP PREEMPT_DYNAMIC Debian 6.11.4-1 (2024-10-20)

### VS Code Configuration (launch.json):
```json
{
    "name": "run/debug: jlink IP",
    "cwd": "${workspaceRoot}",
    "executable": "${workspaceFolder}/build/ga-iridium.elf",
    "request": "launch",
    "type": "cortex-debug",
    "interface": "swd",
    "servertype": "external",
    "gdbPath": "arm-none-eabi-gdb",
    "gdbTarget": "192.168.1.199:2331",
    "svdFile": "${workspaceFolder}/STM32F407.svd",
    "runToMain": true,
    "runToEntryPoint": "main",
    "postLaunchCommands": [
        "monitor reset",
    ],
    "postRestartCommands": [
        "monitor reset",
    ],
    "showDevDebugOutput": "both",
    "rttConfig": {
        "enabled": true,
        "address": "auto",
        "decoders": [
            {
                "label": "rtt_chanel_0",
                "port": 0,
                "type": "console"
            }
        ]
    },
}
```

GDB version: GNU gdb (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7))

voltamund, 2025-01-23:
As far as I know `rttConfig` will be ignored if you use
`"servertype": "external"`

You have to use something like `"servertype": "jlink"`
and instead of `JLinkGDBServer` you start `JLinkRemoteServer` on the remote machine. This makes the JLink accessible over the network.

PaulFirs, 2025-01-23:
Thanks for your help! The issue was resolved by running the J-Link GDB Server (JLinkRemoteServer) remotely and configuring the VS Code launch.json with:

"servertype": "jlink"
"ipAddress": "<remoteIP>" (replace <remoteIP> with the actual remote IP address)

With this configuration, RTT terminal functionality works correctly. 

Truthfully, though, I now have a slightly lesser understanding of how it all works! 😅