Marus/cortex-debug #1085: Slow start, stuck on 10-file-exec-and-symbols

JonasAlmaas, 2025-01-27:
When trying to start a debug session I can see in the output that it stops on "Waiting for gdb connection...". Looking at the debug output with "showDevDebugOutput" and "showDevDebugTimestamps" I can see that 10-file-exec-and-symbols takes a really long time.

The annoying part is that I don't have this on my personal computer, only my work computer. I have tried to get help from internal support, but not knowing exactly what the issue is makes that very hard.
One thing I got from that was that disabling IPv6 system wide, witch resulted in the boot going a lot faster. Going from taking upwards of 5min to around 20s.
This is still a lot, but a big improvement.

Does anyone have any idea as to what could be causing this.

 - Cortex-Debug Version [1.12.1]
 - OS: [Windows 10]
 - GDB Version: [SEGGER J-Link GDB Server V7.98a Command Line Version]
 - Compiler Toolchain Version: [arm-none-eabi V 13.2.rel1]

**`launch.json`**

```json
{
    "name": "Cortex Debug",
    "cwd": "${workspaceFolder}",
    "executable": "${workspaceFolder}/build/analog_conv_app.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "jlink",
    "serverpath": "JLinkGDBServerCL.exe",
    "device": "STM32L431RC",
    "svdFile": "${workspaceFolder}/device/STM32L4x1.svd",
    "interface": "swd",
    "preLaunchTask": "Make",
    "showDevDebugOutput": "raw",
    "showDevDebugTimestamps": true
}
```

[debug-console.txt](https://github.com/user-attachments/files/18557234/debug-console.txt)

haneefdm, 2025-01-27:
> GDB Version: [SEGGER J-Link GDB Server V7.98a Command Line Version]

That is not gdb. that is a gdb-server.  From the text file attached, I can see the following
```
0000000077+00000: -> ~"GNU gdb (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.90.20231008-git\n"
```

The following is ~26 seconds. Well, there is not much we can do.
```
0000000196+00000: 10-file-exec-and-symbols "C:/prd/analog-converter/build/analog_conv_app.elf"
0000026112+25916: -> 10^done
0000026112+00000: Debug Time: GDB Ready...
...
0000026723+00000: 14-target-download
...
0000026849+00120: -> 14^done,address="0x0802c454",load-size="53016",transfer-rate="106032000",write-rate="5301"
````
So, it is not a very big program. We were able to read the symbols very quickly (47 and 65 ms in parallel). So, max of 65 ms. Those are normal looking times.
```
0000000012+00009: Reading symbols from arm-none-eabi-objdump.exe --syms -C -h -w C:/prd/analog-converter/build/analog_conv_app.elf
0000000017+00005: Reading symbols from arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/prd/analog-converter/build/analog_conv_app.elf
0000000059+00008: Finished reading symbols from objdump: Time: 47 ms
0000000082+00004: Finished reading symbols from nm: Time: 65 ms
```

So, I am at a loss. Virus scanners interfere with debuggers. Did you check the Task manager to see what is going on during that time.

https://community.esri.com/t5/arcgis-pro-questions/antivirus-blocks-projects-and-geodatabases/td-p/1193942

We need to test things out of VSCode's and cortex-debug context and use a terminal. Try the following in your favourite terminal (CMD, MSYS, cygwin, etc.) - make sure you are using the same verion of gdb and use the proper slashes (or backslashes) for the shell. Run the following commands (everything shown in our debug output). Anything that starts with a '-' is to be entered at the gdb prompt.

```
"C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.10/bin/arm-none-eabi-gdb.exe" -q --interpreter=mi2
-gdb-version
-file-exec-and-symbols "C:/Users/megab/Desktop/Nuvoton/NuvotonWorkspace/MicroBMS/Debug/SPI_Master.elf"
```

haneefdm, 2025-01-27:
See also...

https://sysprogs.com/w/forums/topic/gdb-version-taking-too-long-to-complete/

JonasAlmaas, 2025-01-27:
Thanks for the quick reply @haneefdm.

I tried what you said:
```
>arm-none-eabi-gdb -q --interpreter=mi2
=thread-group-added,id="i1"
(gdb)
-gdb-version
~"GNU gdb (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.90.20231008-git\n"
~"Copyright (C) 2023 Free Software Foundation, Inc.\n"
~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
~"\nType \"show copying\" and \"show warranty\" for details.\n"
~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
~"Type \"show configuration\" for configuration details.\n"
~"For bug reporting instructions, please see:\n"
~"<https://bugs.linaro.org/>.\n"
~"Find the GDB manual and other documentation resources online at:\n    <"
~"http://www.gnu.org/software/gdb/documentation/>.\n\n"
~"For help, type \"help\".\n"
~"Type \"apropos word\" to search for commands related to \"word\".\n"
^done
(gdb)
-file-exec-and-symbols C:/prd/analog-converter/build/analog_conv_app.elf
```

When I run it here it is just stuck. I tried disabling our VPN/security/malware thing (Zscaler) and it completes in instantly.

Any ideas as to how I should frame this to my support team?

haneefdm, 2025-01-27:
I would show them exactly what you are looking at. I would first reduce it down to which one is causing the issue. Is it VPN, security, or malware.

JonasAlmaas, 2025-01-27:
It is the same thing...

JonasAlmaas, 2025-01-27:
> I would show them exactly what you are looking at. I would first reduce it down to which one is causing the issue. Is it VPN, security, or malware.

It is the same thing...

I'll submit an issue, thanks for the help thus far!

closed 2025-01-27 by haneefdm

JonasAlmaas, 2025-01-31:
**For future reference**

Adding **`set auto-load no`** to **`~/.gdbinit`** seems to fix it, but I'll have to do some more testing before knowing for sure.
_Reference: https://sourceware.org/gdb/current/onlinedocs/gdb.html/Auto_002dloading-safe-path.html_