Marus/cortex-debug #1087: VsCode: Debug Breakpoint Log Message

NINI1988, 2025-02-03:
**Describe the bug**
When adding a debug breakpoint in VsCode and setting a Log Message it says: `Expressions within {} are interpolated`.
But doing so will result either in:

Log Message: `Hello {logStoreLocation.startAddress}`
```
-> =breakpoint-modified,bkpt={number="2",type="dprintf",disp="keep",enabled="y",addr="0x08033152",func="MainProcess_init",file="general/MainProcess.c",fullname="/workspaces/general/MainProcess.c",line="78",thread-groups=["i1"],times="1",script={"printf \"Hello\",{logStoreLocation.startAddress}"},original-location="/workspaces/general/MainProcess.c:78"}
-> &"Wrong number of arguments for specified format-string\n"
Wrong number of arguments for specified format-string
```

Log Message: `Hello {logStoreLocation.startAddress} ,address: {logStoreLocation.startAddress}`
```
-> =breakpoint-modified,bkpt={number="2",type="dprintf",disp="keep",enabled="y",addr="0x08033152",func="MainProcess_init",file="general/MainProcess.c",fullname="/workspaces/general/MainProcess.c",line="78",thread-groups=["i1"],times="1",script={"printf \"Hello\",{logStoreLocation.startAddress},,address:,{logStoreLocation.startAddress}"},original-location="/workspaces/general/MainProcess.c:78"}
-> &"A syntax error in expression, near `,address:,{logStoreLocation.startAddress}'.\n"
A syntax error in expression, near `,address:,{logStoreLocation.startAddress}'.
```

I already saw in https://github.com/Marus/cortex-debug/issues/542 that the format in Log message must be like :`"Hello %d\n" 123`, but this is not obvious for anyone using VsCode.

**To Reproduce**
Steps to reproduce the behavior:
1. Create breakpoint and select Log Message and enter `Hello {123}`
2. Start debug session
3. See issue in Debug console

**Expected behavior**

The log message should be printed.

**Screenshots**

[image]

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension) 1.12.1
 - OS: Linux ubuntu:22.04 devcontainer
 - GDB Version: STMicroelectronics ST-LINK GDB server. Version 7.8.0
 - Compiler Toolchain Version: arm-none-eabi-gcc (Arm GNU Toolchain 13.3.Rel1 (Build arm-13.24)) 13.3.1 20240614

**Please include `launch.json`**

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug ST-LINK application",
            "cwd": "${workspaceFolder}",
            "executable": "debug/App.elf",
            "preLaunchTask": "build application",
            "request": "launch",
            "type": "cortex-debug",
            // "runToEntryPoint": "main",
            // "breakAfterReset": false,
            "servertype": "external",
            "showDevDebugOutput": "raw",
            "gdbPath": "arm-none-eabi-gdb",
            "gdbTarget": "IP:61234",
            "overrideLaunchCommands": ["load", "monitor reset hardware"],
            "overrideRestartCommands": ["load", "monitor reset hardware"],
            "overrideResetCommands": ["monitor reset hardware"],
            "svdFile": "../bootloader/stm32h7driver/Common/STM32H7x5_CM4.svd",
            "device": "",
        },
    ]
}

```

haneefdm, 2025-10-19:
I am sorry but GDB is our backend and you *are* debugging what looks like a C/C++ program. You have to conform to the debugger you are using. This is true for any debugger... the backend influences the user interface. Here the functionality is debug printf and for C programs, this is totally natural. Infact using `{}` is un-natural.

Please lookup documentation for `GDB dprintf`

william19941994, 2025-11-19:
Yes, I have same issue.
have you resoled?