Marus/cortex-debug #1088: Failing to start pyocd gdbserver through cortex-debug

toonst, 2025-02-06:
**Describe the bug**
I cannot debug via pyocd with cortex-debug. Openocd debugging works fine.

**To Reproduce**
Use below launch configuration. Press debug in visual code gui.

**Expected behavior**
Being able to debug.

**Environment**
 - Cortex-Debug Version:  1.12.1
 - OS: macos

**Launch configuration:**
```
        {
            "cwd": "${workspaceFolder}",
            "executable": "build/zephyr/zephyr.elf",
            "name": "debug pyocd",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "pyocd",
            "targetId": "stm32g0b1retx",
            "runToEntryPoint": "main",
            "showDevDebugOutput": "raw"
        }
```

**Attach text from `Debug Console`**

```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "cwd": "/Users/toonprof/code/blinker_fw",
    "executable": "/Users/toonprof/code/blinker_fw/build/zephyr/zephyr.elf",
    "name": "debug pyocd",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "pyocd",
    "targetId": "stm32g0b1retx",
    "runToEntryPoint": "main",
    "showDevDebugOutput": "raw",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "toolchainPath": "myuser/zephyr-sdk-0.17.0//arm-zephyr-eabi/bin",
    "toolchainPrefix": "arm-zephyr-eabi",
    "extensionPath": "myuser/.vscode/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "c12baa91-5c00-473e-919d-fcf5e68e1273",
    "pvtShowDevDebugOutput": "raw"
}
Reading symbols from /myuser/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump --syms -C -h -w .../build/zephyr/zephyr.elf
Reading symbols from .../zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-nm --defined-only -S -l -C -p .../build/zephyr/zephyr.elf
Launching GDB: .../zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: pyocd gdbserver --port 50000 --telnet-port 50001 --target stm32g0b1retx
    Please check TERMINAL tab (gdb-server) for output from pyocd
Failed to launch PyOCD GDB Server: Error: spawn pyocd ENOENT

```

**Additional context**
When I run the same command manually in terminal it runs fine:
```
pyocd gdbserver --port 50000 --telnet-port 50001 --target stm32g0b1retx
0001340 I Target type is stm32g0b1retx [board]
0001388 I DP IDR = 0x0bc11477 (v1 MINDP rev0) [dap]
0001389 I debugvar 'DbgMCU_APB_Fz1' = 0x0 (0) [pack_target]
0001389 I debugvar 'DbgMCU_APB_Fz2' = 0x0 (0) [pack_target]
0001389 I debugvar 'DbgMCU_CR' = 0x6 (6) [pack_target]
0001396 I AHB-AP#0 IDR = 0x04770031 (AHB-AP var3 rev0) [discovery]
0001399 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xf0000000 (designer=020:ST part=467) [rom_table]
0001401 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c0> [rom_table]
0001401 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c0) [rom_table]
0001404 I   [0]<e000e000:SCS v6-M class=14 designer=43b:Arm part=008> [rom_table]
0001405 I   [1]<e0001000:DWT v6-M class=14 designer=43b:Arm part=00a> [rom_table]
0001406 I   [2]<e0002000:BPU v6-M class=14 designer=43b:Arm part=00b> [rom_table]
0001412 I CPU core #0: Cortex-M0+ r0p1, v6.0-M architecture [cortex_m]
0001412 I   Extensions: [MPU] [cortex_m]
0001413 I Setting core #0 (Cortex-M0+) default reset sequence to ResetSystem [pack_target]
0001414 I 2 hardware watchpoints [dwt]
0001416 I 4 hardware breakpoints, 0 literal comparators [fpb]
0001424 I Semihost server started on port 50001 (core 0) [server]
0001504 I GDB server started on port 50000 (core 0) [gdbserver]
```

haneefdm, 2025-02-06:
> Failed to launch PyOCD GDB Server: Error: spawn pyocd ENOENT

That is an OS message. Means pyocd was not found in your PATH. You can force the setting via extension settings for cortex-debug in VSCode as well.

See if pyocd can be found inside a TERMINAL window within VSCode. It can be different from the MacOS Terminal.

toonst, 2025-02-06:
Ok, thanks for the help. Fixed it by adding the following to workspace `settings.json`:

```
    "cortex-debug.pyocdPath": "${workspaceFolder}/.venv/bin/pyocd",
```

closed 2025-02-06 by toonst