Marus/cortex-debug #1090: VS Code Ide, programming the same afx file with ULINK2 and JLINK gives different result

Marbro1965, 2025-02-11:
**Describe the bug**

Microcontroller TIVA Series TM4C1290NCPDT

I have a compiled file which if programmed using ULINK2 is different than if programmed with JLINK

**To Reproduce**

Unfortunately I use a custom board. But the problem should also visible in a 

evaluation board. 

I have afx and hex file but I cannot attach.

In VSCODE use ULINK2 to program.

Read the image and save.

Use JLINK to program

Read the image and save

Compare the two image, they are different.

The program extends from 0x000000 to 0x07E3AF and Ulink2 programs all.

Jlink in VS Code does not program all the memory. It does not program from 0x07DD28 up to 0x07E3AF.

**Expected behavior**

Should be the same

**Screenshots**

**Environment (please complete the following information):**

**Please include `launch.json`**

To flash using ULINK2

{
    "version": "2.0.0",
    "tasks": [
        {
            "type": "arm-debugger.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "programs": "${command:cmsis-csolution.getBinaryFiles}",
            "cmsisPack": "${command:cmsis-csolution.getTargetPack}",
            "deviceName": "${command:cmsis-csolution.getDeviceName}",
            "processorName": "${command:cmsis-csolution.getProcessorName}",
            "problemMatcher": [],
            "label": "Flash Device"
            
        },
        {
            "type": "shell",
            "label": "Reset Device",
            "command": "arm-none-eabi-gdb",
            "args": [
                "--batch",
                "-ex", "target remote localhost:2331",
                "-ex", "monitor reset halt",
                "-ex", "disconnect",
                "-ex", "quit"
            ],
            "problemMatcher": []

        },
        {
            "label": "Flash and Reset Device",
            "dependsOrder": "sequence",
            "dependsOn": [
                "Flash Device",
                "Reset Device"
            ]
        }

    ]
}

To flash using JLINK

        {
            "name": "Debug with JLink",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "device": "TM4C1290NCPDT",
            "interface": "jtag",    
            "runToEntryPoint": "main",
            "showDevDebugOutput": "none",
            "cwd": "${workspaceFolder}",
            "serverpath": "C:/Program Files/SEGGER/JLink_V810a/JLinkGDBServerCL.exe",
            "executable": "./out/HT1000I/Versione6/HT1000I.axf",
        },

**Attach text from `Debug Console`**

**Additional context**

haneefdm, 2025-02-11:
If you look in the Debug Console you will see that we do not program anything. GDB does. If gdb has trouble understanding how to program your axf file, not much we can do.

Good news is that you have a hex file. You can use that to program via JLink. Keep everything else the same but add the following.

```JSON
"loadFiles": [ "your-hex-filename" ],
```

The Debug Console is your friend. With our debugger, you have to get used this area. There is a lot of self help with instructions and documentation. It appears that you have not even enabled  verbose logging. We are not paid for this service. So self help is highly recommended.

You did not attach the Debug Console output. Sue you cannot attach the axf/hex file but surely you can attach what we printed!?!?

haneefdm, 2025-02-11:
No, what I said is that gdb programs the board and we tell gdb to do it. So, in the end, the board is programmed using gdb and we tell gdb to do that for you.

We ourselves do not know anything about programming the board.

haneefdm, 2025-02-11:
Please use the hex file like I said above and it should work.

Marbro1965, 2025-02-11:
There is an evident bug. I think it is an Arm problem. The output file is
programmed in two different way depending on the driver.
I will ask segger))). No one is using Tiva  tm4c with compiler 6, for a
number of reasons.first is that it does not compile the old library which
must be compiled old way. The second is that no support for jlink in vscode.