Marus/cortex-debug #1091: Unexpected disconnect right after connection to pyOCD

gordonhjs, 2025-02-14:
Hi,

Not sure if i should post this issue here or on pyOCD, but since it happens only with the plugin, I chose here. Tell me if it's not suited.

**Unexpected disconnect right after connection to pyOCD**

When I start the debug with the cortex-debug pluging, pyOCD starts and GDB connects to it until step 11 (11-target-select extended-remote localhost:50000) where it fails (-> 11^error,msg="Remote connection closed").

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. See issue

**Expected behavior**

I expect GDB to not disconnect to pyOCD during starting.

**Environment:**

 - Cortex-Debug Version (this extension): 1.12.1
 - OS: Linux Ubuntu 24.04 LTS
 - GDB Version: arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gdb
 - Compiler Toolchain Version:  arm-gnu-toolchain-14.2.rel1 (but I tried with 13.3 and 12.3 as well)

**Please include `launch.json`**
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceFolder}",
            "armToolchainPath": "/usr/local/bin/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/",
            "executable": "./■■■■■■■■■■■■■■■■■■■■■■■/my_fw.axf",
            "request": "attach",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "pyocd",
            "targetId": "LPC54114J256BD64",
            "serverArgs": ["-v"],
            "showDevDebugOutput": "raw",
        }
    ]
}
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Cortex Debug",
    "cwd": "/home/■■■■■■■■■■■■■■■■■■■■■■■/my-project",
    "armToolchainPath": "/usr/local/bin/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/",
    "executable": "./■■■■■■■■■■■■■■■■■■■■■■■/my_fw.axf",
    "request": "attach",
    "type": "cortex-debug",
    "runToEntryPoint": "main",
    "servertype": "pyocd",
    "targetId": "LPC54114J256BD64",
    "serverArgs": [
        "-v"
    ],
    "showDevDebugOutput": "raw",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
[... 197 lines cut ...]
0001066 D added=[] removed=[] [manager]
0001066 D bps after flush={} [manager]
0001070 D resuming core 1 [cortex_m]
0001070 D added=[] removed=[] [manager]
0001070 D bps after flush={} [manager]
0001070 D added=[] removed=[] [manager]
0001070 D bps after flush={} [manager]
0001082 D closing interface [pyusb_backend]
0001084 D gdbserver shutdown event set; waiting for exit [gdbserver]
0001104 D gdbserver exiting connection loop [gdbserver]
0001104 I Client disconnected from port 50000! [gdbserver]
0001104 D gdbserver thread exiting [gdbserver]
0001104 D GDB server cleaning up [gdbserver]
0001204 D Error in packet IO thread: [Errno 9] Bad file descriptor [packet_io]
0001204 D GDB packet thread stopping [packet_io]
[2025-02-14T09:03:43.036Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

**Additional context**

When I start pyOCD and GDB in my terminal, the pyOCD logs differ after `0001062 D GDB received query: [b'Supported', b'multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;QThreadOptions+;no-resumed+;memory-tagging+#e7'] [gdbserver]` line, where I get `0044641 D GDB general set: b'StartNoAckMode' [gdbserver]` instead of `uninit` lines.

I'm available for more information of tests.
Thanks!

haneefdm, 2025-02-14:
Error messages are coming from pyocd. It could not even get started. It failed to open the USB connection because you have not configured the Linux permissions using `udev` rules. This has nothing to do with us. It is possible you have not installed your boards SW completely. 
```
[2025-02-14T09:03:41.774Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
pyocd gdbserver --port 50000 --telnet-port 50001 --target LPC54114J256BD64 -v
0000281 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=5986 PID=211b). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000282 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0bda PID=8179). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000283 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=5986 PID=211b). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000283 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0bda PID=8179). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000283 D Project directory: /home/gordon/work/smartX/sx-firmware [session]
0000603 D Project directory: /home/gordon/work/smartX/sx-firmware [session]
0000603 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=5986 PID=211b). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000603 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0bda PID=8179). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000604 D Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0143). [pyusb_backend]
0000628 D CMSIS-DAP v1 probe QZFQQUT1SELCO: protocol version 1.1.0 [dap_access_cmsis_dap]
0000632 D closing interface [pyusb_backend]
0000822 I Target type is lpc54114j256bd64 [board]
0000823 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=5986 PID=211b). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000823 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0bda PID=8179). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000824 D Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0143). [pyusb_backend]
0000842 D Running task load_svd [sequencer]
```
Make sure pyocd can run from the command line first

gordonhjs, 2025-02-14:
I did install the udev rule manually for my probe `probe QZFQQUT1SELCO` as the logs state, idk why pyocd wants to find other probes.

I'll try to install all rules to see if its better. 

I understand that it's not a cortex-debug issue, thank for your fast reply!

gordonhjs, 2025-02-14:
@haneefdm  I think I got it.

The plugin starts pyocd gdb server with ports 50000 and 50001 (with command `pyocd gdbserver --port 50000 --telnet-port 50001 --target LPC54114J256BD64 -v` above)
This command also makes the server crash at start in command line with error `OSError: [Errno 98] Address already in use`.
I see this error from command line, but not from the vscode terminal window, is this normal?

Since my mcu is dual core, It seems that pyocd opens (or tries to) 2 gdb servers (one for each core), so is using 50000 and 50001 for gdb servers. 

If I run the command with `pyocd gdbserver --port 50000 --telnet-port 50002 --target LPC54114J256BD64 -v`, I indeed get those logs:

```
0000950 D Setting vector catch to 0x00000001 [cortex_m]
0000953 I Semihost server started on port 50002 (core 0) [server]
0000963 I GDB server started on port 50000 (core 0) [gdbserver]
0000965 D Setting vector catch to 0x00000001 [cortex_m]
0000968 I Semihost server started on port 50003 (core 1) [server]
0000968 I GDB server started on port 50001 (core 1) [gdbserver]
0057164 D Starting GDB server packet I/O thread [packet_io]

```

And then pyocd is starting.

In the plugin I didn't find any config item to change those ports ?
What is working is to start pyocd in command line with `pyocd gdbserver --port 50000 --telnet-port 50002 --target LPC54114J256BD64 -v`, and configure the plugin with:

```
            "servertype": "external",
            "gdbTarget": "localhost:50000"
```

haneefdm, 2025-02-14:
> This command also makes the server crash at start in command line with error OSError: [Errno 98] Address already in use.

This means that one of your experiments had pyocd still running. Use the task manager to make sure there are no other instances of pyocd running. Or you did not configure your settings/launch.json correctly

> I see this error from command line, but not from the vscode terminal window, is this normal?

I have no idea.... not sure exactly what you are seeing,

> Since my mcu is dual core, It seems that pyocd opens (or tries to) 2 gdb servers (one for each core), so is using 50000 and 50001 for gdb servers.

Then in launch.json, you need to tell us how many processors you have and which processor you want to debug. See the documentation. There is no (good) way for us to guess and we don't have information for the thousands of processors we support.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

There is even a wiki for multi-core cpus https://github.com/Marus/cortex-debug/wiki

No, you cannot change the ports. We have dozens of ports we have to manage and we are careful not use something already in use.

Please close this issue unless you see a specific issue with cortex-debug.

closed 2025-02-17 by gordonhjs

haneefdm, 2025-02-17:
So, you didn't use `numberOfProcessors` and `targetProcessor` in your launch.json?

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

That would have told us what your processor looks like and how many ports to reserve and how to connect to pyocd or any other gdb-server