Marus/cortex-debug #1094: SWO/RTT usability issues with `"servertype": "external"` connecting to a remote OpenOCD

dgazzoni, 2025-02-19:
**Is your feature request related to a problem? Please describe.**
I have what is apparently an unusual setup: I have some evaluation boards (mostly STM32) connected to a headless Raspberry Pi, and I'd like to debug these remotely (sometimes in my own network, sometimes over the internet). Note: by remote I do not mean something like the "Remote SSH" extension, but actually through a contraption I've created using VS Code tasks, launch targets, shell scripts which connect over ssh, copy ELF files, forward ports, etc. For cortex-debug I'm using `"servertype": "external"`, connecting over the network to the Raspberry Pi.

Somehow all of this is working, except that I need support for some form of `printf`, say SWO or RTT. Up until now I was using further contraptions based on SWO and this script [this script](https://github.com/robertlong13/SWO-Parser), as I cannot get the cortex-debug SWO window to work. Here is my current config:

```
        {
            "name": "Debug (OpenOCD, remote)",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "external",
            "gdbTarget": "127.0.0.1:50000",
            "cwd": "${workspaceFolder}",
            "executable": "${command:cmake.launchTargetPath}",
            "device": "STM32F407VG",
            "runToEntryPoint": "main",
            "svdFile": "STM32F407.svd",
            "preLaunchTask": "SWO parser",
            "postDebugTask": "Stop OpenOCD",

            // "postLaunchCommands": [
            //     "monitor stm32f4x.tpiu configure -protocol uart -output :50003 -traceclk 24000000 -pin-freq 2000000",
            //     "monitor stm32f4x.tpiu enable"
            // ],
            // "swoConfig": {
            //     "enabled": true,
            //     "source": "socket",
            //     "swoPort": "127.0.0.1:50003",
            //     "swoFrequency": 2000000,
            //     "cpuFrequency": 24000000,
            //     "decoders": [
            //         {
            //             "port": 0,
            //             "type": "console",
            //             "label": "SWO output",
            //             "encoding": "ascii",
            //             "showOnStartup": true
            //         }
            //     ]
            // }

            // "postLaunchCommands": [
            //     "monitor rtt setup 0x20000000 2048 \"SEGGER RTT\"",
            //     "monitor rtt start",
            //     "monitor rtt server start 9090 0"
            // ],
            // "rttConfig": {
            //     "enabled": true,
            //     "address": "auto",
            //     "rtt_start_retry": 100,
            //     "decoders": [
            //         {
            //             "label": "Terminal",
            //             "port": 0,
            //             "type": "console"
            //         }
            //     ]
            // }
        },
```

I uncomment the first commented block if trying to use SWO, or the second commented block if trying to use RTT.

Unfortunately there is some kind of issue with OpenOCD and SWO, which looks like a buffer overrun, since some of the characters are "eaten up" in the output. If I use the texane st-link tool `st-trace`, I can read from SWO just fine, without eating up any characters, so it looks like an OpenOCD issue.

Thus I've been investigating RTT as well, but when trying to configure RTT with `"servertype": "external"`, it doesn't even open the RTT viewer window, and apparently doesn't send the required commands to the GDB server either. By issuing some commands manually, I got it to work, but it's quite a bit of a hassle.

Just to make sure that none if this is an issue with e.g. the OpenOCD version on my local Mac, I unplugged the board from the Raspberry Pi 5 and plugged it to the Mac, and then used `"servertype": "openocd"` rather than `"servertype": "external"`. With this configuration, both SWO and RTT work perfectly, without garbling/corrupting a single character. Similarly, I opened the project directly in the Raspberry Pi (from my Mac, using the Remote SSH extension), and ran the debugger again using `"servertype": "openocd"` -- which makes sense since it is now local to the Raspberry Pi which I was connecting via Remote SSH. Again, both RTT and SWO work perfectly. So it appears that the whole software stack, whether on the Raspberry Pi or the Mac (including cortex-debug itself, if using `"servertype": "openocd"`), works perfectly. However, given how much faster the Mac is for e.g. building the project, compounded by issues with network latency if doing it over the internet, Remote SSH is a solution I would rather avoid if at all possible.

Thus, apparently all my issues can be traced to either bugs or limitations on the use of `"servertype": "external"` in Cortex-Debug.

**Describe the solution you'd like**
Being unfamiliar with the architecture of cortex-debug, I have no idea if it would be too much work, or even feasible, but what I'd like to see is a way of combining the features of `"servertype": "external"` and `"servertype": "openocd"`: maybe something like a `"serversubtype"` field for `"external"` which I could set to `"openocd"`, or a way to use `"servertype": "openocd"` over a network connection, or even a `"servertype": "external-openocd"`.

Failing that, I'd like to see SWO working properly with "external" -- I suspect some of the required commands are not being sent to OpenOCD, even though I send the commands which I know are needed manually (and which are enough for my contraption based on the SWO parser script to work), as can be seen from the `launch.json` configuration above. Similarly, it seems like `"rttConfig"` doesn't really do anything in `"external"` mode.

**Describe alternatives you've considered**
I believe I've described everything I tried above. I consider all of the alternatives suboptimal at the very least: either garbled output for SWO, or manual steps for RTT, or using a much slower computer (potentially compounded by network latency) if using Remote SSH to the Raspberry Pi. If I had the budget I might consider a J-Link with proper network connection but unfortunately that's not realistic for me.

**Additional context**
No additional context.

haneefdm, 2025-02-19:
Server type "external" is designed and intended for gdb servers that are either running remotely or some new server that we do not know about. It has a lot of limitations and anything that is not basic debugging (SWO, RTT, etc.) is likely to have issues.

For both RTT and SWO, we use TCP ports that we carefully construct knowing which server you are using and what we have to do over the command line, scripting and gdb commands. FYI: gdb is NOT involved in RTT or SWO. Everything happens bypassing GDB. Openocd's RTT implementation is designed for manual use and not for automated tools -- we tried real hard to automate it and even then it can fail. It is a stupid implementation (I have my own patched openocd for this reason).

I have heard of people using server:external with an swoPort on the external server successfully. In fact that capability was supplied by one of the users. The problem is that of a handshake and what happens at reset. The remote processor probably started running and already emitted some stuff that went missing. then when we take over we interrupt whatever the processor was doing - another way you could potentially get garbage.

> a way to use "servertype": "openocd" over a network connection, or even a "servertype": "external-openocd"

No. If I did that, then I would have twice as many gdb-servers to maintain. Too many things can go wrong because starting and initializing the gdb-server is left up to the user and I cannot rely on them doing everything needed. It is a support nightmare. I don't have that kind of time to volunteer. Besides, that is the wrong way to do it -- called "designed to failure".

If you use something like JLink, they have the capability of running a gdb-server on the remote server but make it look like everything is running locally. And, no one (other SW) will even know that it is being done that way.

https://www.segger.com/products/debug-probes/j-link/tools/j-link-remote-server/

haneefdm, 2025-02-19:
Without experimenting with your specific setip, I cannot tell what is wrong with your post-launch commands. Seems to be the same as what we do, you can see what we do as documented in our wiki.

https://github.com/Marus/cortex-debug/blob/master/support/openocd-helpers.tcl

And, how we use it through gdb. Note "through" gdb. GDB has no clue what commands we are sending and why

https://github.com/Marus/cortex-debug/blob/61f3ba5fbe3b4a470c2de8f3d997f31b46b7d2be/src/openocd.ts#L186

It may be a question of timing as to when the TPIU/ITM is configured.

Also, you can do everything from the command line using command line gdb until you got a working recipe. Also, enable debug output in Cortex-Debug and see what commands we send.

> Lastly: would you consider at least getting minimal RTT support for "servertype": "external", as you have for SWO? Although for some reason it doesn't work for me, at least the support is there.

It is pointless to add RTT support for openocd because it requires polling openocd on a side channel until RTT starts. Like I said, they have the stupidest implementation and far too many things can go wrong

From your messages, this is very confusing to me

```JSON
            "gdbTarget": "127.0.0.1:50000",
            //     "swoPort": "127.0.0.1:50003",
```

You say openocd is running remotely but also telling us it is running on localjost (127.0.0.1). Which is it? Remote openocd or local? Where is openocd running? Do you have output from openocd?

haneefdm, 2025-02-19:
> if I telnet localhost 50003, I get (garbled) output. Still, on the cortex-debug SWO window, not a single character is printed.

That is because SW bytes are encoded and will look like garbage. That is the ARM spec. Anyone who reads SWO output will need a decoder ring :-)

dgazzoni, 2025-02-20:
> It is pointless to add RTT support for openocd because it requires polling openocd on a side channel until RTT starts. Like I said, they have the stupidest implementation and far too many things can [...]

Ah, that's too bad. Maybe I'll open a bug with them suggesting they fix this.

In my particular case I'm putting the call to `SEGGER_RTT_Init` on a function marked `__attribute__((constructor))`, so by the time it stops in `main`, RTT should already be available. But I guess the checks you do are when you reset the MCU.

> From your messages, this is very confusing to me
> 
>             "gdbTarget": "127.0.0.1:50000",
>             //     "swoPort": "127.0.0.1:50003",
> You say openocd is running remotely but also telling us it is running on localjost (127.0.0.1). Which is it? Remote openocd or local? Where is openocd running? Do you have output from openocd?

Elsewhere I've mentioned that I connect to the Raspberry Pi via ssh, and use it to forward ports to my local machine. That way, when developing from outside of my network, I do not have to open these ports from my firewall.

haneefdm, 2025-02-20:
> Ah, that's too bad. Maybe I'll open a bug with them suggesting they fix this.

They should auto poll once RTT is enabled just like JLink (SEGGER) does. This is after all SEGGERs protocol and they should emulate that. Hitting a breakpoint just after reset should not cause RTT to fail. Openocd expects the user to continuously start RTT manually. We should be able to simply enable it and it should auto-poll until the control block is found/initialized.

Know that `main` is not special for us at all. Only the reset vector. At reset we know RTT is not functional or even stale from the previous contents in SRAM. You putting a constructor attribute does nothing for us.

What we do at reset is invalidate any previous RTT buffer and let the gdb-server find it again. Or it can find the wrong (or corrupted) one. We also give RTT the exact address of where it should find the RTT control block so no guessing involved and no slow scanning of the entire SRAM. Btw, how RTT is setup is different for every gdb-server. Note that we enable RTT at reset because that is the only reliable place to do it.

Please close this as this usage pattern is not a supported and if you can get it to work for you, great. With OpenOCD, you have a better chance with SWO than with RTT if you are using an external server type. You should find out why that is not working.

haneefdm, 2025-02-20:
What does the Debug Console say? Make sure you follow the instructions to enable raw debug output. You should always look at both outputs to see what is going on.

haneefdm, 2025-02-20:
You did say that a local telnet to 127.0.0.1:50003 works as in you are getting something even if it looks like garbage?

haneefdm, 2025-02-20:
We have a timeout of 5 minutes and we (re)try to connect every 250ms. No, there is no log for this.

dgazzoni, 2025-02-20:
> You did say that a local telnet to 127.0.0.1:50003 works as in you are getting something even if it looks like garbage?

Yes. Actually, now that I retried it, and I suppose it's somehow linked to me having disconnected the board from the Raspberry Pi to connect to my Mac and then later reconnecting it to the Raspberry Pi, I get perfect ASCII output from telnetting to that port. That is, it looks like OpenOCD is somehow already decoding the SWO output, in contrast to what you said here: https://github.com/Marus/cortex-debug/issues/1094#issuecomment-2670026925.

I'm not sure if it's a "new" behavior from OpenOCD which differed from the assumptions when this was implemented previously -- FYI, I'm using the stock OpenOCD 0.12.0 from apt on the Raspberry Pi 5, which is running Ubuntu 24.04.1 LTS.

I guess the garbled output I was getting (and which I'm still getting from a second board that I use here, a NUCLEO-L4R5ZI) is related to something else, possibly some kind of USB issue on the Raspberry Pi.

Here is the debug console output, after adding `"showDevDebugOutput": "raw"` and `"showDevDebugTimestamps": true` to `launch.json`. What I did was: press the "debug" button (which opened the OpenOCD and the SWO windows;); wait a few seconds; manually telnet into port 50003; press the "continue" button; wait a few seconds; press the pause button. The SWO window never showed anything.

```
0000000002+00002: Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
0000000002+00000: "configuration": {
    "name": "Debug (OpenOCD, remote, DISCOVERY)",
    "type": "cortex-debug",
    "request": "launch",
    "servertype": "external",
    "gdbTarget": "127.0.0.1:50000",
    "cwd": "/Users/...",
    "executable": "/Users/decio/.../build/bench.elf",
    "device": "STM32F407VG",
    "runToEntryPoint": "main",
    "svdFile": "/Users/decio/.../STM32F407.svd",
    "preLaunchTask": "SWO parser",
    "postDebugTask": "Stop OpenOCD",
    "showDevDebugOutput": "raw",
    "showDevDebugTimestamps": true,
    "postLaunchCommands": [
        "monitor stm32f4x.tpiu configure -protocol uart -output :50003 -traceclk 24000000 -pin-freq 2000000",
        "monitor stm32f4x.tpiu enable"
    ],
    "swoConfig": {
        "enabled": true,
        "source": "socket",
        "swoPort": "50003",
        "swoFrequency": 2000000,
        "cpuFrequency": 24000000,
        "decoders": [
            {
                "port": 0,
                "type": "console",
                "label": "SWO output",
                "encoding": "ascii"
            }
        ]
    },
    "__configurationTarget": 6,
    "gdbServerConsolePort": 60001,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "toolchainPrefix": "arm-none-eabi",
    "extensionPath": "/Users/decio/.vscode/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": false,
    "pvtVersion": "1.12.1",
[... 250 lines cut ...]
Info : flash size = 1024 KiB
Info : flash size = 512 bytes
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32f4x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080018c4 msp: 0x2001c000
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32f4x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080018c4 msp: 0x2001c000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32f4x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080018c4 msp: 0x2001c000
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32f4x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080018c4 msp: 0x2001c000
Info : starting trace server for stm32f4x.tpiu on 50003
Info : Listening on port 50003 for tpiu_swo_trace connections
Info : accepting 'tpiu_swo_trace' connection on tcp/50003
Info : dropped 'tcl' connection
```

haneefdm, 2025-02-20:
What we get with SWO is raw output in packet format. If someone is already decoding it, then our decoder will fail. Packets contain more than just text. they can be text, hardware, timestamps, synchronization, overflow, etc.

The header of a packet is a singly byte that is usually unprintable so it may look like telnet is eating them but in fact the console/terminal maybe the one eating it.

But, the main thing is that we are NOT seeing a connection being established. To debug this, you have to put our debugger itself into the VSCode JavaScript debugger.

I don't know how your port forwarding works. Is it possible we are connecting to 5003 but on that was created long time ago and thus seeing nothing? I would do everything fresh just to make sure. When a debug session ends, are you closing all the ports? We never close ports since that is the servers' job and there could be other clients.

haneefdm, 2025-02-20:
```
Info : starting trace server for stm32f4x.tpiu on 50003
Info : Listening on port 50003 for tpiu_swo_trace connections
Info : accepting 'tpiu_swo_trace' connection on tcp/50003
Info : dropped 'tcl' connection
```
What is this? Looks like someone made a connection on 50003. Is this by you using telnet or by cortex-debug.

haneefdm, 2025-02-20:
I want to pay attention to the lifecycle of the 5003 port. I have a feeling that the ssh tunnel is opening the 5003 port before openocd even created one and we probably connected to it. When openocd final create the 5003 port on the remote machine, ssh probably starts forwarding that one while closing the old one. With all that, we probably disconnected and during the short time we were connected there was no output to look at.

dgazzoni, 2025-02-20:
> I want to pay attention to the lifecycle of the 5003 port. I have a feeling that the ssh tunnel is opening the 5003 port before openocd even created one and we probably connected to it. When openocd [...]

Nailed it!

Prompted by this, I did a simple change to `launch.json`: rather than `"swoPort": "127.0.0.1:50003"`, I used `"swoPort": "192.168.0.5:50003"`, where 192.168.0.5 is the IP address of the Raspberry Pi. And now the SWO window works!

I believe you are completely right. What happened is, `ssh` opened the 50003 port on the Mac, and any connections to that port will cause ssh to open a connection to the Raspberry Pi, but of course, until the binary is flashed and the `tpiu` commands are run, that port is closed. So anyone trying to connect before that will get immediately disconnected, as in this example I did manually, in the time window between pressing "debug" and stopping in the breakpoint at `main`:

```
% telnet localhost 50003
Trying ::1...
Connected to localhost.
Escape character is '^]'.
Connection closed by foreign host.
```

So I guess SWO was working all along, and the only problem was with my weird setup. Well, guess I'll have to figure out a different way of using it -- although directly connecting to the Raspberry Pi works while I'm inside my network, it doesn't over the internet, but I'll figure something out.

Thank you very much, and I guess now we can really close this issue.

closed 2025-02-20 by dgazzoni