Marus/cortex-debug #1099: The MCU goes into hard fault directly after debugger startup - STM32MP157C-M4

maxime-gaschet-devmg, 2025-03-13:
*Description*

I'm trying to debug the M4 core of the STM32MP157C using a JLINK probe. The chip is in ENGINEERING mode, so the A7 core shouldn't interfere. The project uses CMAKE.
When I use Stm32CubeIde the debug is done correctly. On startup, the breakpoint on the main is correct and step-by-step works.
When I use VScode with cortex-debug, I can have a breakpoint on the startup.s but once I do a run or a step by step, the chip goes into hard fault (PC = 0x08).
Given that the 2 configurations use exactly the same tools, I wonder if the problem might be with the cortex-debug plugin?
Thanks for your support.

*Environment*
 - Cortex-Debug Version 1.12.1
 - OS: Linux Ubuntu 24.10
 - GDB Version: stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.externaltools.jlink.linux64_2.4.0.202501261557
 - Compiler Toolchain Version: stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.13.3.rel1.linux64_1.0.0.202410170706

*`launch.json`*
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug JLink",
            "cwd": "${workspaceRoot}",
            "executable": "/home/user/repo/ceva_4dot0_needleless_stm32mp157c_m4_cmake/build-debug_mg_lnx/ceva_4dot0_needleless_stm32mp157c_m4_cmake.elf", // CMake.
            // "executable": "/home/user/app/cubeIdeWorkspace/debugger_with_cubeIde_cmake/build/ceva_4dot0_needleless_stm32mp157c_m4_cmake.elf", // CubeIde.
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "STM32MP15xx_M4",
            "interface": "swd",
            // "preLaunchTask": "CMake: build",
            "runToEntryPoint": "main",
            // "breakAfterReset" : true,
            // "liveWatch": {
            //     "enabled": true,
            //     "samplesPerSecond": 4
            // },
            "showDevDebugOutput": "raw",
            "swoConfig": {
                "enabled": false,
            },
           "serverArgs": [
                "-speed", "4000", // kHz.
                "-vd", // Verify download.
            ],
            "numberOfProcessors": 2,
            "targetProcessor": 0, // 0 = M4.
        }
    ]
}
```

*`cortex-debug Console`*
```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Debug JLink",
    "cwd": "/home/user/repo/ceva_4dot0_needleless_stm32mp157c_m4_cmake",
    "executable": "/home/user/repo/ceva_4dot0_needleless_stm32mp157c_m4_cmake/build-debug_mg_lnx/ceva_4dot0_needleless_stm32mp157c_m4_cmake.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "jlink",
    "device": "STM32MP15xx_M4",
    "interface": "swd",
    "runToEntryPoint": "main",
    "showDevDebugOutput": "raw",
    "swoConfig": {
        "enabled": false,
        "source": "probe",
        "decoders": []
    },
[... 398 lines cut ...]
*`cubeIde DebuggerConsole`*
```
New UI allocated
GNU gdb (GNU Tools for STM32 13.3.rel1.20240926-1715) 14.2.90.20240526-git
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) 0x00000008 in g_pfnVectors ()

Temporary breakpoint 1, main ()
    at /home/user/repo/ceva_4dot0_needleless_stm32mp157c_m4_cmake/CM4/Core/Src/main.c:78
78        mainApp_preHalInit();
(gdb)                                                                                                                                                                                                                                   
```

maxime-gaschet-devmg, 2025-03-13:
After a CONTINUE and a PAUSE
[image]

haneefdm, 2025-03-13:
What you are doing may work but not how we do it. We use the command line version of JLink Server

>     "serverpath": "/opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.externaltools.jlink.linux64_2.4.0.202501261557/tools/bin/JLinkGDBServer",

should be something like

>     "serverpath": "<path-to-server>/JLinkGDBServerCLExe",

See https://github.com/Marus/cortex-debug/wiki/J-Link-Specific-Configuration I am curious why you are using JLink from the ST's installation. We normally install JLink and use it from its install directory. Do you know what the difference is?

> When I use Stm32CubeIde the debug is done correctly. On startup, the breakpoint on the main is correct and step-by-step works.
> When I use VScode with cortex-debug, I can have a breakpoint on the [...]

In our deebug console, you can see the EXACT commands being sent. You can do the same with Stm32CubeIde, see what we are doing differently, and customize Cortex-Debug to do the equivalent.

We do not control the execution of your FW. We simply reset the chip and expect it to be halted at the reset vector. We do not have any finer control than that. Then we simply issue gdb commands to break, continue, step, etc. We only talk to gdb after the initial reset.

Are you sure you are loading the whole program (given your CPU is multi-core).

mg-ceva, 2025-03-19:
Hello, thank you for your feedback.

> What you are doing may work but not how we do it. We use the command line version of JLink Server
> should be something like

    "serverpath": "<path-to-server>/JLinkGDBServerCLExe",

Yes I tried but the result is the same. :-(

> I am curious why you are using JLink from the ST's installation.

We want to use the tools provided by ST. I'm pretty sure the result would be the same even if I use the latest SEGGER tools. Furthermore, it's the same tools that works with cubeIde, so it should work with vsCode.

> Are you sure you are loading the whole program (given your CPU is multi-core).

Yes, I have placed the chip in ENGINEERING mode, so the A7 core is supposed to hand over control to the M4. Additionally, I have added the verify download. So, it should be okay at this level.

Well, I think I'm going to give up with JLink.

So I turned to the ST-LINK V3 debugger and openOcd. 
And... once again, under cubeIde with this debugger, it also works but not under VSCode. There is really something I am missing.

The launch.json configuration:
```
{
    "name": "debug-stlink",
    "cwd": "${workspaceRoot}",
    "executable": "./build/build-mg-aurora-ubuntu-vscode/sis2-needleless-m4.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "openocd",
    "interface": "swd",
    "runToEntryPoint": "main",
    "showDevDebugOutput": "raw",
    "numberOfProcessors": 2,
    "targetProcessor": 0, // 0 = M4.
    "configFiles": ["cubeide_cmake debug-stlink-v3.cfg"],
    "searchDir": [
        "/opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.3.100.202501240831/resources/openocd/st_scripts",
        "/opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mpu.debug.openocd_2.2.0.202409171044/resources/openocd/st_scripts"
    ],
    "serverArgs": [
        "-c", "gdb_report_data_abort enable"
    ]
}
```

I added in settings.json:
`"cortex-debug.openocdPath":        "/opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.linux64_2.4.100.202501161620/tools/bin/openocd"`

The idea is still the same, to use the tools provided by cubeIde.

The gdb-server terminal under VSCode:
```
[2025-03-19T18:55:00.751Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.linux64_2.4.100.202501161620/tools/bin/openocd -c "gdb_port 50000" -c "tcl_port 50002" -c "telnet_port 50004" -s /opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.3.100.202501240831/resources/openocd/st_scripts -s /opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mpu.debug.openocd_2.2.0.202409171044/resources/openocd/st_scripts -f /home/user/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f "cubeide_cmake debug-stlink-v3.cfg" -c "gdb_report_data_abort enable"
Open On-Chip Debugger 0.12.0+dev-00608-gd8ed48fef (2025-02-06-11:15) [https://github.com/STMicroelectronics/OpenOCD]
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDLiveWatchSetup
Info : Listening on port 50002 for tcl connections
Info : Listening on port 50004 for telnet connections
Warn : 'srst_nogate' reset_config option is required
Info : STLINK V3J16M7B5S1 (API v3) VID:PID 0483:374F
Info : Target voltage: 3.253723
Info : clock speed 8000 kHz
Warn : 'srst_nogate' reset_config option is required
Info : stlink_dap_op_connect(connect)
Warn : 'srst_nogate' reset_config option is required
[... 166 lines cut ...]
Info : STM32MP157CAAx.cpu0: hardware has 6 breakpoints, 4 watchpoints
Info : [STM32MP157CAAx.cpu0] Examination succeed
Info : STM32MP157CAAx.cpu1: hardware has 6 breakpoints, 4 watchpoints
Info : [STM32MP157CAAx.cpu1] Examination succeed
Info : gdb port disabled
Info : gdb port disabled
Info : gdb port disabled
Info : starting gdb server for STM32MP157CAAx.cpu0 on 3334
Info : Listening on port 3334 for gdb connections
Info : starting gdb server for STM32MP157CAAx.cm4 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : [STM32MP157CAAx.cm4] target was in unknown state when halt was requested
Info : New GDB Connection: 1, Target STM32MP157CAAx.cm4, state: halted
Info : accepting 'gdb' connection on tcp/3333
Info : New GDB Connection: 2, Target STM32MP157CAAx.cm4, state: halted
```

What I don't understand is that when it reaches 
`Info: Listening on port 3333 for gdb connections `
in the case of cubeIde, we directly get: 
`Info: accepting 'gdb' connection on tcp/3333`
whereas under VSCode, it times out... 

Any idea?

haneefdm, 2025-03-19:
> Well, I think I'm going to give up with JLink.
> The idea is still the same, to use the tools provided by cubeIde.

This is a good thing, especially for openocd. Didn't know ST was bundling JLink with ST. JLink releases are standard and they update much more frequently than ST does. In JLink case, I would prefer to use SEGGER releases (Also easier to find on filesystem). Also, in most cases, I prefer JLink to OpenOCD but that doesn't matter.

Something is wrong with OpenOCD here...we issued this command-line.
> /opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.linux64_2.4.100.202501161620/tools/bin/openocd -c "gdb_port 50000" -c "tcl_port 50002" -c "telnet_port 50004" -s [...]

Pay special attention to `-c "gdb_port 50000" -c "tcl_port 50002" -c "telnet_port 50004"`. We select these ports and ask OpenOCD to use them. We have been doing this since the beginning of this extension. Lets esamine the OpenOCD output

>>>
Info : Listening on port 50002 for tcl connections
Info : Listening on port 50004 for telnet connections
>>>

This is good. As expected so the command line options had an effect. And then, we see the following

>>>
Info : [STM32MP157CAAx.ap1] Examination succeed
Info : [STM32MP157CAAx.cm4] Cortex-M4 r0p1 processor detected
Info : [STM32MP157CAAx.cm4] target has 6 breakpoints, 4 watchpoints
Info : [STM32MP157CAAx.ap2] Examination succeed
Info : [STM32MP157CAAx.axi] Examination succeed
Info : STM32MP157CAAx.cpu0: hardware has 6 breakpoints, 4 watchpoints
Info : [STM32MP157CAAx.cpu0] Examination succeed
Info : STM32MP157CAAx.cpu1: hardware has 6 breakpoints, 4 watchpoints
Info : [STM32MP157CAAx.cpu1] Examination succeed
Info : gdb port disabled
Info : gdb port disabled
Info : gdb port disabled
>>>
It disabled all gdb ports (there should be as many ports as there are processors). Are you sure there are two processors? It seems like you have three. Anyways, why are the ports getting disabled? Normally, right after processor discovery, you should see ports being used. Not disabled. But, it did the same thing for ST tools. Regardless, something undid the command line option we passed.
>>>
Info : starting gdb server for STM32MP157CAAx.cpu0 on 3334
Info : Listening on port 3334 for gdb connections
Info : starting gdb server for STM32MP157CAAx.cm4 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
>>>
It ignored our command line argument for `gdb_port`. It is possible that one of your config files did this.

> -f "cubeide_cmake debug-stlink-v3.cfg"

Are you sure you are using the proper config files? On the STMCube side, I can't tell what the command line options were (you should be able to get this from the debug dialog box in Eclipse). Make sure you are using the right config files.

If you are using the right config file, then we have to debug that file to see why the ports are getting disabled and why the gdb_port option is getting overridden. This may be a bug in the config file provided by ST.

haneefdm, 2025-03-19:
If you run a debug session even once in the ST IDE with openocd, it saves the config file alongside your project. I generally just use that and perhaps add/remove any redundant/harmful stuff. You (and ST) own this file btw and your whole debug experience is dependent on how OpenOCD is configured. We don't do any of that because it is very low-level HW/probe/processor/JTAG/ARM/SWD specific. I use the ST IDE as reference and then try to understand and do what they do. Cortex-Debug takes over TCP port management and setting up things like SWO/RTT (which includes TCP port management)

> Given that the 2 configurations use exactly the same tools, I wonder if the problem might be with the cortex-debug plugin?

Tools yes. But how they are configured can make a big difference.

mg-ceva, 2025-03-22:
Hello there,

> Are you sure you are using the proper config files?

Yes I'm using the config file generated by cubeIde.

> On the STMCube side, I can't tell what the command line options were

Here are the command line options used in cubeIde : 

`
/opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.linux64_2.4.100.202501161620/tools/bin/openocd "-f" "cubeide_cmake debug-stlink-v3.cfg" "-s" "/home/user/repo/sis2-needleless-m4" "-s" "/opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.3.100.202501240831/resources/openocd/st_scripts" "-s" "/opt/st/stm32cubeide_1.18.0/plugins/com.st.stm32cube.ide.mpu.debug.openocd_2.2.0.202409171044/resources/openocd/st_scripts" "-c" "gdb_report_data_abort enable" "-c" "gdb_port 3333" "-c" "tcl_port 6666" "-c" "telnet_port 4444"`

They are the same that mine excepted port numbers.

``

haneefdm, 2025-03-22:
> set GDB_PORT 3333

You should comment that out. Do not reset it. Because, what we pass over the command line can change in some cases

About jlink, no idea what is going on. You have to do a similar thing as you did with OpenOCD to see if they are doing something that is missing with Cortex-Debug.

We cannot monitor every change every vendor does to every gdb-server. There is simply not enough resources for us to chase that. With JLink, you can pass an additional script to do customer stuff. Perhaps ST is doing that for this device

haneefdm, 2025-03-22:
Btw, copy (and use) the openocd cfg file generated by ST to your own file name. Next time you use ST tools, it will be wiped out.

stemschmidt, 2025-04-28:
The file cubeide_cmake debug-stlink-v3.cfg sources target/stm32mp1x.cfg. Could you please show the content of this file?

stemschmidt, 2025-04-29:
Hi mg-ceva, 
I meant the content of the file target/stm32mp1x.cfg :-)

Does the M4 Core have any access port protection which might lock the debug interface?

mg-ceva, 2025-04-30:
Hello, the content of stm32mp1x.cfg is:

```
# STMicroelectronics STM32MP15x (Single/Dual Cortex-A7 plus Cortex-M4)
# http://www.st.com/stm32mp1

# HLA does not support multi-cores nor custom CSW nor AP other than 0
if { [using_hla] } {
	echo "ERROR: HLA transport cannot work with this target."
	echo "ERROR: To use STLink switch to DAP mode, as in \"board/stm32mp15x_dk2.cfg\"."
	shutdown
}

source [find target/swj-dp.tcl]
source [find mem_helper.tcl]
source [find gdb_helper.tcl]

if { [info exists CHIPNAME] } {
	set _CHIPNAME $CHIPNAME
} else {
	set _CHIPNAME stm32mp15x
}

if { [info exists CPUTAPID] } {
	set _CPUTAPID $CPUTAPID
} else {
	if { [using_jtag] } {
		set _CPUTAPID 0x6ba00477
	} else {
		set _CPUTAPID 0x6ba02477
	}
}

# Chip Level TAP Controller, only in jtag mode
if { [info exists CLCTAPID] } {
	set _CLCTAPID $CLCTAPID
} else {
	set _CLCTAPID 0x06500041
}

swj_newdap $_CHIPNAME tap -expected-id $_CPUTAPID -irlen 4
if { [using_jtag] } {
	jtag newtap $_CHIPNAME.clc tap -expected-id $_CLCTAPID -irlen 5
}

dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.tap -ignore-syspwrupack

# APNum definition
# set _AP_NUM_CM4 2
# set _AP_NUM_CA7 1
# set _AP_NUM_AXI 0

# FIXME: Cortex-M code requires target accessible during reset, but this is not possible in STM32MP1
# so defer-examine it until the reset framework get merged
# NOTE: keep ap-num and dbgbase to speed-up examine after reset
# NOTE: do not change the order of target create
target create $_CHIPNAME.ap1 mem_ap -dap $_CHIPNAME.dap -ap-num 1
target create $_CHIPNAME.ap2 mem_ap -dap $_CHIPNAME.dap -ap-num 2
target create $_CHIPNAME.axi mem_ap -dap $_CHIPNAME.dap -ap-num 0
target create $_CHIPNAME.cpu0 cortex_a -dap $_CHIPNAME.dap -ap-num 1 -coreid 0 -dbgbase 0xE00D0000
target create $_CHIPNAME.cpu1 cortex_a -dap $_CHIPNAME.dap -ap-num 1 -coreid 1 -dbgbase 0xE00D2000
target create $_CHIPNAME.cm4 cortex_m -dap $_CHIPNAME.dap -ap-num 2 -defer-examine -gdb-max-connections 2

swo  create $_CHIPNAME.swo  -dap $_CHIPNAME.dap -ap-num 1 -baseaddr 0xE0083000
tpiu create $_CHIPNAME.tpiu -dap $_CHIPNAME.dap -ap-num 1 -baseaddr 0xE0093000

if { [info exists GDB_PORT] } {
	set _GDB_PORT $GDB_PORT
} else {
	set _GDB_PORT [ocd_gdb_port]
[... 82 lines cut ...]

proc detect_cpu1 {} {
	set cpu1_prsr [$::_CHIPNAME.ap1 read_memory 0xE00D2314 32 1]
	set dual_core [expr {$cpu1_prsr & 1}]
	if {! $dual_core} {$::_CHIPNAME.cpu1 configure -defer-examine}
}

proc rcc_enable_traceclk {} {
	$::_CHIPNAME.ap2 mww 0x5000080c 0x301
}

# FIXME: most of handler below will be removed once reset framework get merged
$_CHIPNAME.ap1  configure -event reset-deassert-pre  {adapter deassert srst deassert trst;catch {dap init};catch {$::_CHIPNAME.dap apid 1}}
$_CHIPNAME.ap2  configure -event reset-deassert-pre  {dbgmcu_enable_debug;rcc_enable_traceclk}
$_CHIPNAME.cpu0 configure -event reset-deassert-pre  {$::_CHIPNAME.cpu0 arp_examine}
$_CHIPNAME.cpu1 configure -event reset-deassert-pre  {$::_CHIPNAME.cpu1 arp_examine allow-defer}
$_CHIPNAME.cpu0 configure -event reset-deassert-post {toggle_cpu0_dbg_claim0}
$_CHIPNAME.cm4  configure -event reset-deassert-post {$::_CHIPNAME.cm4 arp_examine;if {[$::_CHIPNAME.ap2 curstate] == "halted"} {$::_CHIPNAME.cm4 arp_halt}}
$_CHIPNAME.ap1  configure -event examine-start       {dap init}
$_CHIPNAME.ap2  configure -event examine-start       {dbgmcu_enable_debug}
$_CHIPNAME.cpu0 configure -event examine-end         {detect_cpu1}
$_CHIPNAME.ap2  configure -event examine-end         {rcc_enable_traceclk;$::_CHIPNAME.cm4 arp_examine; if {[$::_CHIPNAME.swo cget -traceclk]} {$::_CHIPNAME.swo enable} }
``` 

To answer your question, the chip in question is an STM32MP157C containing an M4 core and an A7 dual core. The M4 is slave to the A7, which is designed to run Linux. However, there is a mode called ENGINEERING mode, which should make it possible to dispense with loading by the A7. So, yes, I suspect there is a mechanism to debug the M4 core.

stemschmidt, 2025-04-30:
I am pretty sure there is a mechanism to debug the M4 core :-)

My point is that there might be some special handling to enable debug access. In the file you posted I see axi_secure and dbgmcu_enable_debug which might point you to what you have to setup in order to access the debug port of the M4.

I do not know this chip but I think you have to enable the debug port to be able to access the internals of the M4 core. I assume that if the debug port is not enabled, a hardfault will be dispatched when you try to access anything within the M4 core.