Marus/cortex-debug #1100: Need help

mars3142, 2025-03-13:
I do have a testing project: https://github.com/mars3142/stm32_template. Don't except fancy stuff, I just wanted to use VSCode instead of the Eclipse IDE for STM32 ;) 

Now, I have the launch configuration, which uploads both elf files (bootloader.elf and firmware.elf) to my device and starts the bootloader. After 5 seconds, the bootloader calls the firmware, which is expected and working perfect (checked with STMCubeMonitor - TIM2 triggers the led - also visible via blinking led).

My problem is, that I can't set any working breakpoint in firmware.c. I can set one, but it will never be stop there. Also "break firmware.c:LINE" in the GDB shell isn't working - "no file found". For bootloader.c everything is running smooth.

I would be happy, if someone could help with that. I use the stable cortex-debug plugin (the preview isn't working for me) and the latest STMCubeCTL (1.18.0).

PS: This issue contains in the discussion section, but there is none.

haneefdm, 2025-03-13:
First thing you should do is follow the Issue submission template and give everyone relevant information.

On the surface looks like you are not debugging the right elf file and make sure all elf files are loaded (programmed).

closed 2025-03-13 by haneefdm

mars3142, 2025-03-13:
Thanks for nothing. You're comment is a bad reaction for this repository. I did give all context. What was missing?

haneefdm, 2025-03-13:
I gave you as much help as I can base on the information you gave. Sorry, I can't guess what your setup is and what your logs look like. I am not that capable. I posed your exact question to ChatGPT and got the same response as what I gave you, but in a verbse way and it thought I should know about GDB quite a bit which I also wish everyone did. Comes in handy for trouble shooting.

Why do you refuse to use the Issue template? Interesting that you flout our rules and expect us to spend time on your issue.

mars3142, 2025-03-13:
I refuse the templates, because none is working for my issue. You also offer "blank issue", so I used it. Don't offer this one, if you don't want that.

I wrote my setup: vscode, stmcubectl 1.18.0, latest cortex-debug plugin. And I added my complete repository, including the the vscode launch configuration. It starts perfect, but every breakpoint within firmware.c doesn't stop the execution.