Marus/cortex-debug #1102: SWO tracing with ST-Link GDB server

Feliksas, 2025-03-19:
Greetings,

As far as I understood, cortex-debug does not support getting SWO data from ST-Link GDB server in probe mode, but it is still possible to tell the gdb server to actually expose SWO traffic on a separate port (-z parameter).
I see that there is a possibility to configure cortex-debug to get SWO data via a TCP socket, but there is no documentation on it whatsoever, besides the brief mention that it exists. After reading the source, I've discovered that there is an swoPort config variable, that looks like the right place to specify the port that I've assigned via -z switch to the gdb server - but cortex-debug just silently contitues, and an SWO output terminal is not being spawned. I've come up with the following config:
```
 "swoConfig":{
                "enabled":true,
                "source":"socket",
                "swoPort": "127.0.0.1:3443",
                "swoFrequency": 1000000,
                "cpuFrequency":168000000,
                "decoders": [
                    {
                        "port": 0,
                        "type": "console",
                        "label": "SWO output",
                        "encoding":"ascii"
                    }
                ]
            }
```
The rest is pretty much auto-generated by the official STM32 VSCode extension, save for additional args to ST-Link gdb server.

Is such a configuration possible at all? Would be nice to have a proper config example on how to setup getting SWO output from a network socket.

haneefdm, 2025-03-19:
If you need to pass additional arguments to the gdb-server, then use `serverArgs`. Remember that every word is a separate item for serverArgs array just like any unix/windows command line.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md
https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood

>  see that there is a possibility to configure cortex-debug to get SWO data via a TCP socket, but there is no documentation on it whatsoever

We point you to the available docs right there in the Debug Console. There is also a Wiki for more complex situations. Your own gdb server docs should be a reference.

If you were using stlink, the swo setup should have already done anything for you. the `swoConfig.enabled` is the trigger. We already use the `-z` equivalent `--swo-port` to specify the port. You should not be using `swoConfig. swoPort` -- not sure why you used this!?!?

I would have preferred to see the entire launch.json because I have very little context as to what you are doing. If you suspect an issue, please use our issue submission template to submit a proper bug report.

Feliksas, 2025-03-19:
Well, I started digging into that exactly because by default it doesn't seem that SWO tracing was being configured in any way, at least with how the "STM32 VS Code Extension" configures launch.json - for example, serverArgs contain only ['-m', '0'], so I see no way how ST-Link GDB server would automagically bind the SWO port, unless some more arguments are being sneakily mixed in (which is not the case, judging by the full command line in the debug output log). Every `swoConfig` example that I could find focuses on usage of "probe" mode, which is clearly not supported with ST-Link GDB server. So I tried manually configuring it in socket mode, for which (obviously), the target port needs to be specified, that's why I used `swoConfig.swoPort`
I will check again how the default config behaves in detail, when I get to my workstation, and will then post existing configuration - but one could easily replicate my setup just by installing STM32 VS Code Extension and creating a new project with it.

Feliksas, 2025-03-22:
Turns out your response was completely misleading. SWO support for ST-Link with STM32CubeCLT is simply not yet present in the current stable cortex-debug version 1.12.1, which is currently published in Marketplace. Support was only added in https://github.com/Marus/cortex-debug/commit/611752b482232ab98de25e6a8b4a1bf4e9d548a1 . Was this really that hard to mention instead of sending people chasing a red herring down the rabbit hole?

haneefdm, 2025-03-22:
Very sorry about that. Did you try the pre-release available here..

https://github.com/Marus/cortex-debug/releases

Please, please, please. Next time,

> If you suspect an issue, please use our issue submission template to submit a proper bug report.

Feliksas, 2025-03-22:
The pre-release version somehow gets stuck after
```
Launching GDB: "c:\\ST\\STM32CubeCLT_1.18.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb" -q --interpreter=mi2
1-gdb-version
```
Still investigating that issue

haneefdm, 2025-03-22:
My Windows machine is down. Adjusting for the (back)slashes (use proper ones depending on your shell), can you see if the gdb command exists? Maybe we have the '.exe' missing (should not be needed).

We should start seeing messages from GDB. You can also use the `armToolchainPath` but it should work as you have it.

My Windows machine is not accessible right now.

haneefdm, 2025-03-23:
Can you attach the entire Debug Console output (maybe as a file if too large)? I can't duplicate what you are seeing. Complete output of Debug Console is a requirement for us to debug anything.  I have no idea what is going on at this point.

Feliksas, 2025-03-26:
I have resolved the issue, it was due to misconfiguration. My working config for the pre-release version V1.13.0-pre3 is attached (leaving it here for posterity)

There is one strange issue, though - after VSCode is relaunched, debugging freezes right at the line where SWO output is generated. I managed to get over it by randomly changing the decoder type to binary, and then back to ascii, but somehow I can't reproduce it anymore.

[launch.json](https://github.com/user-attachments/files/19474173/launch.json)

Feliksas, 2025-04-01:
A quick follow-up for anyone who comes upon this thread:

I have solved the issue of debug hanging on attempting to output something via SWO: cortex-debug doesn't correctly initialize the debug hardware on STM32 devices (I'm using a STM32F407VG), so one needs to supply their own initialization routine, either via code on the MCU side, or by means of custom GDB init script via `preLaunchCommands` config variable, as described [here](https://github.com/Marus/cortex-debug/blob/10dfac8312ed3b745fdacbf1b3ba5cea8ec57eb4/support/gdb-swo.init#L17)

I have ended up doing it from MCU side like so:
```c
void SWO_Init(void) {
    uint32_t SWOPrescaler = (FCPU / SWO_BAUD) - 1U;

    // Global enable all DWT and ITM features
    CoreDebug->DEMCR = CoreDebug_DEMCR_TRCENA_Msk;                                                       
   
    // Trace IO pin enabled in async mode, Keep debug hardware clocked in Standby, STOP and Sleep modes
    DBGMCU->CR = DBGMCU_CR_TRACE_IOEN | DBGMCU_CR_DBG_STANDBY | DBGMCU_CR_DBG_STOP | DBGMCU_CR_DBG_SLEEP; 
    
    // Serial Wire Output - NRZ (UART)
    TPI->SPPR  = TPI_SPPR_TXMODE_Msk & 0x2U;                                                             
    
    // Set prescaler value for SWO
    TPI->ACPR  = SWOPrescaler;                                                                          
    
    // Unlock write access to other ITM registers
    ITM->LAR   = 0xC5ACCE55u;                                                                            
    
    // ITM Enable, DWT Sync Enable, DWT Stimulus enable, ATB ID 0x1
    ITM->TCR   = ITM_TCR_ITMENA_Msk | ITM_TCR_SYNCENA_Msk | ITM_TCR_DWTENA_Msk | (1UL << 16U);            
    
    // Unmask stimulus ports 7:0
    ITM->TPR   = ITM_TPR_PRIVMASK_Msk;

    // Enable stimulus port 0                                                                    
    ITM->TER   = 0x1UL;                                                                                   
}
```