Marus/cortex-debug #1104: STM32N6 support (boot from RAM)

amurzeau, 2025-03-25:
**Is your feature request related to a problem? Please describe.**
The default cortex-debug configuration doesn't seem to be able to launch a program on STM32N657.
This is a new STM32 which doesn't have internal FLASH, but instead either boot from external flash, or from RAM directly when in debug mode (this can be configured using BOOT0 and BOOT1 pins).

I'm sharing here the required launch.json to be able to launch a program using ST-Link (thanks for the ability to override commands !):
```
{
	"cwd": "${workspaceFolder}",
	"executable": "${command:cmake.launchTargetPath}",
	"name": "Debug STM32N657 with ST-Link",
	"request": "launch",
	"type": "cortex-debug",
	"servertype": "stlink",
	"runToEntryPoint": "main",
	"svdPath": "STM32N657.svd",
	"serverArgs": [
		"-m",
		"1"
	],
	"overrideLaunchCommands": [
		"-target-download",
		"-interpreter-exec console \"monitor halt\""
	]
},
```

The important parts are:
- `-m 1` ST-Link arguments, ST-Link won't attach without that argument.
- `overrideLaunchCommands` override the default commands to avoid a reset after loading. A reset will make the just-loaded binary in RAM lost.

**Describe the solution you'd like**

I'm not sure how this special case can be handled as I guess the reset is otherwise necessary on other devices.
But I guess this might at least help other seeking to debug a STM32N6 using VSCode.

**Describe alternatives you've considered**
I've not tried other gdbservers (like openocd), maybe they can be easier to use in that particular case.

haneefdm, 2025-03-27:
Note. If I am not wrong, the following
```json
	"overrideLaunchCommands": [
		"-target-download",
		"-interpreter-exec console \"monitor halt\""
	]
```
can be
```json
	"overrideLaunchCommands": [
		"-target-download",
		"monitor halt"
	]
```
Anything that does not start with a '-' will become a gdb normal command and the `-interpreter-exec console` prepended to it. Less to type. Can you try it out?

Also, are you sure you need the `overrideLaunchCommands` because what you have is what we do. You do need the '-m' option as we do not know you want to debug the non-default processor. Nothing to do with booting from RAM.

We can add info like this to our Wiki in the ST specific section (we can create one if it does not exist). But first, we should make sure we are guiding people properly

amurzeau, 2025-03-29:
Hi,

I've tried `monitor halt` instead of `-interpreter-exec console \"monitor halt\"` and it works.
```
	"overrideLaunchCommands": [
		"-target-download",
		"monitor halt"
	]
```

About `overrideLaunchCommands`, I've tried without `overrideLaunchCommands` but after the `target-download`, the processor is reset which lose the entry point as PC goes back into the boot ROM code in an infinite loop (at address 0x18003514):
```
11-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"0x18003a1a in ?? ()\n"
0x18003a1a in ?? ()
-> *stopped,frame={addr="0x18003a1a",func="??",args=[],arch="armv8.1-m.main"},thread-id="1",stopped-threads="all",core="1"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 11^connected
12-interpreter-exec console "monitor reset"
-> @"STM32 Successfully completed reset operation (System reset)\n"
STM32 Successfully completed reset operation (System reset)
-> 12^done
13-target-download
-> 13+download,{section=".isr_vector",section-size="844",total-size="11040993"}
-> 13+download,{section=".isr_vector",section-sent="844",section-size="844",total-sent="844",total-size="11040993"}
-> 13+download,{section=".text",section-size="170648",total-size="11040993"}
-> 13+download,{section=".rodata",section-size="16652",total-size="11040993"}
-> 13+download,{section=".init_array",section-size="12",total-size="11040993"}
-> 13+download,{section=".fini_array",section-size="4",total-size="11040993"}
-> 13+download,{section=".data",section-size="1068",total-size="11040993"}
-> 13^done,address="0x341a6bec",load-size="189228",transfer-rate="6728104",write-rate="11131"
14-interpreter-exec console "monitor reset"
Finished reading symbols from nm: Time: 1387 ms
-> @"STM32 Successfully completed reset operation (System reset)\n"
STM32 Successfully completed reset operation (System reset)
-> 14^done
15-interpreter-exec console "monitor halt"
-> @"Trying to halt core...\n"
Trying to halt core...
-> 15^done
16-break-insert *__assert_func
-> 16^done,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x341a8d08",func="__assert_func",file="/build/gnu-tools-for-stm32_11.3.rel1.20230912-1600/src/newlib/newlib/libc/stdlib/assert.c",fullname="/build/gnu-tools-for-stm32_11.3.rel1.20230912-1600/src/newlib/newlib/libc/stdlib/assert.c",line="58",thread-groups=["i1"],times="0",original-location="*__assert_func"}
Returning dummy thread-id to workaround VSCode issue with pause button not working
Returning dummy stack frame to workaround VSCode issue with pause button not working: {"threadId":1,"startFrame":0,"levels":20}
17-exec-continue --all
-> 17^running
-> *running,thread-id="all"
mi2.status = running
Info: Received readMemory request while busy. {"memoryReference":"0x40005c00","count":44}
Failed to update peripheral I2C3: Error: peripheral-viewer: readMemory failed @ 0x40005c00 for 44 bytes: CodeExpectedError: Busy, session=fedc50ab-bb3f-4f1f-8bde-c6bb6d865429
18-exec-interrupt 
-> 18^done
-> ~"\nProgram"

Program
-> ~" received signal SIGINT, Interrupt.\n"
 received signal SIGINT, Interrupt.
-> ~"0x18003514 in ?? ()\n"
0x18003514 in ?? ()
-> *stopped,reason="signal-received",signal-name="SIGINT",signal-meaning="Interrupt",frame={addr="0x18003514",func="??",args=[],arch="armv8.1-m.main"},thread-id="1",stopped-threads="all",core="1"
mi2.status = stopped
```

I guess the target-download, halt, reset sequence comes from this code:
https://github.com/Marus/cortex-debug/blob/b9b557e947922e92c8cc5017c0ded2ce70b28b74/src/common.ts#L393-L407
https://github.com/Marus/cortex-debug/blob/b9b557e947922e92c8cc5017c0ded2ce70b28b74/src/stlink.ts#L153-L160

So I used `overrideLaunchCommands` to only halt after `target-download` but not reset. Indeed maybe this is not related to booting from RAM but specific to the STM32N6 (the boot option used for development specifically goes into an infinite loop in the boot ROM code, waiting debugger access).

haneefdm, 2025-03-29:
Thanks for trying it out and reporting back. What you see in our source code is some generic ST initialization that works for most of their processors. We can't possibly keep up with their entire catalog. That is their job. What I do is see what cortex-debug is doing vs what ST's own tools are doing and reconcile. This is why we have so much customization available..

I will see if I can find a place in the Wiki. Are you familiar with our Wiki? I know that no one reads it :-(

https://github.com/Marus/cortex-debug/wiki

amurzeau, 2025-03-30:
The wiki is where I found about `overrideLaunchCommands` :)

I would suggest in https://github.com/Marus/cortex-debug/wiki/ST-Link-(st-util)-Specific-Configuration
But when searching about a solution to get STM32N6 debug working, I also searched the issues.

Indeed there are many models available, so I'm ok with the idea of keeping this information here in this issue rather than the wiki if you find it better.

hlmd0, 2025-06-26:
Hi any luck doing this with openocd? I am using the following configuration. It is almost working, except for whenever I hit a breakpoint the device seems to get stuck and will not let me step through the code. Almost like hitting breakpoints is causing it to reset. Any experience with this?

``` 
        {
            "cwd": "${workspaceRoot}",
            "executable": "Template_FSBL.elf",
            "name": "Debug with OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "configFiles": [
                "stm32n6_fsbl.cfg"
            ],
            "searchDir": [
                "code/embedded/stm32n6/openocd"
            ],
            "runToEntryPoint": "main",
            "showDevDebugOutput": "none",
             "openOCDLaunchCommands": [
                 "init",
                 "reset halt",
             ],
            "postLaunchCommands": [
                "monitor halt",
                "load",
            ],
            "overrideLaunchCommands": [
                "monitor halt",
            ],
            "svdFile": "STM32N657.svd"
        },
```

closed 2025-10-19 by haneefdm