Marus/cortex-debug #1106: Could not load source file when Debugging

gfcwfzkm, 2025-03-27:
**Describe the bug**
Starting a debug session (with either JLink or STLink, both were tested) results in VSCode being unable to open/find the source file. Adding the additional configuration `"postLaunchCommands": ["set substitute-path / ./"]` at least let VSCode find the source file with the main entry point, but it still keeps failing to display any other source file.

Only when I set the substitute path `./` to the absolute path on my drive with unix-style path seperator characters, it manages to successfully open the right source files where GDB is pointing to.

I don't think this is the right solution, as that would require me to hard-code the projects location into the `launch.json`. I did try `${workspaceFolder}`, but due to it using the native windows path seperator, that too fails.

It wouldn't surprise me if something is broken on my end, but I can't figure out what it might be and I would be very thankful for any help or pointers.

**To Reproduce**

Simply starting any debug session on either my desktop or laptop.

**Expected behavior**

I expect it to correctly figure out which source file to display and open.

**Screenshots**

Without any `postLaunchCommands`:

[image]

With  `"postLaunchCommands": ["set substitute-path / ./"]`:

[image]

**Environment (please complete the following information):**

 - Cortex-Debug Version **1.12.1**
 - OS: **Windows 11 Pro**
 - GDB Version: GNU gdb (GNU Tools for STM32 12.3.rel1.20240612-1315) 13.2.90.20230627-git
 - Compiler Toolchain Version: (GNU Tools for STM32 12.3.rel1.20240612-1315) 12.3.1 20230626

**Please include `launch.json`**

<details>
<summary>launch.json</summary>

```json
{
	// Use IntelliSense to learn about possible attributes.
	// Hover to view descriptions of existing attributes.
	// For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
	"version": "0.2.0",
	"configurations": [
		{
			// Quick Test using JLINK to make sure it's not a issue related to a specific debugger
			"name": "Build & Debug - JLink",
			"cwd": "${workspaceFolder}",
			"executable": "${command:cmake.launchTargetPath}",
			"request": "launch",
			"type": "cortex-debug",
			"runToEntryPoint": "main",
			"servertype": "jlink",
			"windows": {
				"serverpath": "C:/Program Files/SEGGER/JLink_V812g/JLinkGDBServerCL",
				"gdbPath": "${env:STM32CLT_PATH}/GNU-tools-for-STM32/bin/arm-none-eabi-gdb",
				"postLaunchCommands": [
					"set substitute-path / C:/Users/pasca/Music/kontron_spitfair"	// This works fully
					//"set substitute-path / \"${workspaceFolder}\""		// Doesn't work due to mixing linux and windows path separators
					//"set substitute-path / ./"	// Only works on the main.cpp file
				]
			},
			"linux": {
				"serverpath": "JLinkGDBServer",
				"gdbPath":"arm-none-eabi-gdb",
			},
			"serverArgs": ["-speed", "12000"],
			"device": "STM32G071RBTx",
[... 217 lines cut ...]
-> ~"\n"

-> ~"Temporary breakpoint 1, main () at /src/main.cpp:4\n"
Temporary breakpoint 1, main () at /src/main.cpp:4
-> &"4\t/src/main.cpp: No such file or directory.\n"
4	/src/main.cpp: No such file or directory.
-> *stopped,reason="breakpoint-hit",disp="del",bkptno="1",frame={addr="0x0800024e",func="main",args=[],file="/src/main.cpp",fullname="/src/main.cpp",line="4",arch="armv6s-m"},thread-id="1",stopped-threads="all",core="0"
mi2.status = stopped
-> =breakpoint-deleted,id="1"
18-thread-list-ids
-> 18^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
19-thread-info 1
-> 19^done,threads=[{id="1",target-id="Thread 1",name="main",frame={level="0",addr="0x0800024e",func="main",args=[],file="/src/main.cpp",fullname="/src/main.cpp",line="4",arch="armv6s-m"},state="stopped",core="0"}]
20-stack-list-frames --thread 1 0 19
-> 20^done,stack=[frame={level="0",addr="0x0800024e",func="main",file="/src/main.cpp",fullname="/src/main.cpp",line="4",arch="armv6s-m"}]
21-stack-info-frame --thread 1 --frame 0
-> 21^done,frame={level="0",addr="0x0800024e",func="main",file="/src/main.cpp",fullname="/src/main.cpp",line="4",arch="armv6s-m"}
22-stack-list-variables --thread 1 --frame 0 --simple-values
-> 22^done,variables=[]
```

</details>

**Additional context**
-

gfcwfzkm, 2025-03-27:
Thank you for your time and help! Trying out `info sources` resulted in the following output:

<details>
<summary>Output of [info sources]</summary>

```
C:\Users\pasca\Music\kontron_spitfair\lib\stm32cubemx\startup_stm32g071xx.s
 /src/main.cpp
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_gpio.h
 C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\arm-none-eabi\include\machine\_default_types.h
 C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\arm-none-eabi\include\sys\_stdint.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/CMSIS/Device/ST/STM32G0xx/Include/stm32g071xx.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/CMSIS/Device/ST/STM32G0xx/Include/stm32g0xx.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_dma.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_usart.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_utils.h
 /lib/stm32cubemx/Src/cube.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Core/Inc/main.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_rcc.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/CMSIS/Include/core_cm0plus.h
 C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\lib\gcc\arm-none-eabi\12.3.1\include\stdint.h
 C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\arm-none-eabi\include\stdint.h
 C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\arm-none-eabi\include\sys\features.h
 C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\arm-none-eabi\include\_newlib_version.h
 C:\ST\STM32CubeCLT_1.16.0\GNU-tools-for-STM32\arm-none-eabi\include\sys\_intsup.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/CMSIS/Include/cmsis_version.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/CMSIS/Include/cmsis_compiler.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/CMSIS/Include/cmsis_gcc.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/CMSIS/Include/mpu_armv7.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/CMSIS/Device/ST/STM32G0xx/Include/system_stm32g0xx.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_bus.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_system.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_exti.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_cortex.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_pwr.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Drivers/STM32G0xx_HAL_Driver/Inc/stm32g0xx_ll_dmamux.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Core/Inc/gpio.h
 /lib/stm32cubemx/cmake/stm32cubemx/../../Core/Inc/usart.h
 /lib/stm32cubemx/Src/cube.c
 /lib/stm32cubemx/Core/Src/main.c
 /lib/stm32cubemx/Core/Src/gpio.c
 /lib/stm32cubemx/Core/Src/usart.c
 /lib/stm32cubemx/Core/Src/stm32g0xx_it.c
 /lib/stm32cubemx/cmake/stm32cubemx/../../Core/Inc/stm32g0xx_it.h
 /lib/stm32cubemx/Drivers/STM32G0xx_HAL_Driver/Src/stm32g0xx_ll_gpio.c
 /lib/stm32cubemx/Drivers/STM32G0xx_HAL_Driver/Src/stm32g0xx_ll_usart.c
 /lib/stm32cubemx/Drivers/STM32G0xx_HAL_Driver/Src/stm32g0xx_ll_rcc.c
 /lib/stm32cubemx/Drivers/STM32G0xx_HAL_Driver/Src/stm32g0xx_ll_utils.c
 /lib/stm32cubemx/Core/Src/system_stm32g0xx.c
 /build/gnu-tools-for-stm32_12.3.rel1.20240612-1315/src/newlib/newlib/libc/string/memset.c
 /build/gnu-tools-for-stm32_12.3.rel1.20240612-1315/install-native/lib/gcc/arm-none-eabi/12.3.1/include/stddef.h
 /build/gnu-tools-for-stm32_12.3.rel1.20240612-1315/src/newlib/newlib/libc/include/string.h
 /build/gnu-tools-for-stm32_12.3.rel1.20240612-1315/src/newlib/newlib/libc/misc/init.c
```

</details>

Already it seems suspicious that there are a lot of shortened path names. Curious about this, I used this trick to [print preprocessor values during compile time](https://stackoverflow.com/a/10791845), in order to see what the macro `__FILE__` would return in my `main.cpp` - it printed the exact same path, even when applied other local library files!

Since my project is based on a CMake template provided by my professor, I decided to investigate the `compiler.cmake` file a little and found this:
```
-ffile-prefix-map=${CMAKE_SOURCE_DIR}/=/    # shorten `__FILE__`
```
Well, it seems like that using this flag absolutely breaks debugging paths! Simply commenting this flag out not only returns a full path in the preprocessor printing test, but also when running the VSCode debugger! 

**Problem found and solved!** Thank you for your help!

I do wonder if there is a way to make it work? GDB doesn't seem too happy using `set substitute-path` with the windows-style path seperators, that VSCode's `${workspaceFolder}` would return and I couldn't find a way to have `${workspaceFolder}` with unix-style path seperators on Windows. Having to type in the absolute path manually with unix-style path seperators is not a good way.

Microsofts C/C++ plugin seems to [support](https://code.visualstudio.com/docs/cpp/launch-json-reference#_additional-properties) something to fix it on the editor's side, but it does not seem to work / be supported in this case. A low priority "issue" either way.

haneefdm, 2025-03-27:
Something you could do as part of your build is to generate a file you can source into gdb (or create a .gdbinit file).

I have no idea what cpptools does with `sourceFileMap`. Not what I expected. People have better results with gdb's substitute-path when done right. Both gdb and vscode are happy.

And, why did your professor do that? Save disk space? At least that could have been './' instead of '/' the root of a file system.

closed 2025-03-27 by haneefdm

gfcwfzkm, 2025-03-27:
I will look into the gdb option and how well it can be automated to avoid hardcoded paths.

./ instead of / does not work, as it will now try to search for the file at the path location where the .ELF file is.

Probably did that to save space on microcontrollers and I can see why. No need to have the full path, just the filename would already be enough to trace things. If done with some `assert` macro and done often, I could see how this ends up into a lot of strings at the program flash.