Marus/cortex-debug #1110: Provide a way to override SWO configuration commands

Feliksas, 2025-04-01:
**Is your feature request related to a problem? Please describe.**
I would like to use a custom GDB script for SWO initialization sequence, as suggested [here](blob/10dfac8312ed3b745fdacbf1b3ba5cea8ec57eb4/support/gdb-swo.init#L17), but to be able to actually use it, the default SWO configuration sequence that is hardcoded in the respective debug server configuration (like [here](blob/92ad7ce850fb1fdb774e7371f87ff7109fe9d225/src/stlink.ts#L186)) - needs to be unset or overidden, since it will be run anyway  after any custom preLaunchCommands that I specify.

**Describe the solution you'd like**
In my opinion, it would be nice, instead of spreading SWO initialization sequence over a series of atomic commands, as it is currently done in [stlink.ts](blob/92ad7ce850fb1fdb774e7371f87ff7109fe9d225/src/stlink.ts), to make it into a single command named `SWO_Init`, like [here](blob/10dfac8312ed3b745fdacbf1b3ba5cea8ec57eb4/support/gdb-swo.init#L33), which would be possible to easily override by sourcing a custom GDB script that contains the function with the same name.
Alternatively, the list of SWO init commands can be exposed as a config variable, which would override the default list.

**Describe alternatives you've considered**
Solutions I've considered:
* Redefining all the hardcoded commands in my custom GDB script - but let's be real here, that's a very ugly solution
* Configuring SWO in code from MCU side, which effectively overwrites any config set by GDB - a feasible workaround, but I would really like to avoid explicitly adding config code, or additional set of ifdefs to exclude the code from production builds. Configuring this in GDB is much more flexible

**Additional context**
Based in discussion in #1102

haneefdm, 2025-04-02:
I already did this for openocd, which is why that gdb-script even exists. When Stefan contributed his STLink + SWO implementation, he must have used a very old implementation as a template written by the original author many years ago...when there was no  STLink gdb server.

https://github.com/Marus/cortex-debug/blob/92ad7ce850fb1fdb774e7371f87ff7109fe9d225/src/openocd.ts#L196

I think the original author had bigger plans for SWO/ITM -- like enabling statistical profiling, etc.

This same change is needed for pyocd and stutil but I don't have time to test all of these.

I can make these changes for STLink and create a pre-pre-release. Do you have time to test it?

haneefdm, 2025-04-02:
Note that are refences to bunch of BASE registers in the gdb script. Those can be different from the ARM defaults. ARM only recommends those addresses. Some vendors use their own (sometimes for no reason). I worked for one of them and we had two processors doing things in totally non-standard ways but in compliance with ARM spec. Compliance is one thing, but doing it in non-standard ways (when you don't have to) is totally another. ARM gives everyone plenty of rope to hang themselves.

In one case, the gdb script would have zero effect or cause bus fault because the clock to the debug blocks had to be enabled in FW, In another there was no power to these blocks because in ultra low power chips, it has to be enabled manually in FW. This meant that only way to properly initialize SWO/ITM was via FW.

So, as much as people like to think SWO is something standard, it is not. Haven't even brought up multi-core devices and how SWO is dealt with there.

For the BASE registers, one way to get around it is to do a ROM table discovery, but we don't have a mechanism to do that and while all gdb-servers have this info, no way to get it out.

Feliksas, 2025-04-02:
> I can make these changes for STLink and create a pre-pre-release. Do you have time to test it?

Yes, certainly. As far as I understood, the main culprit was that `DBGMCU->CR` was not being setup properly to actually enable physical tracing pin, like so:
```c
DBGMCU->CR = DBGMCU_CR_TRACE_IOEN | DBGMCU_CR_DBG_STANDBY | DBGMCU_CR_DBG_STOP | DBGMCU_CR_DBG_SLEEP;
```
Basically, it is what commented-out line [here](https://github.com/Marus/cortex-debug/blob/master/support/swo-init.c#L21) does.

I have also prepared my own GDB init script, based on `gdb-swo.init`from the repo, but haven't had the chance to test it properly yet:

```
set $ITM_BASE    = 0xE0000000
set $DWT_BASE    = 0xE0001000
set $SCS_BASE    = 0xE000E000
set $SCB_BASE    = $SCS_BASE + 0xD00
set $DCB_BASE    = $SCB_BASE + 0xF0
set $TPI_BASE    = 0xE0040000
set $DBGMCU_BASE = $TPI_BASE + 0x2000

set $TRCENA      = (0x1  << 24)

set $TRACE_IOEN  = (0x1  <<  5)
set $DBG_STANDBY = (0x1  <<  2)
set $DBG_STOP    = (0x1  <<  1)
set $DBG_SLEEP   = (0x1  <<  0)

set $TXMODE_NRZ  = (0x1  <<  1) 

set $ITMENA      = (0x1  <<  0)
set $SYNCENA     = (0x1  <<  2)
set $DWTENA      = (0x1  <<  3)
set $ATB_ID      = (0x7F << 16)

define SWO_Init
    set language c

# Global enable all DWT and ITM features
 	set *($DCB_BASE + 0x0000C)  = $TRCENA

# DBGMCU->CR = DBGMCU_CR_TRACE_IOEN | DBGMCU_CR_DBG_STANDBY | DBGMCU_CR_DBG_STOP | DBGMCU_CR_DBG_SLEEP
    set *($DBGMCU_BASE + 0x0004) = $TRACE_IOEN | $DBG_STANDBY | $DBG_STOP | $ DBG_SLEEP

# TPI->SPPR Pin protocol register to NRZ
	set *($TPI_BASE + 0x000F0)  = $TXMODE_NRZ

# TPI->ACPR = Prescaler. Change the following if your Prescaler calculation is different if your input clock is different
	set *($TPI_BASE + 0x00010)  = ($cpuFreq / $swoFreq) - 1

# ITM->LAR To be able to access rest of the ITM
	set *($ITM_BASE + 0x00FB0)  = 0xC5ACCE55

# ITM->TCR = ITM_TCR_ITMENA_Msk | ITM_TCR_SYNCENA_Msk | ITM_TCR_DWTENA_Msk | ITM_TCR_TraceBusID_Ms
	set *($ITM_BASE + 0x00E80)  = $ITMENA | $SYNCENA | $DWTENA | $ATB_ID

# ITM->TPR = bit-mask calculated by Cortex-Debug. 
	set *($ITM_BASE + 0x00E40)  = $swoPortMask

# ITM->TER = bit-mask calculated by Cortex-Debug. You can set it to 0xFFFFFFFF if you like
	set *($ITM_BASE + 0x00E00)  = $swoPortMask

    set language auto
end

document SWO_Init
Usage: SWO_Init
Initializes relevant CoreSight component registers to enable SWO. It relies
on a few global variables $swoFreq, $cpuFreq and $swoPortMask
end
```

>Note that are refences to bunch of BASE registers in the gdb script. Those can be different from the ARM defaults. ARM only recommends those addresses. Some vendors use their own (sometimes for no reason).

Thankfully, it looks like those base addresses are the same in case of STM32, only the DBGMCU base and associated logic was missing, so this probably can be made optional just for ST-Link case, though I'm not sure if that is gonna be universally applicable for everyone.

haneefdm, 2025-04-14:
We cannot put your specific DBGMCU changes as it is far from generic even for ST devices. Thanks to your reporting this issue, we have use the simpler implementation we had for openocd so you can easily override it using your own gdb-init file.

Since I cannot do anything more I am closing this. You can download the latest build with the simplified SWO_Init for ST now

https://github.com/Marus/cortex-debug/releases

closed 2025-04-14 by haneefdm