Marus/cortex-debug #1111: Destruction hook for advanced decoders

yueduz, 2025-04-03:
I have written an advanced decoder myself to forward data to TCP ports But I can't find a way to destroy the TCP connection He created multiple connections when I repeatedly started debugging These connections will persist unless I restart VSOCDE
```js
var net = require('net');

var client;
var avd = {};
avd.default = function () {
    this.init = function (config, displayOutput, graphData) {
        this.displayOutput = displayOutput;
        this.graphData = graphData;
        client = net.connect({ port: 8080 }, function () {
            console.log('连接到服务器！');
            this.displayOutput('连接到服务器！')
        });

    };
    this.lostSynchronization = function () {
        console.log('lostSynchronization');
    };
    this.synchronized = function () {
        console.log('synchronized');
    };
    this.softwareEvent = function (port, data) {
        client.write(data); // 发送数据
    };
    this.typeName = function () {
        return "自定义Name"
    };
    this.outputLabel = function () {
        return '自定义Label';
    };
}
module.exports = avd;
```

haneefdm, 2025-04-03:
Good point. I will look into it next week

haneefdm, 2025-04-08:
I will be adding a new optional method called `dispose()` that will be called at the end of a session. I don't have a good way of testing it. Please try it out after I make a private release. Watch this page for a release after today.

https://github.com/Marus/cortex-debug/releases

I am hoping that our own dispose is being called properly (and every time) so it can be forwarded to you.

haneefdm, 2025-04-13:
Can you test and give feedback with the following build

https://github.com/Marus/cortex-debug/releases

See the change to the interface...it should be compatible with any old implementations as well

https://github.com/Marus/cortex-debug/blob/a3a5747be7e4af8181da83bbb8de1f5f64089ab1/src/frontend/swo/common.ts#L102

yueduz, 2025-04-14:
I kept the code unchanged, but now I have:
```
Error Initializing Advanced Decoder: TypeError: Cannot read properties of undefined (reading 'append')
```

haneefdm, 2025-04-14:
Do you have any more of a stack trace or just that error? Looks like the constructor for your decoder is failing, Are you able to put a breakpoint either in your decoder constructor or its init function?

Does the error go away if you have a dispose() method?

yueduz, 2025-04-14:
```
    this.dispose = function () {
        client.end(); 
        // this.displayOutput('dispose')
        console.log('dispose');

    };
```
disposal() is not executed.

If I call "this.displayOutput('Name')" in some other method I get:
```
error initializing advanced decoder: type error: cannot read properties of undefined (Reading append)
```

haneefdm, 2025-04-14:
dispose will not be called because it could not even load/create your object. The error is happening while we are initializing your decoder.

haneefdm, 2025-04-14:
Can you attach your launch.json?

yueduz, 2025-04-14:
```json
    {
      "name": "Jlink",
      "type": "cortex-debug",
      "request": "launch",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/build/debug/STM32WLE5CBUx_switch.elf",
      "servertype": "jlink",
      "device": "STM32WLE5CB",
      "interface": "swd",
      "showDevDebugOutput": "raw",
      "svdFile": "${workspaceRoot}/STM32WLE5_CM4.svd",
      "preLaunchTask": "build Debug", //先运行make任务
      "serverArgs": ["-nologtofile", "-s", "-vd", "-speed", "24000"],
      "rttConfig": {
        "enabled": true,
        "address": "auto",
        "decoders": [
          {
            "port": 0,
            "type": "console"
          },
          {
            "port": 0,

            "type": "advanced",
            "decoder": "tool/rtt_forward.js",
            "ports": [0]  // Specify which SWO ports to forward
        }
        ]
      }
    }
```

The error occurred because I called "this. displayOutput()" in "typeName()"

yueduz, 2025-04-14:
If I don't call 'this. showOutput()', the decoder can load normally, but dispose() is not called

haneefdm, 2025-04-14:
Yes, you cannot use {displayOutput, graphData} until the initialization is complete. They may not be ready.

Same goes for dispose(...) you cannot use our stuff in your dispose as they have already been partially destroyed. May or may not work and no guarantees. Dispose is for you to clean up your things.

Now, I have to figure out why your dispose is not being called

haneefdm, 2025-04-14:
Can you attach your whole decoder file again? Including the new dispose function. Because dispose is optional, you will not get errors if you/me have a typo.

yueduz, 2025-04-14:
```js
var net = require('net');

var client = null;
var avd = {};
avd.default = function () {
    this.init = function (config, displayOutput, graphData) {
        this.displayOutput = displayOutput;
        this.graphData = graphData;
        client = net.connect({ port: 8080 }, function () {
            // this.displayOutput('连接到服务器！')
        });
    };
    this.lostSynchronization = function () {
        console.log('lostSynchronization');
    };
    this.synchronized = function () {
        console.log('synchronized');
    };
    this.softwareEvent = function (port, data) {
        client.write(data); // 发送数据

    };
    this.typeName = function () {
        this.displayOutput('自定义Name')

        return "自定义Name"
    };
    this.outputLabel = function () {
        console.log(this);

        return '自定义Label';
    };
    this.dispose = function () {
        client.end(); 
        // this.displayOutput('dispose')
        console.log('dispose');

    };

}
module.exports = avd;
```

haneefdm, 2025-04-15:
Can you try the latest build. Make sure you read the CHANGELOG

https://github.com/Marus/cortex-debug/releases

Please let me know if it works or not.

I am not sure how you run/test your decoder but I use the VSCode debugger with cotex-debug so I can see the console.log() output. You can probably launch VSCode from a terminal as well.

Anyways, I used the following decoder (modified from yours)

```Javascript
const fs = require('fs');
const os = require('os');
const path = require('node:path');

var avd = {};
avd.default = function () {
    this.init = function (config, displayOutput, graphData) {
        this.displayOutput = displayOutput;
        this.graphData = graphData;
        this.fName = path.join(os.tmpdir(), 'test-adv-decoder.txt');
        try {
            // console.log(`Trying to open '${this.fName}' for writing`)
            this.stream = fs.createWriteStream(this.fName);
        } catch (e) {
            console.log(`Failed to open '${this.fName}' for writing`)
            this.stream = undefined;
        }
    };
    this.lostSynchronization = function () {
        console.log('lostSynchronization');
    };
    this.synchronized = function () {
        console.log('synchronized');
    };
    this.softwareEvent = function (port, data) {
        const msg = data.toString();
        if (this.stream) {
            this.stream.write(msg);
        }
        this.displayOutput(msg);
    };
    this.typeName = function () {
        return "自定义Name"
    };
    this.outputLabel = function () {
        return '自定义Label';
    };
    this.dispose = function () {
        if (this.stream) {
            this.stream.end(() => {
                console.log(`Write stream ${this.fName} closed`)
            });
        }
    };
}
module.exports = avd;

```

haneefdm, 2025-04-15:
Also, your init(...) function must not fail. If that fails then a decoder was never created and dispose will not be called.

yueduz, 2025-04-15:
In the decoder you modified, the content output by RTT has been correctly written to test adv-decoder.txt, so the decoder has been initialized correctly

yueduz, 2025-04-15:
I didn't see console. log (` Write stream ${this. fName} closed `) in the dispose() function when I stopped debugging

haneefdm, 2025-04-15:
Note that I did not test this with RTT which may be the issue. I only tested with SWO. I have to create an RTT example.

haneefdm, 2025-04-15:
Yes, that is the way it has always been. That is why I moved all the built-in stuff to the TERMINAL window. I did not write this code and did not want to touch it for compatibility. Also, when in the TERMINAL window, it can be bidirectional. OUTPUT panels are unidirectional. RTT is inherently bidirectional and so is semi-hosting

There are two things I could do.

1. Re-use anything with the same name but continue to use the OUTPUT Panel. Least amount of work
2. Move it to the TERMINAL tab where everything else resides. This also gives bidirectional capability. Lot more work.

What would you prefer? And, no, I don't want to create an option to do it both ways :-)

I am leaning towards option 2, but I need some free time. Remember, I do this in my free time

yueduz, 2025-04-15:
Sorry, I just tested the new version. dispose() has been fixed and the output window is no longer duplicated.

closed 2025-04-15 by yueduz