Marus/cortex-debug #1112: Live Watch - Variables List storage place

sme2rt, 2025-04-03:
Hi all,

first of all I would like to thanks the people who have developed / are developing this extension - it is really great, especially the live watch feature. 

Regarding this feature I have a very simple question: where is the list of the variables shown being stored? The background of my question is, that I would like to apply the same "configuration" (=variable list) to another branch or project without having to add the variables manually each time.
When I re-open a project the variables are still here so it definitely gets stored somewhere but somehow cannot figure out where ;-)

Would be great if someone could help me.

haneefdm, 2025-04-03:
The list of watched variables are stored in your workspace state data (kinda like metadata). This is generally not accessible. I don't know where VSCode stores these. I think your breakpoints, watch variables, etc. are also stored there. Googling, I found this thread

https://stackoverflow.com/questions/57760451/where-does-visual-studio-code-store-information-about-enabled-extensions-per-wor
https://stackoverflow.com/questions/57767800/where-does-breakpoints-stores

It is designed so that you are not supposed to edit it. It is considered as transient data. Note that this is not documented. MS may change their mind and store it elsewhere or in a different format.

sme2rt, 2025-04-04:
Thanks a lot for the fast answer and support. 
I will have a look at it. 
Mickael

closed 2025-04-04 by sme2rt