Marus/cortex-debug #1115: Breakpoint not working

WGR7, 2025-04-10:
**Describe the bug**
Hi,

I'm facing a problem. The thebugger connect, but can't stop on breakpoints. I can pause and see the current line of code executed, but can't step in/out.

Probe: pico debug probe https://github.com/raspberrypi/debugprobe The same probe worked on a old laptop to debug. The same probe can read and write sucessfuly to mcu through pyocd

**To Reproduce**
Only starts a debug session on attached p

**Expected behavior**
Stop on main function when session debug starts

**Environment:**
OS:
Distributor ID:	Linuxmint
Description:	Linux Mint 22.1
Release:	22.1
Codename:	xia

Target: Py32F003x6

GCC Toolchain: 14.2.rel1-x86_64-arm-none-eabi
arm-none-eabi-gcc (Arm GNU Toolchain 14.2.Rel1 (Build arm-14.52)) 14.2.1 20241119
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 
VSCODE: 
1.99.2
4949701c880d4bdb949e3c0e6b400288da7f474b
x64

**Please include `launch.json`**
Included in attached file

```
{ "version": "0.2.0", "configurations": [

    {
        "armToolchainPath": "/opt/gcc-arm/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin",
        "toolchainPrefix": "arm-none-eabi",
        "name": "CortexDebug",
        "cwd": "${workspaceFolder}",
        "executable": "${workspaceFolder}/Build/app.elf",
        "request": "launch",        // can be launch or attach
        "type": "cortex-debug",
        "runToEntryPoint": "main", // "main" or other function name. runToMain is deprecated
        "servertype": "pyocd",  // jlink, openocd, pyocd, pe and stutil
        "device": "PY32F003x6",
        "interface": "swd",
        "preLaunchTask": "build",  // Set this to run a task from tasks.json before starting a debug session
        // "preLaunchCommands": ["Build all"], // Uncomment this if not using preLaunchTask
        "svdFile": "${workspaceFolder}/Misc/SVD/py32f003xx.svd",  // svd for this part number
        "showDevDebugOutput": "raw", // parsed, raw, vscode:vscode log and raw
        "swoConfig":
        {
            "enabled": true,
            "cpuFrequency": 8000000, // Target CPU frequency in Hz
            "swoFrequency":  4000000,
            "source": "probe", // either be “probe” to get directly from the debug probe, 
                               // or a serial port device to use a serial port external to the debug probe.
            "decoders":
            [
                {
                    "label": "ITM port 0 output",
                    "type": "console",
                    "port": 0,
                    "showOnStartup": true,
                    "encoding": "ascii"
[... 284 lines cut ...]
To client: {"seq":0,"type":"response","request_seq":14,"command":"threads","success":true,"body":{"threads":[{"id":1,"name":"Thread"}]}}
From client: stackTrace({"threadId":1,"startFrame":0,"levels":20})
31-stack-list-frames --thread 1 0 19
-> 31^done,stack=[frame={level="0",addr="0x080010d8",func="HAL_Delay",file="Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal.c",fullname="/home/owner/Teste/Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal.c",line="380",arch="armv6s-m"},frame={level="1",addr="0x08000990",func="main",file="User/main.c",fullname="/home/owner/Teste/User/main.c",line="21",arch="armv6s-m"}]
To client: {"seq":0,"type":"response","request_seq":15,"command":"stackTrace","success":true,"body":{"stackFrames":[{"id":4096,"source":{"name":"Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal.c","path":"/home/owner/Teste/Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal.c","sourceReference":0},"line":380,"column":0,"name":"HAL_Delay@0x080010d8","instructionPointerReference":"0x080010d8"},{"id":4097,"source":{"name":"User/main.c","path":"/home/owner/Teste/User/main.c","sourceReference":0},"line":21,"column":0,"name":"main@0x08000990","instructionPointerReference":"0x08000990"}]}}
From client: scopes({"frameId":4096})
32-stack-info-frame --thread 1 --frame 0
-> 32^done,frame={level="0",addr="0x080010d8",func="HAL_Delay",file="Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal.c",fullname="/home/owner/Teste/Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal.c",line="380",arch="armv6s-m"}
To client: {"seq":0,"type":"response","request_seq":16,"command":"scopes","success":true,"body":{"scopes":[{"name":"Local","variablesReference":4096,"expensive":false},{"name":"Global","variablesReference":4294967295,"expensive":false},{"name":"Static: ./Libraries/PY32F0xx_HAL_Driver/Src/py32f0xx_hal.c","variablesReference":1052672,"expensive":false},{"name":"Registers","variablesReference":2101248,"expensive":false}]}}
From client: variables({"variablesReference":4096})
33-stack-list-variables --thread 1 --frame 0 --simple-values
-> 33^done,variables=[{name="Delay",arg="1",type="uint32_t",value="0x3e8"},{name="Delay@entry",arg="1",type="uint32_t",value="0x3e8"},{name="tickstart",type="uint32_t",value="0x177d"},{name="wait",type="uint32_t",value="0x3e9"}]
34-var-create --thread 1 --frame 0 var_Delay_4096 @ "Delay"
-> 34^done,name="var_Delay_4096",numchild="0",value="0x3e8",type="uint32_t",has_more="0"
35-var-create --thread 1 --frame 0 var_Delay@entry_4096 @ "Delay@entry"
-> 35^done,name="var_Delay@entry_4096",numchild="0",value="0x3e8",type="uint32_t",has_more="0"
36-var-create --thread 1 --frame 0 var_tickstart_4096 @ "tickstart"
-> 36^done,name="var_tickstart_4096",numchild="0",value="0x177d",type="uint32_t",has_more="0"
37-var-create --thread 1 --frame 0 var_wait_4096 @ "wait"
-> 37^done,name="var_wait_4096",numchild="0",value="0x3e9",type="uint32_t",has_more="0"
To client: {"seq":0,"type":"response","request_seq":17,"command":"variables","success":true,"body":{"variables":[{"name":"Delay","evaluateName":"Delay","value":"0x3e8","type":"uint32_t Delay;\ndec: 1000\nhex: 0x000003e8\noct: 000000001750\nbin: 00000000 00000000 00000011 11101000","presentationHint":{},"variablesReference":0},{"name":"Delay@entry","evaluateName":"Delay@entry","value":"0x3e8","type":"uint32_t Delay@entry;\ndec: 1000\nhex: 0x000003e8\noct: 000000001750\nbin: 00000000 00000000 00000011 11101000","presentationHint":{},"variablesReference":0},{"name":"tickstart","evaluateName":"tickstart","value":"0x177d","type":"uint32_t tickstart;\ndec: 6013\nhex: 0x0000177d\noct: 000000013575\nbin: 00000000 00000000 00010111 01111101","presentationHint":{},"variablesReference":0},{"name":"wait","evaluateName":"wait","value":"0x3e9","type":"uint32_t wait;\ndec: 1001\nhex: 0x000003e9\noct: 000000001751\nbin: 00000000 00000000 00000011 11101001","presentationHint":{},"variablesReference":0}]}}

[Teste.zip](https://github.com/user-attachments/files/19695217/Teste.zip)

```

haneefdm, 2025-04-14:
I don't see VSCode request ANY breakpoints. We do not control what VSCode GUI does. We only respond to requests made to the GUI and I see none.

Silly question but do you have any breakpoints? Enabled?

```
From client: setFunctionBreakpoints({"breakpoints":[]})
To client: {"seq":0,"type":"response","request_seq":5,"command":"setFunctionBreakpoints","success":true,"body":{"breakpoints":[]}}
From client: setInstructionBreakpoints({"breakpoints":[]})
From client: setDataBreakpoints({"breakpoints":[]})
```

Normally, the requests should look like these (I removed the responses). I had just one breakpoint.

```
From client: setBreakpoints({"source":{"name":"main.c","path":"/Users/hdm/mtb3.1/HAL_FreeRTOS-POSIX_Demo/source/main.c"},"lines":[95],"breakpoints":[{"line":95}],"sourceModified":false})
From client: setFunctionBreakpoints({"breakpoints":[]})
From client: setInstructionBreakpoints({"breakpoints":[]})
From client: setDataBreakpoints({"breakpoints":[]})
```

You can also try the latest build done with a more recent version of VSCode.

https://github.com/Marus/cortex-debug/releases

haneefdm, 2025-04-16:
Well, we don't know any more than you do. GDB reports setting the breakpoints where it thought it should set them. And, without errors.

If your program never reached those lines or those lines got optimized out, then it won't hit those breakpoints. It is now between you, the compiler and the firmware. Use tools like objdump to see the machine code (assembly).

Are you seeing your print statements? You can also add the breakpoint instruction in your while loop to force a halt.
```
__asm__("BKPT");
```
https://developer.arm.com/documentation/dui0552/a/the-cortex-m3-instruction-set/miscellaneous-instructions/bkpt

We are only a graphics front end. We don't control what your MCU does. Neither does gdb other than setting the breakpoints hope that your processor will do the right thing.

```
-> 15^done
16-break-insert "/home/user/Teste/User/main.c:21"
-> 16^done,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x08000990",func="main",file="User/main.c",fullname="/home/user/Teste/User/main.c",line="21",thread-groups=["i1"],times="0",original-location="/home/user/Teste/User/main.c:21"}
17-break-insert "/home/user/Teste/User/main.c:23"
-> 17^done,bkpt={number="2",type="breakpoint",disp="keep",enabled="y",addr="0x080009a0",func="main",file="User/main.c",fullname="/home/user/Teste/User/main.c",line="23",thread-groups=["i1"],times="0",original-location="/home/user/Teste/User/main.c:23"}
```

Use purely command-line tools to see if we are doing something wrong. If so, let us know. As far as I can tell, everyone has done what they are supposed to do. Since we don't have your Hardware, we can't really move forward. I am closing this issue for now....unless you have more information.

closed 2025-04-16 by haneefdm