Marus/cortex-debug #1118: Feature Request: Support for Black Magic Debug App

ztihon, 2025-04-20:
Hello,

I'd like to request support for the Black Magic Debug App for cortex-debug.

Black Magic Debug App aka blackmagic-bmda is the Black Magic Debug compiled as a standalone application that runs on the host PC rather than as a firmware on the debug probe. This architecture allows the tool to interface with a variety of debug probes, including:

-Black Magic Probe (via USB-serial interface)

-ST-Link v2, v2.1, and v3 (with recent firmware)

-CMSIS-DAP compatible probes

-J-Link probes

-FTDI MPSSE-based probes

This usage model is similar to OpenOCD, acting as a "server in the middle" between GDB and the hardware.

More details can be found in the [here](https://github.com/blackmagic-debug/blackmagic?tab=readme-ov-file#black-magic-debug-app).

Motivation

I’ve previously used OpenOCD with cortex-debug, but its slow release cycle has delayed support for newer MCUs like the STM32H5 series. In contrast, Black Magic Debug provides timely updates and already supports the STM32H5, making it a strong alternative to OpenOCD.

Furthermore, prebuilt binaries for bmda are readily available for Linux, Windows, and macOS, which significantly lowers the barrier to entry: [Releases](https://github.com/blackmagic-debug/blackmagic/releases)

Proposed Implementation

Since bmda behaves similarly to OpenOCD—where you launch the server in one terminal and connect via GDB in another—it might be relatively straightforward to add support.
Maybe a new "servertype" could be added to cortex-debug as "bmda" so the existing "bmp" could be left as is ? With "bmda" you would not give the serial port as an argument but the path to the server.

Thank you for considering this request! I’d be happy to help test or provide feedback if needed.

haneefdm, 2025-04-20:
Someone familiar with Typescript and BM Debug App has to do this work. You can do this. I mean looking for volunteers that can create this support and maintain it going forward.

I don't own any BMPs HW setup either. Already busy with many other things.

> I’ve previously used OpenOCD with cortex-debug, but its slow release cycle has delayed support for newer MCUs like the STM32H5 series

Now, this is interesting and hard to believe. OpenOCD should be used from ST's SW distribution and NOT from releases available on the internet. There is also STLink. I am pretty sure that these two mechanisms are supported by ST from day one of a release of a device.

Also, why not make this request to the BMP folks?

ztihon, 2025-04-20:
Yes, ST's fork of OpenOCD does support the H5.
However, using it requires installing ST's CubeIDE, which is a bit of a pain—it’s a bulky IDE and even needs an ST account just to download.

If I have to go through all that just to use OpenOCD, it kind of defeats the whole point of using VS Code with cortex-debug: having a lightweight and customizable setup for MCU development.

ztihon, 2025-04-20:
And it's the same thing with ST-Link GDB Server, it is not available as a standalone package but part of CubeIDE.

haneefdm, 2025-04-20:
I'm sorry you have to download it. I do it once per release and just copy the OpenOCD directory. They might put it into the CubeCLT package, which may be helpful. I also like to use ST supported/validated tools. STLink server is available as part of their CLT package. I think they are moving more and more stuff into this package.

https://www.st.com/en/development-tools/stm32cubeclt.html

This is also true with other device manufacturers because OpenOCD is so slow in their release cycle that they have to fork OpenOCD.

> If I have to go through all that just to use OpenOCD, it kind of defeats the whole point of using VS Code with cortex-debug: having a lightweight and customizable setup for MCU development.

Where you get tools has very little to do with what debugger/editor I use. GNU Tools are super bulky as well, but I have to use them.

haneefdm, 2025-04-20:
I marked it as 'help needed'. I will keep this issue alive for a while and close it if there are no volunteers.

As you can tell, I don't agree with your justification. But lets not argue about that.

ztihon, 2025-04-20:
> I'm sorry you have to download it. I do it once per release and just copy the OpenOCD directory. They might put it into the CubeCLT package, which may be helpful. I also like to use ST [...]

It is not really a personal problem to me. I have used ST's tools for years and I actually use CubeIDE in the board bring-up phase where I just verify the HW and don't do any real development work.

However, it would be beneficial to have an easily accessible, open-source tool set for debugging MCUs, one that gets support for the latest MCU models very fast and supports variety of debug probes. Cortex-debug nearly fulfills this, and adding support for BMDA would complete it.

But maybe I'll contact the BMP folks and see if they have interest in this. I have no experience in Typescript so I can't really help with this.