Marus/cortex-debug #1119: pherepherial view don't open

bobovich, 2025-04-20:
Hello, you can help  me I have lost all chance to find an answer. How open pherepherial viewer? 

[image]

haneefdm, 2025-04-20:
Wrong Repo. https://github.com/mcu-debug/peripheral-viewer

Make sure it is installed. See if you hid the view by mistake If you installed cortex-debug manually then it will not pull in other related extensions...

bobovich, 2025-04-20:
Yes, I have this extenension, all possible views on screen, how right configure pherepherial view extension?

[image]

haneefdm, 2025-04-20:
> Yes, I have this extenension

What do you mean "this" extension? Peripheral viewer is its own extension and it can work with multiple debuggers...

Is it installed though? Make sure that it is installed and not disabled (can happen accidentally)

[image]

Also check the OUTPUT tab and enable the debug flag...here are the settings

[image]

How did you even install cortex-debug? From inside VSCode or using some command line tool>

bobovich, 2025-04-20:
debug-tracker-vscode and Peripheral Viewer installed and configured but cortex peripheral not available
[image]
[image]
[image]
[image]
[image]

haneefdm, 2025-04-21:
So, did you increase the debug level and see what is going on? I even pasted the setting in my previous post.

I don't know what is going one. Perhaps VSCode install is corrupted. Zero ideas

bobovich, 2025-04-21:
here is the output of DEBUG CONSOLE, I noticed that it stopped working when I installed it PlatformIO

> Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from C:/Program Files/xpack-arm-none-eabi-gcc/bin/arm-none-eabi-objdump.exe --syms -C -h -w D:/yd/swp/sl_fw/sk/build/Debug/SafeLink.elf
Reading symbols from c:/program files/xpack-arm-none-eabi-gcc/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p D:/yd/swp/sl_fw/sk/build/Debug/SafeLink.elf
Launching GDB: "C:\\Program Files\\xpack-arm-none-eabi-gcc\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: "C:\\openocd\\bin\\openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "D:/yd/swp/sl_fw/sk" -f "c:/Users/Ivan/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl" -f interface/cmsis-dap.cfg -f target/geehy/apm32f4x.cfg -c CDLiveWatchSetup
    Please check TERMINAL tab (gdb-server) for output from C:\openocd\bin\openocd.exe
Finished reading symbols from objdump: Time: 486 ms
Finished reading symbols from nm: Time: 417 ms
Output radix now set to decimal 10, hex a, octal 12.
Input radix now set to decimal 10, hex a, octal 12.
0x08001074 in SysTick_Handler () at ./FreeRTOS/portable/GCC/ARM_CM4F/port.c:512
512	            portNVIC_INT_CTRL_REG = portNVIC_PENDSVSET_BIT;
Program stopped, probably due to a reset and/or halt issued by debugger
[apm32f4x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080057c8 msp: 0x20020000
[apm32f4x.cpu] halted due to breakpoint, current mode: Thread
xPSR: 0x01000000 pc: 0x080057c8 msp: 0x20020000
LiveGDB: Output radix now set to decimal 10, hex a, octal 12.
LiveGDB: Input radix now set to decimal 10, hex a, octal 12.
LiveGDB: Reset_Handler () at ./startup_apm32f40x.S:60
LiveGDB: 60	  ldr   sp, =_end_stack
LiveGDB: Program stopped, probably due to a reset and/or halt issued by debugger
Started live-monitor-gdb session
warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
Note: automatically using hardware breakpoints for read-only addresses.
Temporary breakpoint 1, main () at ./main.c:36
36	{

bobovich, 2025-04-21:
Thankyou for answer, I'm clean VS Code, reinstall extensions and reconfigure. It's take effect! XPeripherial come back!

[image]

closed 2025-04-21 by bobovich