Marus/cortex-debug #1126: Jlink multi core chained debug issue

salfe, 2025-05-14:
When i debug multicore with cortex-debug + jlink, i found only core0 can be attach&program successfully, core1 always **fail**

**Expected behavior**

core0 and core1 can be attach & program successfully

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension) [1.12.1]
 - OS: windows
 - GDB Version: [10.3-2021.10) 10.2.90.20210621-git]
 - Compiler Toolchain Version: [e.g. arn-none-eabi V 10.3.1]

**Please include `launch.json`**
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "name": "core0_debug",
      "type": "cortex-debug",
      "request": "launch",
      "runToEntryPoint": "main",
      "servertype": "jlink",
      "device": "XXXX_M7_0",
      "gdbPath": "C:\\Tools\\gcc_10\\bin\\arm-none-eabi-gdb.exe",
      "armToolchainPath": "C:\\Tools\\gcc_10\\bin",
      "cwd": "C:\\Program Files\\SEGGER\\JLink",
      "executable": "\output\\cm7_0\\example.elf",
      "svdPath": "xxxx.svd",
      "numberOfProcessors": 2,
      "targetProcessor": 0,
      "showDevDebugOutput": "raw",
      "preLaunchCommands": [
        "monitor halt",
        "set {int}0x40001000 = 0xA5A5A500",
        "x/4xw 0x40001000"
      ],
      "chainedConfigurations": {
        "enabled": true,
        "delayMs": 5000,
        "waitOnEvent": "postInit",
        "detached": true,
        "lifecycleManagedByParent": true,
        "launches": [
          {
            "name": "core1_debug",
            "folder": "${workspaceFolder}"
          }
        ]
      }
    },
    {
      "name": "core1_debug",
      "type": "cortex-debug",
      "request": "launch",
      "runToEntryPoint": "Reset_Handler",
      "servertype": "jlink",
      "device": "XXXX_M7_1",
      "gdbPath": "C:\\Tools\\gcc_10\\bin\\arm-none-eabi-gdb.exe",
      "armToolchainPath": "C:\\Tools\\gcc_10\\bin",
      "cwd": "C:\\Program Files\\SEGGER\\JLink",
      "executable": "\output\\cm7_1\\example.elf",
      "numberOfProcessors": 2,
      "showDevDebugOutput": "raw",
      "targetProcessor": 1
    }
  ]
}
```

**Attach text from `Debug Console`**

```
LOG FROM core0_debug:

Launching gdb-server: JLinkGDBServerCL.exe -singlerun -nogui -if swd -port 50000 -swoport 50002 -telnetport 50004 -device XC27X_M7_0
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServerCL.exe
Finished reading symbols from objdump: Time: 77 ms
Finished reading symbols from nm: Time: 76 ms
C:\Tools\gcc_10\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source c:/Users/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source c:/Users/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-file-exec-and-symbols "cm7_0/example.elf"
-> 10^done
11-target-select extended-remote localhost:50000

LOG FROM core1_debug:

Launching gdb-server: JLinkGDBServerCL.exe -singlerun -nogui -if swd -port 50009 -swoport 50011 -telnetport 50013 -device XC27X_M7_1
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServerCL.exe
Finished reading symbols from objdump: Time: 75 ms
Finished reading symbols from nm: Time: 73 ms
C:\Tools\gcc_10\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source c:/Users/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source c:/Users/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-file-exec-and-symbols "output/cm7_1/example.elf"
-> 10^done
11-target-select extended-remote localhost:50010

```

From the log, i see
for core0, the jlink gdb server using port 50000 and also arm gdb.
but core1, the jlink gdb server using port 50009, but arm gdb using 50010

i guess this is the root cause, but not sure, please help on this. Thank you very much.

haneefdm, 2025-05-14:
I agree, that appears wrong.

> 11-target-select extended-remote localhost:50010

That should be 50009. In JLink, every core is treated as a separate MCU. It is the only server that does that. As a workaround, can you try the following? Set `"targetProcessor": 0`, for the second core as well.  This is what we should be doing internally -- ignore `targetProcessor` when `"detached": true`

haneefdm, 2025-05-14:
Please report back with your findings.

salfe, 2025-05-14:
> I agree, that appears wrong.
>
>
> That should be 50009. In JLink, every core is treated as a separate MCU. It is the only server that does that. As a workaround, can you try the following? Set [...]

Set `"targetProcessor": 0` works, but there is another issue i am not sure if it is the side effect,
after core1 attach & program done, core1 **go** directly, but core0 still halt,

 Here is the core1_debug log:

Launching gdb-server: JLinkGDBServerCL.exe -singlerun -nogui -if swd -port 50009 -swoport 50011 -telnetport 50013 -device XXXX_M7_1
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServerCL.exe
Finished reading symbols from objdump: Time: 72 ms
Finished reading symbols from nm: Time: 69 ms
C:\Tools\gcc_10\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source c:/Users/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source c:/Users/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-file-exec-and-symbols "C:/output/cm7_1/example.elf"
-> 10^done
11-target-select extended-remote localhost:50009
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"0x10002d96 in ?? ()\n"
0x10002d96 in ?? ()
-> *stopped,frame={addr="0x10002d96",func="??",args=[],arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 11^connected
12-interpreter-exec console "monitor halt"
-> 12^done
13-interpreter-exec console "monitor reset"
-> @"Resetting target\r\n"
Resetting target
-> 13^done
14-target-download
-> 14+download,{section=".flash",section-size="12276",total-size="2582571"}
-> 14+download,{section=".flash",section-sent="12276",section-size="12276",total-sent="12276",total-size="2582571"}
-> 14^done,address="0x10200400",load-size="12276",transfer-rate="49104000",write-rate="12276"
15-interpreter-exec console "monitor reset"
-> @"Resetting target\r\n"
Resetting target
-> 15^done
Returning dummy thread-id to workaround VSCode issue with pause button not working
Returning dummy stack frame to workaround VSCode issue with pause button not working: {"threadId":1,"startFrame":0,"levels":20}
16-break-insert -t --function Reset_Handler
-> 16^done,bkpt={number="1",type="breakpoint",disp="del",enabled="y",addr="0x10200400",at="<_start>",thread-groups=["i1"],times="0",original-location="-function Reset_Handler"}
17-exec-continue --all
-> 17^running
-> *running,thread-id="all"
mi2.status = running
18-exec-interrupt 
-> 18^done
-> ~"\nProgram"

Program
-> ~" received signal SIGTRAP, Trace/breakpoint trap.\n"
 received signal SIGTRAP, Trace/breakpoint trap.
-> ~"0x10003096 in ?? ()\n"
0x10003096 in ?? ()
-> *stopped,reason="signal-received",signal-name="SIGTRAP",signal-meaning="Trace/breakpoint trap",frame={addr="0x10003096",func="??",args=[],arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
19-thread-list-ids
-> 19^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
20-thread-info 1
-> 20^done,threads=[{id="1",target-id="Thread 57005",frame={level="0",addr="0x10003096",func="??",args=[],arch="armv7e-m"},state="stopped"}]
21-stack-list-frames --thread 1 0 19
-> 21^done,stack=[frame={level="0",addr="0x10003096",func="??",arch="armv7e-m"},frame={level="1",addr="0x10002f12",func="??",arch="armv7e-m"}]
22-stack-info-frame --thread 1 --frame 0
-> 22^done,frame={level="0",addr="0x10003096",func="??",arch="armv7e-m"}
23-stack-list-variables --thread 1 --frame 0 --simple-values
-> 23^done,variables=[]
24-stack-info-frame --thread 1 --frame 1
-> 24^done,frame={level="1",addr="0x10002f12",func="??",arch="armv7e-m"}
25-stack-list-variables --thread 1 --frame 1 --simple-values
-> 25^done,variables=[]

haneefdm, 2025-05-14:
Do you mind temporarily switching to the latest build? Even if something is to be fixed, I would like it to be based on the current code-base.

> but there is another issue i am not sure if it is the side effect, after core1 attach & program done, core1 go directly, but core0 still halt,

From my experience, I would program all cores with the first configuration and disable programming from the others. Depends on your memory layout and what is being shared. I also don't know what you are doing in your `preLaunchCommands` without looking into the huge ARM manual.

I am not sure if your description, logs and launch.json all matches what I am looking at. I am a bit lost. Am I looking at the complete text from the Debug Console for the first core? Because it is supposed to do many things after the `11-target-select extended-remote localhost:50000`

salfe, 2025-05-14:
where is latest build?

haneefdm, 2025-05-14:
Oops...forgot to paste link

https://github.com/Marus/cortex-debug/releases

This is where all of very early builds go

salfe, 2025-05-15:
> Do you mind temporarily switching to the latest build? Even if something is to be fixed, I would like it to be based on the current code-base.
>
>
> From my experience, I would program all cores with [...]

I tried the latest build [V1.13.0-pre6](https://github.com/Marus/cortex-debug/releases/tag/v1.13.0-pre6), the behavior is same,

core0 halt, core1 go directly

Here is the detail log:

core0_debug:
```
Cortex-Debug: VSCode debugger extension version 1.13.0-pre6 git(48913dc+dirty). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "core0_debug",
    "type": "cortex-debug",
    "request": "launch",
    "runToEntryPoint": "main",
    "servertype": "jlink",
    "device": "XXXX_M7_0",
    "gdbPath": "C:\\Tools\\gcc_10\\bin\\arm-none-eabi-gdb.exe",
    "armToolchainPath": "C:\\Tools\\gcc_10\\bin",
    "cwd": "C:\\Program Files\\SEGGER\\JLink",
    "executable": "C:/cm7_0/example.elf",
    "svdPath": "C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.4\\arm\\config\\debugger\\XHSC\\XXXX.svd",
    "numberOfProcessors": 2,
    "targetProcessor": 0,
    "showDevDebugOutput": "raw",
    "preLaunchCommands": [
        "monitor halt",
        "set {int}0x40001000 = 0xA5A5A500",
        "x/4xw 0x40001000"
    ],
    "chainedConfigurations": {
        "enabled": true,
        "delayMs": 5000,
        "waitOnEvent": "postInit",
        "detached": true,
        "lifecycleManagedByParent": true,
        "launches": [
            {
                "name": "core1_debug",
                "folder": "C:/Users/.vscode/extensions/xhsc.xhsc-xstudio-extension-1.0.2/resources/workspace",
                "enabled": true,
                "delayMs": 5000,
                "detached": true,
                "waitOnEvent": "postInit",
                "lifecycleManagedByParent": true
            }
        ]
    },
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
[... 428 lines cut ...]
      "gdbPath": "C:\\Tools\\gcc_10\\bin\\arm-none-eabi-gdb.exe",
      "armToolchainPath": "C:\\Tools\\gcc_10\\bin",
      "cwd": "C:\\Program Files\\SEGGER\\JLink",
      "executable": "C:\\cm7_1\\example.elf",
      "numberOfProcessors": 2,
      "showDevDebugOutput": "raw",
      "targetProcessor": 0
    }
  ]
}
```

The prelaunch command is used to release core1 from reset state, so core1 can run flash loader

```
      "preLaunchCommands": [
        "monitor halt",
        "set {int}0x40001000 = 0xA5A5A500",
        "x/4xw 0x40001000"
      ],
```

> From my experience, I would program all cores with the first configuration and disable programming from the others. Depends on your memory layout and what is being shared. I also don't know what you are doing in your `preLaunchCommands` without looking into the huge ARM manual.

is there any example launch.json i can reference?

haneefdm, 2025-05-15:
> so are you going to fix this Jink multi core debug issue in later release?

Probably not. It is actually not a defect. Note that our SW is not JLink specific. There is a scenario where what it is doing is actually correct and intended. The chained configurations can also be used for multi-board/multi-mcu scenarios where each item can itself be multi-core. Instead, I will update the Wiki like we already do.

https://github.com/Marus/cortex-debug/wiki/Multi-core-debugging

I have to think some more because it might be a breaking change.

> is there any example launch.json i can reference?

Since there are literally millions of combinations on top of how people write their FW, I am not sure how I can create a sample launch.json. If you understand your device, your FW (bootloaders, security), gdb and your gdb-server, those are the requirements. I don't even own much hardware. I just have to imagine what you may be doing.

>Set "targetProcessor": 0 works, but there is another issue i am not sure if it is the side effect,
after core1 attach & program done, core1 go directly, but core0 still halt

But isn't this what you requested? core0 seems to have hit a breakpoint (due to `"runToEntryPoint": "main"`) and is halted there .... waiting for you to do a continue manually. You can remove `runToEntryPoint` or set it to an empty string.

I saw this in my email where you attached new logs for both core0 and core1.  don't see that now on github.

salfe, 2025-05-16:
.

> Probably not. It is actually not a defect. Note that our SW is not JLink specific. There is a scenario where what it is doing is actually correct and intended. The chained configurations can also be [...]

OK, understand, please update document for JLINK chain configuration. 

Thank you very much.

haneefdm, 2025-05-16:
You could have offered to edit the Wiki. It is a Wiki. Anyway, I just updated it.

closed 2025-05-16 by haneefdm