Marus/cortex-debug #1137: rtt advanced decoder (for defmt-print) opens new output windows at every run instead of reusing the old one

mfp20, 2025-06-09:
**Describe the bug**

Every debug session creates a new output item in VScode Output tab.

**To Reproduce**

1. Configure as seen in the configuration pasted below. 
2. Add the script
3. Start/stop the debug sessions to see multiple items in the output tab.

**Expected behavior**

Debug session re-use the same output item.

**Environment (please complete the following information):**

 - Cortex-Debug Version, see below
 - OS, Windows 11
 - GDB&Toolchain Version, GNU gdb (Arm GNU Toolchain 14.2.Rel1 (Build arm-14.52)) 15.2.90.20241130-git

**Please include `launch.json`**

The configuration can be seen below. I attach here the js script I'm using to send the output to defmt-print and, once decoded, output in VSCode Output tab's output item (named `rtt-defmt.js` in the launch.json configuration shown below).

```
import { spawn } from 'child_process';
import { resolve } from 'path';
//import vscode from 'vscode';

class DefmtDecoder {
  constructor() {
    this.config = {};
    this.displayOutput = () => {};
    this.graphData = () => {};
    this.process = null;
  }

  init(config, displayOutput, graphData) {
    this.config = config;
    this.displayOutput = displayOutput;
    this.graphData = graphData;

    this.cwd = config.config.cwd || process.cwd();
    if (!config.config.executable) {
      throw new Error(JSON.stringify(config));
    }

    this.executable = resolve(this.cwd, config.config.executable);

    this.process = spawn(
      'defmt-print',
      ['-e', this.executable],
      { cwd: this.cwd }
    );

    this.process.stdout.on('data', (data) => {
      this.displayOutput(data.toString());
    });

    this.process.stderr.on('data', (data) => {
        this.displayOutput(`stderr: ${data.toString()}`);
    });

    this.process.on('error', (error) => {
        this.displayOutput(`Error: ${error.message}\n`);
    });
  }

  softwareEvent(_port, data) {
    if (!this.process || !this.process.stdin) {
[... 296 lines cut ...]
Thread 1 "rp2040.core0" hit Temporary breakpoint 1, count::__cortex_m_rt_main_trampoline () at src\bin/count.rs:10
-> ~"10\t#[cortex_m_rt::entry]\n"
10	#[cortex_m_rt::entry]
-> *stopped,reason="breakpoint-hit",disp="del",bkptno="1",frame={addr="0x100002d4",func="count::__cortex_m_rt_main_trampoline",args=[],file="src\\bin/count.rs",fullname="C:\\Users\\user\\dev\\repo-pico-rust\\knurling-app-template\\src\\bin\\count.rs",line="10",arch="armv6s-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
-> =breakpoint-deleted,id="1"
25-thread-list-ids
-> 25^done,thread-ids={thread-id="1",thread-id="2"},current-thread-id="1",number-of-threads="2"
26-thread-info 1
-> 26^done,threads=[{id="1",target-id="Thread 1",details="Name: rp2040.core0, state: breakpoint",name="rp2040.core0",frame={level="0",addr="0x100002d4",func="count::__cortex_m_rt_main_trampoline",args=[],file="src\\bin/count.rs",fullname="C:\\Users\\user\\dev\\repo-pico-rust\\knurling-app-template\\src\\bin\\count.rs",line="10",arch="armv6s-m"},state="stopped"}]
27-thread-info 2
-> 27^done,threads=[{id="2",target-id="Thread 2",details="Name: rp2040.core1, state: debug-request",name="rp2040.core1",frame={level="0",addr="0x0000012a",func="??",args=[],arch="armv6s-m"},state="stopped"}]
28-stack-list-frames --thread 1 0 19
-> 28^done,stack=[frame={level="0",addr="0x100002d4",func="count::__cortex_m_rt_main_trampoline",file="src\\bin/count.rs",fullname="C:\\Users\\user\\dev\\repo-pico-rust\\knurling-app-template\\src\\bin\\count.rs",line="10",arch="armv6s-m"}]
LiveGDB: 13-var-update  --all-values *
LiveGDB: -> 13^done,changelist=[]
29-stack-info-frame --thread 1 --frame 0
-> 29^done,frame={level="0",addr="0x100002d4",func="count::__cortex_m_rt_main_trampoline",file="src\\bin/count.rs",fullname="C:\\Users\\user\\dev\\repo-pico-rust\\knurling-app-template\\src\\bin\\count.rs",line="10",arch="armv6s-m"}
30-stack-list-variables --thread 1 --frame 0 --simple-values
-> 30^done,variables=[]
```

**Additional context**

Everything else seems to work good.

mfp20, 2025-06-09:
> See [#1111](https://github.com/Marus/cortex-debug/issues/1111) (scroll to bottom)
> 
> Please close this if it addressed your issue.

I added a dispose() function but I don't know what code I need in order to write a message and leave it open for next debug session. See below

```
import { spawn } from 'child_process';
import { resolve } from 'path';
//import vscode from 'vscode';

class DefmtDecoder {
  constructor() {
    this.config = {};
    this.displayOutput = () => {};
    this.graphData = () => {};
    this.process = null;
  }

  init(config, displayOutput, graphData) {
    this.config = config;
    this.displayOutput = displayOutput;
    this.graphData = graphData;

    this.cwd = config.config.cwd || process.cwd();
    if (!config.config.executable) {
      throw new Error(JSON.stringify(config));
    }

    this.executable = resolve(this.cwd, config.config.executable);

    this.process = spawn(
      'defmt-print',
      ['-e', this.executable],
      { cwd: this.cwd }
    );

    this.process.stdout.on('data', (data) => {
      this.displayOutput(data.toString());
    });

    this.process.stderr.on('data', (data) => {
        this.displayOutput(`stderr: ${data.toString()}`);
    });

    this.process.on('error', (error) => {
        this.displayOutput(`Error: ${error.message}\n`);
    });
  }

  softwareEvent(_port, data) {
    if (!this.process || !this.process.stdin) {
      if (this.outputChannel) {
        this.displayOutput('Error: defmt-print process is not running.\n');
      }
      return;
    }
    let bufferInput = Buffer.isBuffer(data) ? data : Buffer.from(data);
    this.process.stdin.write(bufferInput);
  }

  outputLabel() {
    return this.config.label;
  }

  typeName() {
    return 'defmt-print';
  }

  dispose() {
    if (this.stream) {
      this.stream.end(() => {
          // need code to dispose the output tag (or leave it open for reusing on next debug session)
      });
    }
  };
}

export default DefmtDecoder;
```

haneefdm, 2025-06-09:
You can do nothing in `dispose`. Others were closing connections/files they opened and freeing up any other resources you allocated, end child processes, etc.