Marus/cortex-debug #1143: Reading Symbols from .elf file is very Slow with GNU 14.2

0shelter0, 2025-06-28:
**Bug Description**

This bug is similar with issue #539 . When I load the .elf file the reading process takes a long time (about 18

[debug_console_gnu_v10.2.txt](https://github.com/user-attachments/files/20958144/debug_console_gnu_v10.2.txt)
[debug_console_gnu_v14.2.txt](https://github.com/user-attachments/files/20958145/debug_console_gnu_v14.2.txt)

s), where the loaded `elf` file is size of 385 KB, GNU toolchain version is `14.2`， and using JLinkGDBServer.  However, when I change the GNU tools to version `10.2`, the time cost of reading symbols from elf file decrease significantly (less than 1s). I'm wondering if it's the Cortex-Debug or the gdb itself causing the long reading times.

**To Reproduce**
Steps to reproduce the behavior:

1. Setup the `launch.json` supplied  below.
2. Config the argument in Cortex-debug extension setting as `"cortex-debug.armToolchainPath": "D:\\path\\to\\your\\arm-gnu-toolchain-14.2\\bin"` .
3. Connect JLink with your target in type of SWD interface.
4. Start debug session.
5. See issue in the `Debug Console` window.

**Expected Output**

The expected **partial** output of Debug Console is shown as following **when using GNU 14.2**.

```shell
0000000603+00000: 8-interpreter-exec console "set output-radix 0xa"
0000000615+00012: -> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
0000000615+00000: Output radix now set to decimal 10, hex a, octal 12.
0000000615+00000: -> 8^done
0000000615+00000: 9-interpreter-exec console "set input-radix 0xa"
0000000630+00015: -> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
0000000630+00000: Input radix now set to decimal 10, hex a, octal 12.
0000000630+00000: -> 9^done
0000000630+00000: 10-interpreter-exec console "symbol-file \"D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf\""
0000000646+00016: -> ~"Reading symbols from D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf...\n"
0000000646+00000: Reading symbols from D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf...
0000018830+18184: -> 10^done
0000018830+00000: Debug Time: GDB Ready...
0000019342+00512: Debug Time: GDB Server post start events done...
0000019342+00000: Debug Time: objdump and nm done...
0000019342+00000: Debug Time: All pending items done, proceed to gdb connect...
```

As we can see, the process of "Reading symbols from `.elf` file" takes a extremely long time, about 18s, when using `GNU 14.2`.
In contrast, the time used of that is less than 1s when using `GNU 10.2`(see the attached file `debug_console_gnu_v10.2.txt`).

**Environment**

- Cortex-Debug Version: [1.13.0-pre3]
- Core: [ARM Cortex-M4]
- VSCode: [1.89.1]
- OS: [Windows 11]
- GDB: [14.2]
- Compiler Toolchain: [`arm-gnu-toolchain-14.2.rel1-mingw-w64-x86_64-arm-none-eabi`]
- JLink GDB Server: [`V7.98a`]

**launch.json**

The `launch.json` in VSCode is configured as below.

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug with J-Link",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/build/Test.hex",
            "symbolFiles": ["${workspaceFolder}/build/Test.elf"],
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "BS9146AM",
            "interface": "swd",
            "showDevDebugOutput": "raw",
            "showDevDebugTimestamps": true,
            "runToEntryPoint": "main",
            "preLaunchCommands": [
                "set mem inaccessible-by-default off",
                "monitor reset",
            ],
            "svdFile": "${workspaceFolder}/../../../Device/Arm/Svd/BS9X46AMXX.svd",
            "serverArgs": [
                "-speed",
                "4000",
                "-localhostonly",
                "1",
            ]
        }
    ]
}
```

0shelter0, 2025-06-28:
> Also, try to use the lastest Cortex-Debug to make sure issue still exists. We have been chaning how we use GNU tools forever and can't keep track. We have not found the magic decoder ring yet.
> 
> https://github.com/Marus/cortex-debug/releases

Thanks for your attention to this issue. I'd changed to the lastest Cortex-Debug  [V1.13.0-pre6](https://github.com/Marus/cortex-debug/releases/tag/v1.13.0-pre6), and the time used of reading symbols from `.elf` is magically reduced to **about 9s** when using GNU v14.2.  The output in debug console window is shown below.
```bash
0000000326+00000: -> 9^done
0000000326+00000: 10-interpreter-exec console "add-symbol-file \"D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf\""
0000000326+00000: -> ~"add symbol table from file \"D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf\"\n"
0000000326+00000: add symbol table from file "D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf"
0000000326+00000: -> ~"(y or n) [answered Y; input not from terminal]\n"
0000000326+00000: (y or n) [answered Y; input not from terminal]
0000000328+00002: -> ~"Reading symbols from D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf...\n"
0000000328+00000: Reading symbols from D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf...
0000009450+09122: -> 10^done
0000009450+00000: Debug Time: GDB Ready...
0000009965+00515: Debug Time: GDB Server post start events done...
0000009965+00000: Debug Time: objdump and nm done...
0000009965+00000: Debug Time: All pending items done, proceed to gdb connect...
0000009966+00001: 11-target-select extended-remote localhost:50004
0000010040+00074: -> =thread-group-started,id="i1",pid="42000"
``` 
Another aspect, I just used `gcc v14.2` command line to get the time used of process `objdump`, here the result is.
```bash
time "D:\arm-gnu-toolchain-14.2.rel1-mingw-w64-x86\bin\arm-none-eabi-objdump.exe" -Wi --syms -C Test.elf  > outfile.txt
real    0m0.542s
user    0m0.030s
sys     0m0.168s
```
It seems like that the time cost in `objdump` instruction didn't take long actually. Could there be a further reason here？

haneefdm, 2025-06-28:
Focus on these two lines. It is pure gdb and nobody else
```
0000000326+00000: 10-interpreter-exec console "add-symbol-file \"D:/work/temp/git_repo/MCAL/V1.1.2/bs9x46am_mcal/Projects/Can_Example/CMake/build/Test.elf\""

...
0000009450+09122: -> 10^done
```

Run gdb from the command line, enter the same commands we did (you can see them all) and observe the 10th command.

The 9s vs 18s maybe something to do with your Virus scanner. 9s is still a very long time for such a small elf file. Where is your D drive? Network?

On Windows, we had to disable virus scanners on directories containing our code and tools because it also affected gcc. An order-of-magnitude slowdown in compile time.

I don't see an issue with Cortex-Debug yet. The only variable you changed is the GNU tools version. If the issue is elsewhere, please close this issue. Try other versions of GNU/GCC

0shelter0, 2025-06-28:
Thank you for the advice. I will switch other versions of GCC to avoid this issue. 
Respect!