Marus/cortex-debug #1144: GDB server is not started even launch.json is configured

aaronjin, 2025-06-28:
I have a newbie's question and struggled a few days to get to this point. Looking for some tips before giving up.

I am able to run ST-LINK-gdbserver to debug my STM32F429 Discovery kit just fine command line. But I am unable to get GDB server running inside VS Code.

But I struggled to get the GDB server running from VS Code. I downloaded the latest code (with minor change in order to build VSIX file on Windows) but it didn't help.

The logs from VS Code debug console
--------------------------------------------------------------------------------------------------------------
Cortex-Debug: VSCode debugger extension version undefined git(0074bab+dirty). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Build & Debug Microcontroller - ST-Link",
    "cwd": "D:\\xfer2\\stm32_work\\my_test1",
    "type": "cortex-debug",
    "executable": "D:\\xfer2\\stm32_work\\my_test1\\build\\Debug\\my_test1.elf",
    "request": "launch",
    "servertype": "stlink",
    "device": "STM32F429ZITx",
    "interface": "swd",
    "serialNumber": "",
    "runToEntryPoint": "main",
    "showDevDebugOutput": "both",
    "svdFile": "d:\\STM32CubeCLT_1.18.0/STMicroelectronics_CMSIS_SVD/STM32F429.svd",
    "v1": false,
    "serverpath": "d:\\STM32CubeCLT_1.18.0/STLink-gdb-server/bin/ST-LINK_gdbserver",
    "stm32cubeprogrammer": "d:\\STM32CubeCLT_1.18.0/STM32CubeProgrammer/bin",
    "stlinkPath": "d:\\STM32CubeCLT_1.18.0/STLink-gdb-server/bin/ST-LINK_gdbserver",
    "armToolchainPath": "d:\\STM32CubeCLT_1.18.0/GNU-tools-for-STM32/bin",
    "gdbPath": "d:\\STM32CubeCLT_1.18.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb",
    "serverArgs": [
        "-m",
        "0"
    ],
    "__configurationTarget": 6,
    "__sessionId": "a7675b13-18c1-4a21-b342-40698b477ef5",
    "pvtShowDevDebugOutput": "both",
    "graphConfig": []
}
Reading symbols from d:/stm32cubeclt_1.18.0/gnu-tools-for-stm32/bin/arm-none-eabi-objdump --syms -C -h -w D:\xfer2\stm32_work\my_test1\build\Debug\my_test1.elf
Reading symbols from d:/stm32cubeclt_1.18.0/gnu-tools-for-stm32/bin/arm-none-eabi-nm --defined-only -S -l -C -p D:\xfer2\stm32_work\my_test1\build\Debug\my_test1.elf
Launching GDB: "d:\\STM32CubeCLT_1.18.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb" -q --interpreter=mi2
1-gdb-version
----------------------------------------------------------------------------
<NO more messages after the line above in DEBUG CONSOLE>

Can you share some lights what's quick worth trying? Thank you!!

haneefdm, 2025-06-28:
Looks like you are actually having trouble starting GDB. Make sure your gdb installation is working by running gdb from the command line. When we query for the gdb-version, it is not returning anything...you should see some stuff printed by gdb.

> Cortex-Debug: VSCode debugger extension version undefined git(0074bab+dirty). Usage info: https://github.com/Marus/cortex-debug#usage

Btw, I can't tell what version of Cortex-Debug you are running!?!?!

Are you using our latest from either VSCode marketplace or

https://github.com/Marus/cortex-debug/releases

Khaamy, 2025-07-01:
@haneefdm Thanks for a very quick answer. I was indeed using wrong GDB. After running the line you proposed I got an error message about missing libncurses.so.5, which is weird, because I tried installing libncurses today after reading another post... Now that I got some comprehensive error code i can continue on my own without wasting any more of your time. Thanks again! And good luck with your problem @aaronjin

haneefdm, 2025-07-01:
Yup, 9 out of 10 times. I don't know why Linux GDB releases are always incomplete. And, why OS gives no indication something is wrong on stdout/stderr. Good luck

aaronjin, 2025-07-07:
> "d:\\STM32CubeCLT_1.18.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb" -q --interpreter=mi2
> 1-gdb-version

Here is the output. What do you suggest next? (sorry for the delayed response as I was on vacation..)

```
D:\xfer2>"d:\\STM32CubeCLT_1.18.0\\GNU-tools-for-STM32\\bin\\arm-none-eabi-gdb" -q --interpreter=mi21-gdb-version
Interpreter `mi21-gdb-version' unrecognized
```

haneefdm, 2025-07-07:
This is the command you enter on the shell/cmd prompt
```
D:\xfer2>"d:\\STM32CubeCLT_1.18.0\\GNU-tools-for-STM32\\bin\\arm-none-eabi-gdb" -q --interpreter=mi2
```
Once gdb starts, this is the "GDB" command you enter
```
1-gdb-version
```
You mixed it all up. If you notice our debug log file, all gdb commands are prefixed by a command sequence number, but you haven't gotten that far.

aaronjin, 2025-07-07:
the gdb-server didn't seem to start at all nor any logs I can copy-n-paste to test from command line. That's why I am lost. 

Here is log output from VS code debug console. Meanwhile, I will examine the JSON file and see I missed anything important. 

```
Cortex-Debug: VSCode debugger extension version undefined git(0074bab+dirty). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Build & Debug Microcontroller - ST-Link",
    "cwd": "D:\\xfer2\\stm32_work\\my_test1",
    "type": "cortex-debug",
    "executable": "D:\\xfer2\\stm32_work\\my_test1\\build\\Debug\\my_test1.elf",
    "request": "launch",
    "servertype": "stlink",
    "device": "STM32F429ZITx",
    "interface": "swd",
    "serialNumber": "",
    "runToEntryPoint": "main",
    "showDevDebugOutput": "both",
    "svdFile": "d:\\STM32CubeCLT_1.18.0/STMicroelectronics_CMSIS_SVD/STM32F429.svd",
    "v1": false,
    "serverpath": "d:\\STM32CubeCLT_1.18.0\\STLink-gdb-server\\bin\\ST-LINK_gdbserver",
    "stm32cubeprogrammer": "d:\\STM32CubeCLT_1.18.0\\STM32CubeProgrammer\\bin",
    "stlinkPath": "d:\\STM32CubeCLT_1.18.0\\STLink-gdb-server\\bin\\ST-LINK_gdbserver",
    "armToolchainPath": "d:\\STM32CubeCLT_1.18.0\\GNU-tools-for-STM32\\bin",
    "gdbPath": "d:\\STM32CubeCLT_1.18.0\\GNU-tools-for-STM32\\bin\\arm-none-eabi-gdb",
    "serverArgs": [
        "-m",
        "0"
    ],
    "__configurationTarget": 6,
    "__sessionId": "d6453533-368d-46cd-befe-6cf7ac43c99a",
    "pvtShowDevDebugOutput": "both",
    "graphConfig": []
}
Reading symbols from d:/stm32cubeclt_1.18.0/gnu-tools-for-stm32/bin/arm-none-eabi-objdump --syms -C -h -w D:\xfer2\stm32_work\my_test1\build\Debug\my_test1.elf
Reading symbols from d:/stm32cubeclt_1.18.0/gnu-tools-for-stm32/bin/arm-none-eabi-nm --defined-only -S -l -C -p D:\xfer2\stm32_work\my_test1\build\Debug\my_test1.elf
Launching GDB: "d:\\STM32CubeCLT_1.18.0\\GNU-tools-for-STM32\\bin\\arm-none-eabi-gdb" -q --interpreter=mi2
1-gdb-version
```

Hossein-M98, 2025-09-18:
I am using [STM32Cube for Visual Studio Code](https://marketplace.visualstudio.com/items?itemName=stmicroelectronics.stm32-vscode-extension) extension and have the same problem. I tested it on two different systems, two different boards with STM32F429 and STM32U575, two different ST-Link v2.1 and v3, and I got the same answer.  This is my debug console:
```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Build & Debug Microcontroller - ST-Link",
    "cwd": "/tmp/test-stm32f4",
    "type": "cortex-debug",
    "executable": "/tmp/test-stm32f4/build/Debug/test-stm32f4.elf",
    "request": "launch",
    "servertype": "stlink",
    "device": "STM32F429ZITx",
    "interface": "swd",
    "serialNumber": "",
    "runToEntryPoint": "main",
    "svdFile": "/opt/st/stm32cubeclt_1.19.0/STMicroelectronics_CMSIS_SVD/STM32F429.svd",
    "v1": false,
    "serverpath": "/opt/st/stm32cubeclt_1.19.0/STLink-gdb-server/bin/ST-LINK_gdbserver",
    "stm32cubeprogrammer": "/opt/st/stm32cubeclt_1.19.0/STM32CubeProgrammer/bin",
    "stlinkPath": "/opt/st/stm32cubeclt_1.19.0/STLink-gdb-server/bin/ST-LINK_gdbserver",
    "armToolchainPath": "/opt/st/stm32cubeclt_1.19.0/GNU-tools-for-STM32/bin",
    "gdbPath": "/opt/st/stm32cubeclt_1.19.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb",
    "serverArgs": [
        "-m",
        "0"
    ],
    "showDevDebugOutput": "raw",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "toolchainPath": "/opt/st/stm32cubeclt_1.19.0/GNU-tools-for-STM32/bin",
    "toolchainPrefix": "arm-none-eabi",
    "extensionPath": "/home/hossein/.vscode/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "92de57a1-1385-4ff4-803a-c0f8b6a43011",
    "pvtShowDevDebugOutput": "raw"
}
Reading symbols from /opt/st/stm32cubeclt_1.19.0/GNU-tools-for-STM32/bin/arm-none-eabi-objdump --syms -C -h -w /tmp/test-stm32f4/build/Debug/test-stm32f4.elf
Reading symbols from /opt/st/stm32cubeclt_1.19.0/GNU-tools-for-STM32/bin/arm-none-eabi-nm --defined-only -S -l -C -p /tmp/test-stm32f4/build/Debug/test-stm32f4.elf
Launching GDB: /opt/st/stm32cubeclt_1.19.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
```
Everything is installed, I can run the GDB server by myself or flash the microcontroller, but debugging with the VSCODE does not work! It seems to me that every time I try to use the debugger, a process starts and listens on port number 55878.
[image]

Finally, it might be good to add that I am using the Debian 13 and VSCODE version 1.104.0.

haneefdm, 2025-09-18:
Don't think it is your gdb server that is the problem. Your GDB install may be an issue. Very common on Linux

> 1-gdb-version

If it does not go past that line, GDB never responds. Make sure your gdb installation is complete (like no missing libraries like libcurses)

> /opt/st/stm32cubeclt_1.19.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb -q --interpreter=mi2

Make sure the above command line works properly and you can run the command `1-gdb-version`

Btw, the `55878` is a port used between the backend of Cortex-Debug and the Console Terminal Window that is the front end (UI). Nothing to do with any gdb-server or gdb.

haneefdm, 2025-09-18:
Note that if you look at the commands, we never even gotten to even starting the gdb-server.

Hossein-M98, 2025-09-18:
@haneefdm 
Thank you for your quick and helpful answer.
As you mentioned, the problem was "arm-none-eabi-gdb" and its dependency on some old libraries that were not installed on my PC. I replaced "/opt/st/stm32cubeclt_1.19.0/GNU-tools-for-STM32/bin/arm-none-eabi-gdb" with a symbolic link to "/usr/bin/gdb-multiarch" and the problem disappeared.

haneefdm, 2025-09-18:
This happened in dozens of issues. Search the Issues for this problem.  We tried to detect it but on Linux, we can't evem tell that something is wrong.

I am closing this issue as the original author for this Issue ihas not responded in a while