Marus/cortex-debug #1150: RFC: Automatically detect connected devices and provide a selection menu when starting a debug session

axel-lebourhis, 2025-07-29:
Hello !

I’d like to propose a feature based on my personal experience, and I’m happy to contribute to its development. I'm opening this issue as a Request for Comments (RFC) to gather your thoughts on its feasibility and whether you'd be open to including such a feature in the project.

**Is your feature request related to a problem? Please describe.**

I use cortex-debug daily to debug multiple targets with J-Link. My setup involves building and debugging on a Linux PC, while using Visual Studio Code remotely from a Windows machine.
Managing multiple targets becomes cumbersome due to the need to handle different serial numbers. It’s hard to remember which serial number corresponds to which board, and switching between targets multiple times a day adds to the complexity.
To avoid duplicating debug configurations for each serial number, I created a solution using:
* Python scripts and udev rules to automatically populate a JSON file with all connected J-Link devices.
* The file is updated dynamically whenever a device is connected or disconnected.
* I use another extension (command-variable) that reads from this JSON file and injects the selected serial number into the serialNumber field of the launch configuration.

This setup works great for me, and I believe it could be a valuable native feature in cortex-debug.

**Describe the solution you'd like**

I’d like to integrate this device detection mechanism directly into the extension. The idea is to:
* Automatically detect connected J-Link devices.
* Populate a JSON file (or similar structure) with their details.
* Allow the user to select a device at debug time, with the selected serial number passed to the configuration.

This feature could be toggled via a new boolean setting, or perhaps by assigning a special value to serialNumber to enable detection mode.

I’m still exploring how best to integrate with udev or whether the extension could respond to external triggers (e.g., udev events). I'm also unsure how feasible this would be on Windows or macOS, as I’m less familiar with those platforms.

Let me know what you think! I’d love to hear your feedback and see if this is something you'd consider for the project.

haneefdm, 2025-08-09:
This. Please do what @hwmaier suggests. I really don't want to get too involved with JLink, USB or how things work in various OSes/WSL (udev). It just opens up a Pandora's box, and I can't handle the support issues.

You can also create your own extension that even supplies the entire configuration or provides substitutions for launch.json as shown above by @hwmaier. There are other methods too built into VSCode

hwmaier, 2025-08-10:
@axel-lebourhis Based on @haneefdm's comment, why not consider implementing your idea into its own extension? You could have configuration and USB interrogation of  Jlinks in such an extension and then feed the results in form of input variables into launch`.json`. I think it would be a useful add-on as I also wrote my own Python script to do exactly that, so there is a need for it for people with multiple targets and multiple JLinks.

Btw, instead of interacting with *udev* it may be easier to use the Jlink API to check what's connected. There is a Python API called `pylink` which has a `connected_emulators()` method. Maybe there is a similar API available for Javascript. That would make the USB code more portable.

axel-lebourhis, 2025-08-11:
@haneefdm I understand your concern, it makes sense.

@hwmaier thanks for your suggestions, I didn't know about the cmake inputs. I also work with Zephyr, so I would be interested to see how you link zephyr's cmake generation with the cmake extension ?
I will also look at `pylink`, sounds promising ! If we can avoid os specific dependencies it's indeed the way to go.

I'll think about starting an extension for that :) 

Thanks to the both of you for the feedback, appreciate it.

closed 2025-08-11 by haneefdm